awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #307: RTMP Support not available

philross88, 2019-12-15:
The RMTP support is not built into docker image or even if it's compiled with the instructions using install script. Please build the rtmps plugins as well.

/gst-launch-1.0 -v rtmpsrc
ERROR: pipeline could not be constructed: no element "rtmpsrc".

MushMal, 2019-12-15:
@philross88 thanks for reporting. This belongs to plugins-bad. We can evaluate whether to include it in the SDK itself or not. Meanwhile, you can simply pull down the plugin directly and use it - GStreamer and it's plugins are not really part of the SDK.

philross88, 2020-01-09:
Thanks @MushMal.
Can you please explain with instructions on how to pull down the plugins-bad in the docker image?
Yum in the docker is broken and I don't see an option to pull down the plugins-bad.

Any details would be great.

MushMal, 2020-03-17:
Resolving as we have no immediate plans on supporting RTMP in the KVS SDK. RTMP can still be supported via GStreamer

closed 2020-03-17 by MushMal