awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #308: Cannot successfully run kinesis_video_gstreamer_sample_app.exe

istranic, 2019-12-17:
Hello,
 
I'm having trouble running kinesis_video_gstreamer_sample_app.exe (./kinesis_video_gstreamer_sample_app.exe pc_stream). I followed the instruction and ran it using MinGw. I get the following error:

[INFO ] [17-12-2019 01:20:07:055.849 Eastern Standard Time] Using region: us-west-2
[INFO ] [17-12-2019 01:20:07:055.849 Eastern Standard Time] Using aws credentials for Kinesis Video Streams
[INFO ] [17-12-2019 01:20:07:055.849 Eastern Standard Time] No session token was detected.
[INFO ] [17-12-2019 01:20:07:059.838 Eastern Standard Time] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [17-12-2019 01:20:07:059.838 Eastern Standard Time] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [17-12-2019 01:20:07:059.838 Eastern Standard Time] heapInitialize(): Creating AIV heap.
[INFO ] [17-12-2019 01:20:07:059.838 Eastern Standard Time] heapInitialize(): Heap is initialized OK
[DEBUG] [17-12-2019 01:20:07:059.838 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [17-12-2019 01:20:07:060.835 Eastern Standard Time] getSecurityTokenHandler invoked
[DEBUG] [17-12-2019 01:20:07:060.835 Eastern Standard Time] Refreshing credentials. Force refreshing: 0 Now time is: 1576545607060835600 Expiration: 0
[INFO ] [17-12-2019 01:20:07:060.835 Eastern Standard Time] New credentials expiration is 1576548007
[DEBUG] [17-12-2019 01:20:07:060.835 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [17-12-2019 01:20:07:060.835 Eastern Standard Time] createDeviceResultEvent(): Create device result event.
[DEBUG] [17-12-2019 01:20:07:060.835 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [17-12-2019 01:20:07:060.835 Eastern Standard Time] clientReadyHandler invoked
[DEBUG] [17-12-2019 01:20:07:060.835 Eastern Standard Time] Client is ready
[INFO ] [17-12-2019 01:20:07:060.835 Eastern Standard Time] Creating Kinesis Video Stream pc_stream
[INFO ] [17-12-2019 01:20:07:060.835 Eastern Standard Time] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [17-12-2019 01:20:07:060.835 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [17-12-2019 01:20:07:443.359 Eastern Standard Time] writeHeaderCallback(): RequestId: c91d98b2-cac6-4a70-8163-1629fd3582b1
[DEBUG] [17-12-2019 01:20:07:443.359 Eastern Standard Time] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.576543255123E9,"DataRetentionInHours":24,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-west-2:132756671989:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:132756671989:stream/pc_stream/1576543255123","StreamName":"pc_stream","Version":"QdD5LaFjaoidmErfR0xe"}}
[INFO ] [17-12-2019 01:20:07:446.350 Eastern Standard Time] describeStreamResultEvent(): Describe stream result event.
[WARN ] [17-12-2019 01:20:07:446.350 Eastern Standard Time] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [17-12-2019 01:20:07:446.350 Eastern Standard Time] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [17-12-2019 01:20:07:446.350 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [17-12-2019 01:20:07:782.450 Eastern Standard Time] writeHeaderCallback(): RequestId: 5aeff8dc-b078-4c3d-8303-53af6a4b3ce9
[DEBUG] [17-12-2019 01:20:07:782.450 Eastern Standard Time] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ] [17-12-2019 01:20:07:784.446 Eastern Standard Time] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [17-12-2019 01:20:07:784.446 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x00000000000
(kinesis_video_gstreamer_sample_app.exe:25084): GLib-GObject-WARNING **: 20:20:08.075: g_object_set_is_valid_property: object class 'GstKsVideoSrc' has no property named 'device'
Exception code=0xc0000005 flags=0x0 at 0x000000000C337F00. Access violation - attempting to execute data at address 0x000000000C337F00
Segmentation fault

I also tried:

gst-launch-1.0 filesrc location="*******.mp4"  ! decodebin ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="pc_stream" storage-size=512 access-key="AKIAR52HKBX2YAP37BMB" secret-key="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" aws-region="us-west-2"

and I get the same access violation at the end

Any help would be greatly appreciated!

MushMal, 2019-12-17:
PLEASE DO NOT INCLUDE SECRET KEYS!!! Delete this key right away.

@istranic seems like a GStreamer plugin GstKsVideoSrc fails with access violation attempting to get or set a property called device. 

Can you please give us a little more info on how you've installed the GStreamer and it's plugins? Can you run with env variable GST_DEBUG=4 to get more GStreamer logs?

chehefen, 2019-12-17:
can you run gst-device-monitor-1.0 and see what it prints?

istranic, 2019-12-17:
Thanks for the quick response 

@MushMal Noob move re. the keys. Thx for the reminder. I never explicitly installed the GStreamer plugin. I just followed the instruction in the "Building the Producer SDK Using MinGW" section of https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-win.html and assumed that installs GStreamer. Is that not correct?

Here's the output with GST_DEBUG = 4

$ ./kinesis_video_gstreamer_sample_app.exe pc_stream
[INFO ] [17-12-2019 01:47:23:712.422 Eastern Standard Time] Using region: us-west-2
[INFO ] [17-12-2019 01:47:23:712.422 Eastern Standard Time] Using aws credentials for Kinesis Video Streams
[INFO ] [17-12-2019 01:47:23:712.422 Eastern Standard Time] No session token was detected.
[INFO ] [17-12-2019 01:47:23:717.419 Eastern Standard Time] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [17-12-2019 01:47:23:717.419 Eastern Standard Time] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [17-12-2019 01:47:23:717.419 Eastern Standard Time] heapInitialize(): Creating AIV heap.
[INFO ] [17-12-2019 01:47:23:717.419 Eastern Standard Time] heapInitialize(): Heap is initialized OK
[DEBUG] [17-12-2019 01:47:23:717.419 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [17-12-2019 01:47:23:717.419 Eastern Standard Time] getSecurityTokenHandler invoked
[DEBUG] [17-12-2019 01:47:23:717.419 Eastern Standard Time] Refreshing credentials. Force refreshing: 0 Now time is: 1576547243717419000 Expiration: 0
[INFO ] [17-12-2019 01:47:23:717.419 Eastern Standard Time] New credentials expiration is 1576549643
[DEBUG] [17-12-2019 01:47:23:717.419 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [17-12-2019 01:47:23:717.419 Eastern Standard Time] createDeviceResultEvent(): Create device result event.
[DEBUG] [17-12-2019 01:47:23:717.419 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [17-12-2019 01:47:23:717.419 Eastern Standard Time] clientReadyHandler invoked
[DEBUG] [17-12-2019 01:47:23:717.419 Eastern Standard Time] Client is ready
[INFO ] [17-12-2019 01:47:23:717.419 Eastern Standard Time] Creating Kinesis Video Stream pc_stream
[INFO ] [17-12-2019 01:47:23:718.378 Eastern Standard Time] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [17-12-2019 01:47:23:718.378 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [17-12-2019 01:47:24:094.371 Eastern Standard Time] writeHeaderCallback(): RequestId: 26a02613-a3f9-495d-b9bd-ac15647ded7c
[DEBUG] [17-12-2019 01:47:24:094.371 Eastern Standard Time] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.576543255123E9,"DataRetentionInHours":24,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-west-2:132756671989:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:132756671989:stream/pc_stream/1576543255123","StreamName":"pc_stream","Version":"QdD5LaFjaoidmErfR0xe"}}
[INFO ] [17-12-2019 01:47:24:096.367 Eastern Standard Time] describeStreamResultEvent(): Describe stream result event.
[WARN ] [17-12-2019 01:47:24:096.367 Eastern Standard Time] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [17-12-2019 01:47:24:096.367 Eastern Standard Time] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [17-12-2019 01:47:24:096.367 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [17-12-2019 01:47:24:439.450 Eastern Standard Time] writeHeaderCallback(): RequestId: b057979a-0960-4be4-aad4-a7993122d4ae
[DEBUG] [17-12-2019 01:47:24:439.450 Eastern Standard Time] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ] [17-12-2019 01:47:24:441.453 Eastern Standard Time] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [17-12-2019 01:47:24:441.453 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x000000000000:00:00.001923000 24636        36d4000 INFO                GST_INIT gst.c:586:init_pre: Initializing GStreamer Core Library version 1.16.2
0:00:00.001982400 24636        36d4000 INFO                GST_INIT gst.c:587:init_pre: Using library installed in C:\msys64\mingw64\lib
0:00:00.002340400 24636        36d4000 INFO                GST_INIT gstmessage.c:128:_priv_gst_message_initialize: init messages
0:00:00.002983900 24636        36d4000 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.006659500 24636        36d4000 INFO      GST_PLUGIN_LOADING gstplugin.c:318:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.006805800 24636        36d4000 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: registered static plugin "staticelements"
[... cut ...]
0:00:00.528280400 24636        36d4000 INFO              GST_STATES gstelement.c:2575:_priv_gst_element_state_changed:<source> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:00.528338300 24636        36d4000 INFO              GST_STATES gstbin.c:2952:gst_bin_change_state_func:<live-kinesis-pipeline> child 'source' changed state to 4(PLAYING) successfully
Exception code=0xc0000005 flags=0x0 at 0x000000000C537F00. Access violation - attempting to execute data at address 0x000000000C537F00
Segmentation fault

istranic, 2019-12-17:
@chehefen Here's the output of gst-device-monitor-1.0
$ gst-device-monitor-1.0
0:00:00.000286900 27704        2594000 INFO                GST_INIT gst.c:586:init_pre: Initializing GStreamer Core Library version 1.16.2
0:00:00.000354800 27704        2594000 INFO                GST_INIT gst.c:587:init_pre: Using library installed in C:\msys64\mingw64\lib
0:00:00.000693900 27704        2594000 INFO                GST_INIT gstmessage.c:128:_priv_gst_message_initialize: init messages
0:00:00.001287600 27704        2594000 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.005024700 27704        2594000 INFO      GST_PLUGIN_LOADING gstplugin.c:318:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.005164200 27704        2594000 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.005203200 27704        2594000 INFO      GST_PLUGIN_LOADING gstplugin.c:228:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.020395000 27704        2594000 INFO            GST_REGISTRY gstregistry.c:1720:ensure_current_registry: reading registry cache: C:\Users\istranic\AppData\Local\Microsoft\Windows\INetCache\gstreamer-1.0\registry.x86_64.bin
0:00:00.036691500 27704        2594000 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded C:\Users\istranic\AppData\Local\Microsoft\Windows\INetCache\gstreamer-1.0\registry.x86_64.bin in 0.016197 seconds
0:00:00.036878000 27704        2594000 INFO            GST_REGISTRY gstregistry.c:1579:scan_and_update_registry: Validating plugins from registry cache: C:\Users\istranic\AppData\Local\Microsoft\Windows\INetCache\gstreamer-1.0\registry.x86_64.bin
0:00:00.241515500 27704        2594000 INFO            GST_REGISTRY gstregistry.c:1678:scan_and_update_registry: Registry cache has not changed
0:00:00.241558900 27704        2594000 INFO            GST_REGISTRY gstregistry.c:1755:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.241589100 27704        2594000 INFO                GST_INIT gst.c:806:init_post: GLib runtime version: 2.62.2
0:00:00.241616900 27704        2594000 INFO                GST_INIT gst.c:808:init_post: GLib headers version: 2.62.2
0:00:00.241643200 27704        2594000 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
Probing devices...

0:00:00.241789600 27704        2594000 WARN                 default gstdevicemonitor.c:458:gst_device_monitor_start:<devicemonitor0> No filters have been set, will expose all devices found
0:00:00.249923600 27704        2594000 INFO      GST_PLUGIN_LOADING gstplugin.c:902:_priv_gst_plugin_load_file_for_registry: plugin "C:\msys64\mingw64\lib\gstreamer-1.0\libgstdirectsound.dll" loaded
0:00:00.249968700 27704        2594000 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:271:gst_device_provider_factory_get: getting device provider "directsoundsinkdeviceprovider"
0:00:00.252219600 27704        2594000 INFO      GST_PLUGIN_LOADING gstplugin.c:902:_priv_gst_plugin_load_file_for_registry: plugin "C:\msys64\mingw64\lib\gstreamer-1.0\libgstdirectsoundsrc.dll" loaded
0:00:00.252294300 27704        2594000 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:271:gst_device_provider_factory_get: getting device provider "directsoundsrcdeviceprovider"
0:00:00.255008600 27704        2594000 INFO      GST_PLUGIN_LOADING gstplugin.c:902:_priv_gst_plugin_load_file_for_registry: plugin "C:\msys64\mingw64\lib\gstreamer-1.0\libgstwinks.dll" loaded
0:00:00.255051100 27704        2594000 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:271:gst_device_provider_factory_get: getting device provider "ksdeviceprovider"
0:00:00.256781100 27704        2594000 INFO      GST_PLUGIN_LOADING gstplugin.c:902:_priv_gst_plugin_load_file_for_registry: plugin "C:\msys64\mingw64\lib\gstreamer-1.0\libgstwasapi.dll" loaded
0:00:00.256822900 27704        2594000 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:271:gst_device_provider_factory_get: getting device provider "wasapideviceprovider"
0:00:00.289232800 27704        2594000 INFO          device-monitor gst-device-monitor.c:345:main: Took 0.05 seconds
0:00:00.310205300 27704        2594000 INFO                 default gstdirectsounddevice.c:118:gst_directsound_enum_callback:<directsoundsinkdeviceprovider0> sound device name: Primary Sound Driver,  (GUID (NULL))
0:00:00.310334400 27704        2594000 INFO                 default gstdirectsounddevice.c:118:gst_directsound_enum_callback:<directsoundsinkdeviceprovider0> sound device name: DELL S3219D (Intel(R) Display Audio), {0.0.0.00000000}.{1261580d-c3cf-4b33-b481-51008c01c150} (GUID {1261580D-C3CF-4B33-B481-51008C01C150})
0:00:00.310401600 27704        2594000 INFO                 default gstdirectsounddevice.c:118:gst_directsound_enum_callback:<directsoundsinkdeviceprovider0> sound device name: Speakers (Realtek High Definition Audio(SST)), {0.0.0.00000000}.{3fafe4f9-b578-4994-ba43-455f9b7f1a2b} (GUID {3FAFE4F9-B578-4994-BA43-455F9B7F1A2B})
0:00:00.339496700 27704        2594000 INFO                 default gstdirectsounddevice.c:118:gst_directsound_enum_callback:<directsoundsrcdeviceprovider0> sound device name: Primary Sound Capture Driver,  (GUID (NULL))
[... cut ...]
0:00:00.355095400 27704        2594000 INFO         GST_REFCOUNTING gstelement.c:3296:gst_element_dispose:<wasapisink3> 000000000308f800 parent class dispose
0:00:00.355143800 27704        2594000 INFO         GST_REFCOUNTING gstelement.c:3328:gst_element_finalize:<wasapisink3> 000000000308f800 finalize
0:00:00.355177400 27704        2594000 INFO         GST_REFCOUNTING gstelement.c:3333:gst_element_finalize:<wasapisink3> 000000000308f800 finalize parent
        gst-launch-1.0 ... ! wasapisink device="\{0.0.0.00000000\}.\{1261580d-c3cf-4b33-b481-51008c01c150\}"

chehefen, 2019-12-18:
can you try run this?
```
gst-launch-1.0 ksvideosrc device-path="\\?\display#int3470#4&5b5cba1&0&uid13424#{6994ad05-93ef-11d0-a3cc-00a0c9223196}\{7c9bbcea-909c-47b3-8cf9-2aa8237e1d4b}" ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="pc_stream" storage-size=512 access-key="XXXXXX" secret-key="XXXXXXX" aws-region="us-west-2"
```

istranic, 2019-12-19:
@chehefen Thanks for your help. FYI, I'm in Windows. I abandoned the MinGW installation and switched to MSVC. After going through all the steps, I get the following errors:

**When I run:
gst-launch-1.0 ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name="test-stream" access-key=YY secret-key=XX**

WARNING: erroneous pipeline: no element "kvssink"

**When I run gst-inspect-1.0 kvssink, i get:**

Plugin Details:
  Name                     kvssink
  Description              GStreamer AWS KVS plugin
  Filename                 XXXX\Aws\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\gstkvssink.dll
  Version                  1.0
  License                  Proprietary
  Source module            kvssinkpackage
  Binary package           GStreamer
  Origin URL               http://gstreamer.net/

**When I run kinesis_video_gstreamer_sample_app.exe test-stream, I get:**

[INFO ] [19-12-2019 20:18:07:314.557 Eastern Standard Time] Using region: us-west-2
[INFO ] [19-12-2019 20:18:07:314.557 Eastern Standard Time] Using aws credentials for Kinesis Video Streams
[INFO ] [19-12-2019 20:18:07:314.557 Eastern Standard Time] No session token was detected.
[INFO ] [19-12-2019 20:18:07:314.557 Eastern Standard Time] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [19-12-2019 20:18:07:314.557 Eastern Standard Time] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [19-12-2019 20:18:07:314.557 Eastern Standard Time] heapInitialize(): Creating AIV heap.
[INFO ] [19-12-2019 20:18:07:314.557 Eastern Standard Time] heapInitialize(): Heap is initialized OK
[DEBUG] [19-12-2019 20:18:07:314.557 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [19-12-2019 20:18:07:314.557 Eastern Standard Time] getSecurityTokenHandler invoked
[DEBUG] [19-12-2019 20:18:07:314.557 Eastern Standard Time] Refreshing credentials. Force refreshing: 0 Now time is: 15767866873285908 Expiration: 0
[INFO ] [19-12-2019 20:18:07:314.557 Eastern Standard Time] New credentials expiration is 1576789087
[DEBUG] [19-12-2019 20:18:07:314.557 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [19-12-2019 20:18:07:330.179 Eastern Standard Time] createDeviceResultEvent(): Create device result event.
[DEBUG] [19-12-2019 20:18:07:330.179 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [19-12-2019 20:18:07:330.179 Eastern Standard Time] clientReadyHandler invoked
[DEBUG] [19-12-2019 20:18:07:330.179 Eastern Standard Time] Client is ready
[INFO ] [19-12-2019 20:18:07:330.179 Eastern Standard Time] Creating Kinesis Video Stream test-stream
[INFO ] [19-12-2019 20:18:07:330.179 Eastern Standard Time] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [19-12-2019 20:18:07:330.179 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [19-12-2019 20:18:07:744.445 Eastern Standard Time] writeHeaderCallback(): RequestId: 1845420f-f454-48ae-8616-4a196771128d
[DEBUG] [19-12-2019 20:18:07:745.410 Eastern Standard Time] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.576539230926E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:132756671989:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:132756671989:stream/test-stream/1576539230926","StreamName":"test-stream","Version":"cvfjZj5hVXHmmbn793aE"}}
[INFO ] [19-12-2019 20:18:07:748.402 Eastern Standard Time] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [19-12-2019 20:18:07:748.402 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [19-12-2019 20:18:08:103.453 Eastern Standard Time] writeHeaderCallback(): RequestId: 57f1f093-3576-472f-99a3-b239b92f2176
[DEBUG] [19-12-2019 20:18:08:104.450 Eastern Standard Time] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ] [19-12-2019 20:18:08:107.442 Eastern Standard Time] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [19-12-2019 20:18:08:107.442 Eastern Standard Time] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [19-12-2019 20:18:08:109.437 Eastern Standard Time] getStreamingTokenHandler invoked
[DEBUG] [19-12-2019 20:18:08:111.432 Eastern Standard Time] Refreshing credentials. Force refreshing: 1 Now time is: 15767866881115045 Expiration: 1576789087
[INFO ] [19-12-2019 20:18:08:115.421 Eastern Standard Time] New credentials expiration is 1576789088
[INFO ] [19-12-2019 20:18:08:117.416 Eastern Standard Time] getStreamingTokenResultEvent(): Get streaming token result event.
[... cut ...]
[DEBUG] [19-12-2019 20:18:08:241.085 Eastern Standard Time] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512

(kinesis_video_gstreamer_sample_app.exe:21968): GLib-GObject-WARNING **: g_object_set_is_valid_property: object class 'GstKsVideoSrc' has no property named 'device'

The webcam light turns on, but I can't see the stream in AWS kinesis. In the media playback, i get errors like "Verify that you are streaming to us-west-2 and the fragment timecodes are correct, or download the Kinesis Video Streams Producer SDK to set up your device."

zhiyua-git, 2019-12-19:
Hi @istranic ,

Can you run gst-launch-1.0 command and kinesis_video_gstreamer_sample_app.exe test-stream again with **set GST_DEBUG=4**?

Can you run kinesis_video_gstreamer_sample_app.exe for more than 2 mins? If you have connection issue to KVS backend, we will see error message when you let it run for 2 mins.

Thanks,
-ZH

istranic, 2019-12-19:
@zhiyua-git Sure. Thanks for the help

For the gst-launch-1.0 command, the full output is:
0:00:00.310217300 28716 0000000002644000 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "C:\Users\istranic\Aws\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstwinks.dll" loaded
0:00:00.326106300 28716 0000000002644000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "ksvideosrc"
0:00:00.332907000 28716 0000000002644000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseSrc@0000000003439570> adding pad 'src'
0:00:00.349082600 28716 0000000002644000 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "C:\Users\istranic\Aws\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstvideoconvert.dll" loaded
0:00:00.361583900 28716 0000000002644000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "videoconvert"
0:00:00.367961700 28716 0000000002644000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0000000002733630> adding pad 'sink'
0:00:00.373274900 28716 0000000002644000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0000000002733630> adding pad 'src'
0:00:00.401924000 28716 0000000002644000 INFO                 x264enc gstx264enc.c:2953:plugin_init: linked against x264 build: 148
0:00:00.413077600 28716 0000000002644000 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "C:\Users\istranic\Aws\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstx264.dll" loaded
0:00:00.427170100 28716 0000000002644000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "x264enc"
0:00:00.438721400 28716 0000000002644000 INFO                 x264enc gstx264enc.c:227:gst_x264_enc_add_x264_chroma_format: 8-bit depth supported
0:00:00.445994000 28716 0000000002644000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoEncoder@00000000034447B0> adding pad 'sink'
0:00:00.451545100 28716 0000000002644000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoEncoder@00000000034447B0> adding pad 'src'
0:00:00.460890900 28716 0000000002644000 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:456:gst_element_factory_make: no such element factory "kvssink"!
0:00:00.469370700 28716 0000000002644000 ERROR           GST_PIPELINE grammar.y:816:priv_gst_parse_yyparse: no element "kvssink"
0:00:00.474112400 28716 0000000002644000 ERROR           GST_PIPELINE grammar.y:901:priv_gst_parse_yyparse: link has no sink [source=@00000000034447B0]
0:00:00.480072300 28716 0000000002644000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "pipeline"
0:00:00.489010600 28716 0000000002644000 INFO            GST_PIPELINE grammar.y:652:gst_parse_perform_link: linking some pad of GstKsVideoSrc named ksvideosrc0 to some pad of GstVideoConvert named videoconvert0 (0/0) with caps "video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1"
0:00:00.505040700 28716 0000000002644000 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "C:\Users\istranic\Aws\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\gstreamer\1.0\x86_64\lib\gstreamer-1.0\libgstcoreelements.dll" loaded
0:00:00.515973500 28716 0000000002644000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "capsfilter"
0:00:00.523733200 28716 0000000002644000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@00000000027380E0> adding pad 'sink'
0:00:00.530499300 28716 0000000002644000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@00000000027380E0> adding pad 'src'
0:00:00.538275800 28716 0000000002644000 INFO              GST_STATES gstbin.c:2087:gst_bin_get_state_func:<pipeline0> getting state
0:00:00.546961200 28716 0000000002644000 INFO              GST_STATES gstelement.c:2676:gst_element_continue_state:<capsfilter0> completed state change to NULL
0:00:00.560738800 28716 0000000002644000 INFO               GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event
0:00:00.572719700 28716 0000000002644000 INFO        GST_ELEMENT_PADS gstutils.c:1774:gst_element_link_pads_full: trying to link element ksvideosrc0:(any) to element capsfilter0:sink
0:00:00.590953500 28716 0000000002644000 INFO        GST_ELEMENT_PADS gstelement.c:920:gst_element_get_static_pad: found pad capsfilter0:sink
0:00:00.600900900 28716 0000000002644000 INFO                GST_PADS gstutils.c:1588:prepare_link_maybe_ghosting: ksvideosrc0 and capsfilter0 in same bin, no need for ghost pads
0:00:00.613537000 28716 0000000002644000 INFO                GST_PADS gstpad.c:2378:gst_pad_link_prepare: trying to link ksvideosrc0:src and capsfilter0:sink
0:00:00.621962700 28716 0000000002644000 INFO                GST_PADS gstpad.c:4230:gst_pad_peer_query:<capsfilter0:src> pad has no peer
[... cut ...]
0:00:00.769983100 28716 0000000002644000 INFO                GST_PADS gstpad.c:2586:gst_pad_link_full: linked videoconvert0:src and x264enc0:sink, successful
0:00:00.774986200 28716 0000000002644000 INFO               GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event
0:00:00.780041600 28716 0000000002644000 INFO               GST_EVENT gstpad.c:5785:gst_pad_send_event_unchecked:<videoconvert0:src> Received event on flushing pad. Discarding
WARNING: erroneous pipeline: no element "kvssink"

zhiyua-git, 2019-12-19:
Hi @istranic ,

The path you provided contains both C:\Users\ZZZ\Aws\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build and C:\Users\istranic\Aws\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build, are both of them correct? 

By saying “ I see activity in the Kinesis Video web dashboard”, what activity do you mean?

Thanks
-ZH

istranic, 2019-12-19:
Yes, they are both correct. They are the same path. I wrote in the ZZZ in order to obscure something. 

Here's what the dashboard looks like:

[image]

zhiyua-git, 2019-12-20:
Hi @istranic ,

I checked the backend log and saw your gstreamer app was able to reach KVS backend but your fragment is more than 10 seconds.

Can you send your complete log with GST_DEBUG=4 on your kinesis_video_gstreamer_sample_app.exe test-stream to kinesis-video-support@amazon.com so we can help? Thanks.

istranic, 2019-12-20:
@zhiyua-git  Email sent. Thank you.

zhiyua-git, 2019-12-21:
Hi @istranic , by setting environment variable set DEBUG_DUMP_FRAME_INFO=TRUE, the log shows your camera is sending at 1 fps even after gstreamer try to set it to 30 fps, by changing this line: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L798
To g_object_set(G_OBJECT (encoder), "bframes", 0, "key-int-max", 5, "bitrate", bitrateInKBPS, NULL); 
It forms a 5 second fragment on 1 fps and then it works. 

Closing the issue as you can run the app now, thanks for the detailed logs.

closed 2019-12-21 by zhiyua-git