awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #309: Test Failed on GM8136

hsinghaowang, 2019-12-19:
I pulled the latest code today, and build tests on our platform.
It had segmentation fault. Any idea?

```
[root@GM]# ./kvspic_test
[==========] Running 1825 tests from 55 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from ClientApiFunctionalityTest
[ RUN      ] ClientApiFunctionalityTest.createKinesisVideoClient_AuthIntegration

[0x0000000076f0e000] [level 3] KinesisVideoClient createKinesisVideoClient(): Creating Kinesis Video Client
[0x0000000076f0e000] [level 3] Heap heapInitialize(): Initializing native heap with limit size 10485760, spill ratio 0% and flags 0x00000001
[0x0000000076f0e000] [level 3] Heap heapInitialize(): Creating AIV heap.
[0x0000000076f0e000] [level 3] Heap heapInitialize(): Heap is initialized OK
[0x0000000076f0e000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[0x0000000076f0e000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[0x0000000076f0e000] [level 3] platform-utils CreateClient(): Create client returned status code is 00000000

Segmentation fault

```

_Originally posted by @hsinghaowang in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/285#issuecomment-567320089_

MushMal, 2019-12-19:
@hsinghaowang the Ubuntu tests: The first two are memory stress tests comparing the internal low-fragmentation heap implementation performance we use for the content store against the process system default heap implementation performance. While this is surprising that on your Ubuntu platform they seem to run slower than the system version. However, you can ignore the randomAllocXXX tests. The miniStreamTest is similar. It's hard to tell what's wrong without the full logs - please include the logs.

The crash however is something different. Can you please clear the directory

git reset --hard origin/master
git clean -dfx
git pull

build it and run this particular test under a debugger to dump out the stack.

./kvspic_test --gtest_filter="ClientApiFunctionalityTest.createKinesisVideoClient_AuthIntegration"

This test is fairly straight forward and I am not sure what's going on.

hsinghaowang, 2019-12-20:
I switch to use the _branched_ version in webrtc SDK.
It is much better but still crashed.

I will continue to integrate SDK into our platform.

```
.
.
.
Setting up test: StreamApiFunctionalityTest

[0x0000000076fde000] [level 3] KinesisVideoClient createKinesisVideoClient(): Creating Kinesis Video Client
[0x0000000076fde000] [level 3] Heap heapInitialize(): Initializing native heap with limit size 10485760, spill ratio 0% and flags 0x00000001
[0x0000000076fde000] [level 3] Heap heapInitialize(): Creating AIV heap.
[0x0000000076fde000] [level 3] Heap heapInitialize(): Heap is initialized OK
[0x0000000076fde000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[0x0000000076fde000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[0x0000000076fde000] [level 3] platform-utils CreateClient(): Create client returned status code is 00000000

[0x0000000076fde000] [level 3] KinesisVideoClient createDeviceResultEvent(): Create device result event.
[0x0000000076fde000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[0x0000000076fde000] [level 3] KinesisVideoClient createKinesisVideoStream(): Creating Kinesis Video Stream.
[0x0000000076fde000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[0x0000000076fde000] [level 3] KinesisVideoClient kinesisVideoStreamFormatChanged(): Stream format changed.
[0x0000000076fde000] [level 3] KinesisVideoClient describeStreamResultEvent(): Describe stream result event.
[0x0000000076fde000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[0x0000000076fde000] [level 3] KinesisVideoClient kinesisVideoStreamFormatChanged(): Stream format changed.
[0x0000000076fde000] [level 3] KinesisVideoClient getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[0x0000000076fde000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[0x0000000076fde000] [level 3] KinesisVideoClient kinesisVideoStreamFormatChanged(): Stream format changed.
[0x0000000076fde000] [level 3] KinesisVideoClient getStreamingTokenResultEvent(): Get streaming token result event.
[0x0000000076fde000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[0x0000000076fde000] [level 3] KinesisVideoClient kinesisVideoStreamFormatChanged(): Stream format changed.
[0x0000000076fde000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
Segmentation fault

```

the complete log is here:
[kvspic_test.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3987565/kvspic_test.txt)

MushMal, 2019-12-20:
@hsinghaowang I am not sure what you mean by "branched" version. The main changes are in the master branch - that's the most stable version.

It seems that you are using KVS WebRTC project and not KVS Producer SDK project. The KVS WebRTC project utilizes KVS Producer SDK for some of the functionality. It includes the PIC which is the reason you have the kvs_pic tests.

Without stack trace under a debugger there is nothing I can do to understand what's crashing in the PIC unit tests.

If I have to guess, your platform might not support unaligned memory access and/or your OS kernel is not configured with an unaligned memory access trap/fixup. 

KVS Producer SDK requires unaligned memory access for now.

WebRTC SDK (the project that you are trying which is not this repository) itself does not require unaligned memory access.

Which assets are you interested in? Are you trying to use the WebRTC Embedded SDK or you are trying to use KVS Producer SDK?

hsinghaowang, 2019-12-23:
There is an "aligned_mem" branch. And amazon-kinesis-video-streams-webrtc-sdk-c depend on it. 
Am I right?

We are trying to use WebRTC Embedded SDK.

zhiyua-git, 2019-12-23:
Hi @hsinghaowang ,

We worked on getting aligned memory access on KVS WebRTC. You can start building KVS WebRTC on your device if that is what you are going to use. But KVS C/C++ producer SDK requires more work to support aligned memory access, so it is not ready on KVS C producer SDK yet. The failure you got in PIC unit tests is not on the components KVS WebRTC depending on. 

We will notify you when KVS C producer SDK also support aligned memory access if you are interested. Thanks.

MushMal, 2019-12-23:
@hsinghaowang as @zhiyua-git mentioned, currently, we only support WebRTC C SDK in an aligned mode. WebRTC SDK utilizes a aligned_mem branch of the PIC and Common libraries from the KVS Producer SDK but the KVS Producer SDK overall doesn't support aligned only access.

You are attempting to execute common tests in the PIC which would fail. WebRTC doesn't link against those so WebRTC doesn't use them.

In order to test it, try to run the WebRTC tests or simply run the sample applications

hsinghaowang, 2019-12-25:
I will close this issue and open another issue in WebRTC SDK.

closed 2019-12-25 by hsinghaowang