awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #311: Pre-built binaries

AVancans, 2019-12-21:
I am still not completely sure why Kinesis Video Streams doesn't come with pre-built samples? It seems overly troublesome to set up the whole build environment just to run a simple RTSP forwarding app. 
Why not simply release the binaries for each architecture for every public release?

zhiyua-git, 2019-12-21:
Hi @AVancans , now we provide prebuilt docker image for the use case when you just want to do RTSP. You can refer to https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-docker for the steps. Hope it helps.

MushMal, 2020-03-17:
There are multiple variables when producing a binary for a low-level system code that's portable. I am resolving this as won't fix for now

closed 2020-03-17 by MushMal