awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #312: FragmentLengthInMilliseconds becomes 0 in listfragments.

HigashidaTomoyuki, 2019-12-23:
I am doing putFrame with the following parameters. However, FragmentLengthInMilliseconds is all 0 in listFragments.
How do I get the correct values?

`StreamDefinition` (Partially omitted)
```
content_type="audio/flac"
fragment_duration=4000
timecode_scale=1
codec_id="A_FLAC"
track_type=MKV_TRACK_INFO_TYPE_AUDIO
default_track_id=1
```

`Frame` (index is not set)
```
duration = 4
flags = FRAME_FLAG_KEY_FRAME
size = buffer_size
frameData = buffer_data
trackId = 1
decodingTs = timestamp
presentationTs = timestamp
```

MushMal, 2019-12-23:
@HigashidaTomoyuki the C++ putFrame API is using Frame object from C SDK which uses 100ns time units. So, if you want to specify 4ms you should specify 4 * HUNDREDS_OF_NANOS_IN_A_MILLISECOND. Also, could you make sure that listFragments API actually returns the data? Depending whether the time range contains valid fragments it will return those or an empty set.

HigashidaTomoyuki, 2019-12-24:
Thank you @MushMal 
I want to set the duration to 4 seconds, so I set it below.
```
duration = static_cast<UINT64>(4000 * duration * HUNDREDS_OF_NANOS_IN_A_MILLISECOND);
```

As shown below, FragmentLengthInMilliseconds is all 0.
```
$ aws kinesis-video-archived-media list-fragments --stream-name "stream_name" --endpoint-url "https://hogehoge.kinesisvideo.ap-northeast-1.amazonaws.com" --fragment-selector FragmentSelectorType="PRODUCER_TIMESTAMP",Time
stampRange=\{StartTimestamp=2019-12-24T00:00:00.000Z,EndTimestamp=2019-12-25T00:00:00.000Z\} --debug
2019-12-24 12:10:28,554 - MainThread - awscli.clidriver - DEBUG - CLI version: aws-cli/1.16.307 Python/3.8.0 Windows/10 botocore/1.13.43
2019-12-24 12:10:28,554 - MainThread - awscli.clidriver - DEBUG - Arguments entered to CLI: ['kinesis-video-archived-media', 'list-fragments', '--stream-name', 'stream_name', '--endpoint-url', 'https://hogehoge.kinesisvideo.ap-northeast-1.amazonaws.com', '--fragment-selector', 'FragmentSelectorType=PRODUCER_TIMESTAMP,TimestampRange={StartTimestamp=2019-12-24T00:00:00.000Z,EndTimestamp=2019-12-25T00:00:00.000Z}', '--debug']
2019-12-24 12:10:28,555 - MainThread - botocore.hooks - DEBUG - Event session-initialized: calling handler <function add_scalar_parsers at 0x03DFD190>
2019-12-24 12:10:28,556 - MainThread - botocore.hooks - DEBUG - Event session-initialized: calling handler <function register_uri_param_handler at 0x039FC9B8>
2019-12-24 12:10:28,557 - MainThread - botocore.hooks - DEBUG - Event session-initialized: calling handler <function inject_assume_role_provider_cache at 0x03A3BF58>
2019-12-24 12:10:28,559 - MainThread - botocore.hooks - DEBUG - Event session-initialized: calling handler <function attach_history_handler at 0x03C17418>
2019-12-24 12:10:28,561 - MainThread - botocore.loaders - DEBUG - Loading JSON file: C:\users\user_name\appdata\local\programs\python\python38-32\lib\site-packages\botocore\data\kinesis-video-archived-media\2017-09-30\service-2.json
2019-12-24 12:10:28,562 - MainThread - botocore.hooks - DEBUG - Event building-command-table.kinesis-video-archived-media: calling handler <function add_waiters at 0x03DFDDF0>
2019-12-24 12:10:28,604 - MainThread - awscli.clidriver - DEBUG - OrderedDict([('stream-name', <awscli.arguments.CLIArgument object at 0x03E6A790>), ('max-results', <awscli.arguments.CLIArgument object at 0x03E6A7D8>), ('next-token', <awscli.arguments.CLIArgument object at 0x03E6ACA0>), ('fragment-selector', <awscli.arguments.CLIArgument object at 0x03E6A910>)])
2019-12-24 12:10:28,605 - MainThread - botocore.hooks - DEBUG - Event building-argument-table.kinesis-video-archived-media.list-fragments: calling handler <function add_streaming_output_arg at 0x03DFD2F8>
2019-12-24 12:10:28,606 - MainThread - botocore.hooks - DEBUG - Event building-argument-table.kinesis-video-archived-media.list-fragments: calling handler <function add_cli_input_json at 0x03A43610>
2019-12-24 12:10:28,606 - MainThread - botocore.hooks - DEBUG - Event building-argument-table.kinesis-video-archived-media.list-fragments: calling handler <function unify_paging_params at 0x03DAA808>
2019-12-24 12:10:28,651 - MainThread - botocore.loaders - DEBUG - Loading JSON file: C:\users\user_name\appdata\local\programs\python\python38-32\lib\site-packages\botocore\data\kinesis-video-archived-media\2017-09-30\paginators-1.json
2019-12-24 12:10:28,652 - MainThread - awscli.customizations.paginate - DEBUG - Modifying paging parameters for operation: ListFragments
2019-12-24 12:10:28,653 - MainThread - botocore.hooks - DEBUG - Event building-argument-table.kinesis-video-archived-media.list-fragments: calling handler <function add_generate_skeleton at 0x03D95A00>
2019-12-24 12:10:28,653 - MainThread - botocore.hooks - DEBUG - Event before-building-argument-table-parser.kinesis-video-archived-media.list-fragments: calling handler <bound method OverrideRequiredArgsArgument.override_required_args of <awscli.customizations.cliinputjson.CliInputJSONArgument object at 0x03E6AE80>>
2019-12-24 12:10:28,654 - MainThread - botocore.hooks - DEBUG - Event before-building-argument-table-parser.kinesis-video-archived-media.list-fragments: calling handler <bound method GenerateCliSkeletonArgument.override_required_args of <awscli.customizations.generatecliskeleton.GenerateCliSkeletonArgument object at 0x03E7B0B8>>
2019-12-24 12:10:28,657 - MainThread - botocore.hooks - DEBUG - Event operation-args-parsed.kinesis-video-archived-media.list-fragments: calling handler functools.partial(<function check_should_enable_pagination at 0x03DAA898>, ['next-token', 'max-results'], {}, OrderedDict([('stream-name', <awscli.arguments.CLIArgument object at 0x03E6A790>), ('max-results', <awscli.arguments.CLIArgument object at 0x03E6A7D8>), ('next-token', <awscli.arguments.CLIArgument object at 0x03E6ACA0>), ('fragment-selector', <awscli.arguments.CLIArgument object at 0x03E6A910>), ('cli-input-json', <awscli.customizations.cliinputjson.CliInputJSONArgument object at 0x03E6AE80>), ('starting-token', <awscli.customizations.paginate.PageArgument object at 0x03E6AB80>), ('page-size', <awscli.customizations.paginate.PageArgument object at 0x03E6ABF8>), ('max-items', <awscli.customizations.paginate.PageArgument object at 0x03E6AC28>), ('generate-cli-skeleton', <awscli.customizations.generatecliskeleton.GenerateCliSkeletonArgument object at 0x03E7B0B8>)]))
[... cut ...]
            "FragmentNumber": "91343852333181432447151424353830790231443122322",
            "FragmentSizeInBytes": 21546,
            "ProducerTimestamp": 1577156961.111,
            "ServerTimestamp": 1577156962.11,
            "FragmentLengthInMilliseconds": 0
        },
        {
            "FragmentNumber": "91343852333181432417440863410981655075981163572",
            "FragmentSizeInBytes": 64324,
            "ProducerTimestamp": 1577156929.496,
            "ServerTimestamp": 1577156930.23,
            "FragmentLengthInMilliseconds": 0
        }
    ]
}
```

zhiyua-git, 2020-01-10:
Hi @HigashidaTomoyuki ,

Can you provide how are you setting the duration? Is it by frame.duration = static_cast<UINT64>(4000 * HUNDREDS_OF_NANOS_IN_A_MILLISECOND); It is cause drop frame due to frame timestamp + duration is more than next frame timestamp, are you seeing [0x30000005](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html) in your log? Thanks.

HigashidaTomoyuki, 2020-01-21:
The code which rewritten Gstreamer dependent part from the sample of the following URL is used.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_audio_video_sample_app.cpp

`create_kinesis_video_frame` and `kinesis_video_stream_init` of this sample are changed as follows.

```cpp
void create_kinesis_video_frame(Frame *frame, void *data, size_t len, float buffer_seconds, float fragment_start_timestamp) {
    UINT64 duration = static_cast<UINT64>(buffer_seconds * HUNDREDS_OF_NANOS_IN_A_SECOND);
    UINT64 start_timestamp = static_cast<UINT64>(fragment_start_timestamp * HUNDREDS_OF_NANOS_IN_A_SECOND);
    UINT64 end_timestamp = start_timestamp + duration;
    frame->duration = duration;
    frame->flags = FRAME_FLAG_KEY_FRAME;
    frame->size = static_cast<UINT32>(len);
    frame->frameData = reinterpret_cast<PBYTE>(data);
    frame->trackId = DEFAULT_AUDIO_TRACKID;
    frame->presentationTs = start_timestamp;
    frame->decodingTs = end_timestamp;
}
```
```cpp
void kinesis_video_stream_init(CustomData *data) {
    unique_ptr<StreamDefinition> stream_definition(new StreamDefinition(
        data->stream_name,
        hours(2),
        nullptr,
        "",
        STREAMING_TYPE_REALTIME,
        "audio/flac",
        duration_cast<milliseconds> (seconds(150)),
        milliseconds(4000),
        milliseconds(1),
        TRUE,
        TRUE,
        FALSE,
        TRUE,
        TRUE,
        TRUE,
        NAL_ADAPTATION_FLAG_NONE,
        100,
        (4 * 1024 * 1024),
        seconds(300),
        seconds(40),
        seconds(60),
        A_FLAC,
        "audio",
        nullptr,
        0,
        MKV_TRACK_INFO_TYPE_AUDIO,
        vector<uint8_t>(),
        1));
    data->kinesis_video_stream = data->kinesis_video_producer->createStreamSync(move(stream_definition));
}
```

zhiyua-git, 2020-01-22:
Hi @HigashidaTomoyuki ,

Can you add **export DEBUG_DUMP_FRAME_INFO=TRUE** and rerun the app and send your log to kinesis-video-support@amazon.com? With the log we will be able to tell why this is giving you 0 as fragment duration. Thanks.

HigashidaTomoyuki, 2020-01-22:
Thank you @zhiyua-git 

Sorry, information was lacking.
I have already contacted support.
The answer from support is as follows. I'm sorry if it's hard to read because I'm translating from Japanese to English.

> Investigating the behavior of FragmentLengthInMilliseconds for Kinesis Video Streams in our test environment, we confirmed that the value matches the timestamp difference of the frames included in the fragment, not the Duration value of create_kinesis_video_frame.

> Investigating past cases, etc., confirmed that when Fragment contains only one Frame, FragmentLengthInMilliseconds becomes 0.

Examining the fragment obtained by get-media-for-fragment-list revealed that only one frame was included.

How do I add multiple frames to a fragment without using Gstreamer?

zhiyua-git, 2020-01-22:
Hi @HigashidaTomoyuki ,

From your code, I can see you are using key-frame-fragmentation: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/0888c666efbe7f20891a86c9e64a677f73e33b17/kinesis-video-producer/src/StreamDefinition.h#L57 and you set all the frames as key-frame (frame->flags = FRAME_FLAG_KEY_FRAME;). 

In key-frame-fragmentation mode, every key frame will start a new fragment, so in your case, you saw one frame per fragment. You can set key frame for some numbers of frames and others as non-key frame.

Alternatively, you can set frames as key frame but don't use key-frame-fragmentation mode (set to FALSE in StreamDefinition), then producer SDK will form fragment based on given fragment duration in StreamDefinition. 

Also, the frame duration is used to avoid overlap, but it is easy to get really dense output for audio frames, I would recommend using frame duration as 0 for every frame to avoid producer SDK dropping frames due to timestamp overlap. The fragment duration would still be correct while frame duration set to 0 if there are multiple frames inside the fragment and having correct timestamp for those frames.

closed 2020-03-17 by MushMal