awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #313: Documentation and results are very different in several places, maybe outdated?

jeremysmitherman, 2019-12-24:
On Ubuntu 18, I was following the instructions led me to https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html#examples-gstreamer-plugin-download and when the gstreamer-plugin-install-script complains about needing the SDK installed that finally led me to https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-cpp-test.html and following the instructions from both the github page linux instructions and the docs in the link above have led to inconsistencies.  

For one, the install-script does not generate a "start" binary as purported, after successfully installing the SDK, the gstreamer-plugin-install-script fails with "fatal error: gstkvssink.h: No such file or directory", and it's unclear where the resulting libgstkvsink.so goes after installation.  Any guidance would be appreciated.

pratamatama, 2020-01-02:
It happens to me as well on MacOS Catalina 10.15.2

disa6302, 2020-01-08:
Hello,

The steps to install the SDK is currently up to date in the README.md of Github repo. Follow the steps in the repo to install the SDK. There are 2 ways to go about installing the SDK - using the install-script and using the min-install script. min-install-script takes lesser time to install the SDK and gets you to running the applications faster, and install-script takes more time since it installs gstreamer from the source. 

My suggestion would be to use the min-install-script to install the SDK. Let me know if have any questions.

parampopat, 2020-04-02:
The min-install-script fails in Windows.

disa6302, 2020-04-02:
@parampopat , 

Can you open a new issue detailing the failure and adding the logs? Thanks!

disa6302, 2020-04-30:
Closing this because documentation related item is tracked #423

closed 2020-04-30 by disa6302