awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #314: Problem with playback in AWS when trying to integrate cproducer into MIPS platform

henryChmn, 2019-12-25:
[image]

attached our video frame log files as the attached file.
[video_sample.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3999437/video_sample.zip)

We need your help for it,
Thanks

MushMal, 2019-12-27:
@henryChmn can you please provide more info what you are trying to do? Are you using the video only sample or doing something of your own? Are you specifying the codec private data? The frames you've supplied look like in Annex-B format and they are h264 from what it looks like. The Producer SDK would automatically set the codec private data from the frames if the NAL adaptation flags are set to adapt the CPD. Can you please also attach the logs?

henryChmn, 2019-12-27:
We fixed this problem in our program

closed 2019-12-27 by henryChmn