awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #317: how can i restart kinesis video stream

hameedriswan7, 2020-01-02:
i am using Putmedia to run RTSP camera in AWS Kinesis Video Stream. i need feature for restart stream Every One Hour duration.

chehefen, 2020-01-03:
Can you give us some more detail on how you are using putMedia to run RTSP camera to kvs? Thanks

hameedriswan7, 2020-01-07:
@chehefen  For RTSP camera you need Gstreamer Plugin for run RTSP camera  in PUTMEDIA.

Refer this code:
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html.

so that you need to install gstreamer  plugin from https://gstreamer.freedesktop.org/download/
it contains two files. download and install in. 

then integrate gestremer class with media source with these paramters 
eg:
final MediaSource mediaSource = new GStreamerRtspMediaSource(STREAM_NAME, new URI(RTSP_URI), RETENTION_PERIOD,
				new Tag[] { new Tag("Produced-By", "Kinesis-Video-GStreamer-Demo") });

chehefen, 2020-03-16:
To restart the stream, you just need to restart the gstreamer pipeline. You can checkout their documentation here: https://gstreamer.freedesktop.org/documentation/tutorials/index.html?gi-language=c

MushMal, 2020-03-23:
Resolving. Please re-open or create a separate issue if need any further info

closed 2020-03-23 by MushMal