awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #318: Video Stream RTSP stopping AWS Kinesis "fragment_timecode_lesser_than_previous"

Fernando-Oliveira86, 2020-01-03:
@Fernando-Oliveira86 can you open a new issue going over what your video source is and how are you streaming to kvs? Thanks.

_Originally posted by @chehefen in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/138#issuecomment-570307172_

Hello,
I'm using the pipeline "gst-launch-1.0 rtspsrc location=URL short-header=TRUE ! rtph264depay ! video/x-h264,stream-format=byte-stream ! h264parse ! kvssink stream-name="Name" storage-size=512" to send a stream to kinesis.
The problem is that randomly after a while the transmission is cut off due to a "pause" that the rtsp protocol has, which makes the algorithm understand that there is no video being sent as the timestamp is very close. Returning: 

"ErrorCode": "FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS", "ErrorId": 4004
[DEBUG] [03-01-2020 12:59:11:730.835 E. South America Standard Time] defaultStreamErrorReportCallback(): Reported streamError callback for stream handle 1736953. Upload handle 0. Fragment timecode in 100ns: 217980000. Error status: 0x52000060.
I tried to change the code "kinesis_video_gstreamer_sample_app" mentioned in the previous issue but without sucess.
Any solution? thanks

Fernando-Oliveira86, 2020-01-06:
@chehefen , Some ideia for this?

zhiyua-git, 2020-01-10:
Hi @Fernando-Oliveira86 ,

Can you add environment variable **export DEBUG_DUMP_FRAME_INFO=TRUE** and send the log to kinesis-video-support@amazon.com? We can check the log to see if it is an timestamp issue we are trying to fix now for the RTSP streams.

Thanks,
-ZH

zhiyua-git, 2020-01-17:
Hi @Fernando-Oliveira86 ,

This [branch](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/rtsp-timestamp-fixup) with a timecode fix-up should help you resolve the issue. Can you check it out and have a try? 

We are also working on a long-term fix in backend service, so in the future just using master branch should not hit this issue. Thanks.

MushMal, 2020-03-17:
Resolving. Please try the new master branch

closed 2020-03-17 by MushMal