awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #319: kvsVideoOnlyStreamingSample run on embedded device error:wrong signature length

guohuhu, 2020-01-05:
embeded device is Ingnic T31Z with Linux version 3.10.14 +BusyBox v1.22.1. I cross comple amazon-kinesis-video-streams-producer-sdk-cpp use mips-linux-uclibc-gnu-gcc. Run kvsVideoOnlyStreamingSample on this embeded device got this error:
2020-01-05 15:55:34 DEBUG   curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result SSL connect error: error:04091077:rsa routines:int_rsa_verify:wrong signature length
2020-01-05 15:55:34 DEBUG   curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIA3RQEAED63ZWHDCFU/20200105/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=acd7e0fd729db6e77f3355008677c31fbe94d731affa3d9c24cdd124920c049d
    content-length: 28
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/2.1.0 GCC/4.7.2 Linux/3.10.14-Archon mips
    X-Amz-Date: 20200105T155533Z

I compile  amazon-kinesis-video-streams-producer-sdk-cpp on my x86 ubuntu pc. Run kvsVideoOnlyStreamingSample is ok.
The same sample why got the error on my embedded device?

closed 2020-01-08 by guohuhu

Remainfox, 2020-02-22:
I meet this problem also in mips T31X Arch Embeded device

MushMal, 2020-02-22:
Interesting, this somehow fell off our radar. @guohuhu could you please suggest what was the root cause in your scenario? Do we have a mix of libCurl <-> OpenSSL <-> libCrypto that get cross-compiled wrong? This issue seems to come from the libCrypto itself and I am not yet sure what might cause it.