awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #321: Build does not yield installable headers or linkable library

snarkeystarkey, 2020-01-07:
Ubuntu 18.04 / GCC 7.4 / Cmake 3.15.0

Perhaps I'm missing something, but I'm unable to find any installable headers or linkable modules from this SDK.

Is this purely a referential implementation? How does one integrate this SDK into an existing project?

zhiyua-git, 2020-01-08:
Hi @snarkeystarkey ,

After running min-install-script/install-script successfully, you should be able to find dynamic libraries (libcproducer.so, libproducer.so, libgstkvssink.so). You can use those depending on what function you need.

snarkeystarkey, 2020-01-08:
I have found those libraries, yes - however I'm unable to locate the corresponding headers. Attempting to include headers in my project yields a lot of errors and path issues. I'm able to make this work if I combine most of the "com" includes in my system include path, as well as pull a few from various locations in this project. Is there something I'm not understanding regarding the headers?

zhiyua-git, 2020-01-08:
Hi @snarkeystarkey ,

We will try to improve the install path so header files and libraries would be easier to use. For now you need to find the header files. What you were doing is correct right now.

If you are using libcproducer then you will just need header files from https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/kinesis-video-c-producer/src/include

If you are using libproducer then you will need header files from https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/kinesis-video-producer/src

If you are using libgstkvssink then you will need header files from https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/kinesis-video-gstreamer-plugin/src

snarkeystarkey, 2020-01-08:
OK! Now we're talking --- THANK YOU!

disa6302, 2020-04-30:
Closing this issue.

closed 2020-04-30 by disa6302