awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #323: (WARNING: erroneous pipeline: no element "kvssink") ubuntu 18

Mano009das, 2020-01-08:
i am having this issue i have tired solutions to solve it with no success. is it a laptop driver issue?

zhiyua-git, 2020-01-08:
Hi @Mano009das ,

Can you try setting environment variables as below?
export GST_PLUGIN_PATH=/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
export LD_LIBRARY_PATH=/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build:$LD_LIBRARY_PATH

Mano009das, 2020-01-09:
thank you for the response .

yes i just did then shall i run  this code? after i did this it gave me the same error. or do i have to install something else.

Ubuntu:

gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,width=640,height=480,framerate=30/1,format=I420 ! x264enc  bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name="my-stream-name" access-key="YourAccessKey" secret-key="YourSecretKey" alsasrc device=hw:1,0 ! audioconvert ! avenc_aac ! queue ! sink.

zhiyua-git, 2020-01-09:
Hi @Mano009das ,

Can you provide the output of “ls /path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build“? Want to check if you have all the files built. Thanks.

Mano009das, 2020-01-09:
ok sure give me a min

Mano009das, 2020-01-09:
~# ls /path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
ls: cannot access '/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build': No such file or directory

Mano009das, 2020-01-09:
may be this is the issue???

zhiyua-git, 2020-01-09:
Hmm, you need to put your actual directory path in the environment variable, say if your file is in /home/username/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build then 

export GST_PLUGIN_PATH= /home/username/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
export LD_LIBRARY_PATH= /home/username/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build:$LD_LIBRARY_PATH

Mano009das, 2020-01-09:
[image]

Mano009das, 2020-01-09:
okay thank let me try and get back to u

Mano009das, 2020-01-09:
[image]

Mano009das, 2020-01-09:
i am getting this output :(

zhiyua-git, 2020-01-09:
As you are already in kinesis-video-native-build directory, you can do it through
export GST_PLUGIN_PATH=$PWD
export LD_LIBRARY_PATH= $PWD:$LD_LIBRARY_PATH

The path "/root/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build" you provided is wrong.

Mano009das, 2020-01-27:
thank you for the solution it was a hardware compatibility issue I tired in raspberry pi and it worked fine :)

closed 2020-01-27 by Mano009das