awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #326: High fps reported by kvssink

js1972, 2020-01-12:
I find that the FPS logged to the console when running a simple gstreamer pipeline and kvssink is always much higher than the actual rtsp sources fps.

Using this pipeline as an example:
```
```
The reported fps value range from around 21 to 90. The actual Hikvision source camera is set to 640x360 and 10fps.
Sometimes I see kvssink report values as high as 180.

The fps value tend to trend lower over time.... I've been running this pipeline just now for about 10 minutess and I'm starting to see the fps reported by kvssink come down to 12 - 14 which is much closer at least.

Why is this and is it a problem.

MushMal, 2020-01-20:
@js1972 the SDK has collects some metrics internally as the streaming progresses - it doesn't have internal threading so the frame rate information is generated when a higher-level logic calls put frame API. The PIC uses EMA (exponential mean averaging) to add some "inertia" to the rolling aggregates. The C producer layer calls public PIC API to extract the metrics and simply prints them with every key frame after certain time elapses. In this case, the GStreamer pipeline uses RTSP which is "bursty" - aka, it produces frames not at a constant rate as the encoder produces but at the rate the RTSP plugin receives them over the network which produces bursts of frames, resulting in the observed frame rate to fluctuate. This is similar to an offline intermittent streaming mode when a number of frames are being produces in bursts resulting in high FPS. The reality, though, the reported FPS is the observed FPS - each frame should have its own timestamp so it shouldn't cause problems with the persistence or indexing. 

Please let us know if you have more questions. Resolving for now.

closed 2020-01-20 by MushMal

js1972, 2020-01-20:
Thanks for the explanation.