awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #327: Disable retry mechanism

indoorrobot, 2020-01-14:
Hi @zhiyua-git , I think this issue deserves a separate thread :)

I would like to make sure that whenever a stream fails to stream, for example connection loss or INVALID_MKV_DATA, the gst process that runs it will stop.
I have followed the steps suggested in [this issue](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/315), but it seems that it does not work as expected and the gst process keeps running.

I guess it would be great if the retry time will be configurable or at least the amount of retries, since I would like to lose a minimal amount of data.

In addition, is there a buffer or some backup stream for data that was gathered but not streamed due to an error? if so, what is the size of it? can it be adjusted?

the way I am running the stream is:
`gst-launch-1.0 -v v4l2src device=/dev/video2 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1920,height=1080,framerate=30/1 ! kvssink name=sink stream-name=${THING_ID}_rc access-key=$AWS_ACCESS_KEY_ID secret-key=$AWS_SECRET_ACCESS_KEY max-latency=10 restart-on-error=FALSE
`

Thanks alot

MushMal, 2020-01-15:
@indoorrobot you can configure the buffer duration specifying buffer-duration parameter https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp#L527

you might also need to increase the storage size appropriately https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp#L567

If you want to dive a little deeper, you can override the default behavior (which is nothing) of the buffer pressure (gets called after the buffer accumulates https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp#L477 duration) and do something that's specific to your logic.

We are constantly analyzing the usage scenarios and we are looking for ways to make things easy for customers. What would you consider your desired behavior should be when the networking bandwidth doesn't allow for streaming? There are multiple cases - no network at all for example or limited bandwidth, temporary or constant. Some customers do react on latency pressure (setting it to be 80% of the buffer duration for example) and re-configure the upstream to produce less dense stream by "thinning out" the frames or lowering resolution (for H264 this will require re-streaming).

indoorrobot, 2020-01-15:
@MushMal thanks for assisting with this.
I have 2 h264 cameras which I am trying to stream to KVS using gst.
`gst-launch-1.0 -v v4l2src device=/dev/video2 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1920,height=1080,framerate=30/1 ! kvssink name=sink stream-name=${ROBOT_CLOUD_ID}_rc access-key=$AWS_ACCESS_KEY_ID secret-key=$AWS_SECRET_ACCESS_KEY max-latency=10`

The problem is that I encounter the INVALID MKV error due to an unclear reason, and the stream won't auto recover....
So in order to rerun it manually I would like the gst process to fail so I will monitor the process and rerun it by myself when fails.

I have made the changes suggested in the previous thread but it still won't make the process to exit.

What I am trying to achieve is:
The constant stream of 2 H.264 cameras in 1080P and making sure I have a constant stream so that I will have a minimal data loss.

MushMal, 2020-01-17:
@indoorrobot you seem to configure the max-latency to 10 which might be too aggressive and kicks off the default auto-recovery early. Could you please provide the logs to see whether it's the network bandwidth or anything else - like invalid auth that does not allow the producer to continue.

kvs sink has some configuration you can specify through parameters but it uses the default policy as it can't readily interact with the application. We are planning on emitting GST messages matching the producer SDK callbacks that the application can use to act. For example, the same buffer pressure callback can be issued as a GST event that will allow upstream elements to "thin out" the frames or to stop/restart the pipeline.

indoorrobot, 2020-01-18:
What would be a reasonable max-latency value?

MushMal, 2020-01-20:
Apologies for the delay. It truly depends on the application. The main issue is to try to react on the network conditions before the pressure forces the SDK to drop the frames. 10 seconds might be too aggressive for the default behavior which is in GStreamer plugin. If the application is using the SDK directly, configuring max latency to 10 seconds might be OK and overriding the default behavior by the custom logic. For example, resetting the connection, thinning out the bitrate, re-streaming or simply pausing the media pipeline until the bandwidth is enough. 

If, however, the application can tolerate more latency, then it's useful to specify the buffer duration (max temporal content duration) - default is 120 seconds and set max latency to something like 40 - 90 seconds (depending on the application) - making sure there is enough storage available on the device which we have a general formula for: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-reference-structures-producer.html

Size = NumberOfStreams * AverageFrameSize * FramesPerSecond * BufferDurationInSeconds * DefragmentationFactor

GStreamer plugin is a powerful tool but not as flexible as using the SDK directly. We are planning on introducing different retry "policies" in the near future that can be selected in the GStreamer plugin.

indoorrobot, 2020-01-20:
Thank you very much for the detailed answer!!

Will follow, and will be waiting for an update regarding the retry policy!

MushMal, 2020-01-20:
Absolutely! Meanwhile, could you please let us know what the desired behavior you expect for your application:

* How long would you need the buffer duration to be?
* How long would you need the max latency to be (the time at which the latency pressure is fired and the SDK will start the policy)
* What the correct remedy you think should be - thinning out the frames (dropping P- and B- frames or dumping entire fragments with Idr frames)? 
* Any other information that would be useful for us to take into account

MushMal, 2020-03-17:
@indoorrobot Is there any further update on this issue?

MushMal, 2020-03-21:
Resolving as this issue seems to be stale. Please re-open if needed

closed 2020-03-21 by MushMal