awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #329: Trying: ./min-install-script; Error: make: *** [Makefile:10: jsmn.o] Error 127

AakashBasu, 2020-01-23:
I'm trying to install the Kinesis Producer SDK, but facing the following error:

Machine: Windows 10 Pro, 64 bits.

Link of steps followed: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-win.html#producer-sdk-win-mingw

Note: I installed it before using minGW which completed successfully after multiple error outs, but I was getting errors while trying to start GStreamer, this command: "./kinesis_video_gstreamer_sample_app.exe my-stream-name"

Then I connected with AWS and they suggested me to do a fresh install by running migGW as Admin, so I cleared everything, ran git clone and then tried running minGW 64 as Admin and then run the below, but it isn't working.

Error:
a.b@DT-0206-20KVTW2 MSYS /c/msys64/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
$ ./min-install-script
Setting up for KVS Producer SDK installation
Required libraries for MSYS_NT-10.0-18362 unknown
jsmn lib not found. Installing
jsmn-1.0.0/
jsmn-1.0.0/.travis.yml
jsmn-1.0.0/LICENSE
jsmn-1.0.0/Makefile
jsmn-1.0.0/README.md
jsmn-1.0.0/example/
jsmn-1.0.0/example/jsondump.c
jsmn-1.0.0/example/simple.c
jsmn-1.0.0/jsmn.c
jsmn-1.0.0/jsmn.h
jsmn-1.0.0/library.json
jsmn-1.0.0/test/
jsmn-1.0.0/test/test.h
jsmn-1.0.0/test/tests.c
jsmn-1.0.0/test/testutil.h
rm -f *.o example/*.o
rm -f *.a *.so
rm -f simple_example
rm -f jsondump
cc -c -fPIC jsmn.c -o jsmn.o
make: cc: Command not found
make: *** [Makefile:10: jsmn.o] Error 127

Any help?

zhiyua-git, 2020-01-23:
Hi @AakashBasu ,

According to your log, you are not running the min-install-script on mingw64/mingw32 shell.

Please follow the steps in the doc: Launch the MinGW shell (mingw64.exe) from the C:\msys32 or C:\msys64 directory. Make sure that you are opening the mingw64.exe or mingw32.exe based on your platform, and not the MSYS2 application. The MSYS2 application is the default application that is opened after you finish installing MSYS2.

Thanks.

AakashBasu, 2020-01-24:
Thanks, closing the issue.

closed 2020-01-24 by AakashBasu