awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #330: 2 cams on same device - not time synced

indoorrobot, 2020-01-26:
Hi,

I have 2 H264 UVC cams, connected to same device (Intel UpBoard) streaming simultaneously to KVS.
`gst-launch-1.0 -v v4l2src device=/dev/video1 do-timestamp=true ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1280,height=720,framerate=15/1 ! kvssink name=sink stream-name=${ROBOT_CLOUD_ID}_rc access-key=$AWS_ACCESS_KEY_ID secret-key=$AWS_SECRET_ACCESS_KEY`

When calling getMedia on the 2 streams (and in the KVS console), i see the streams are not from the same time (there might be a difference of more than 10 seconds sometimes).

In addition when watching the stream in KVS console (using SERVER TIMESTAMP), I sometimes see that the time "hops" every few seconds in intervals of 2 to 7 seconds, even though the stream itself is fine. moreover, when switching to producer timestamp (the time on the device when streaming) there is nothing there or some weirdly timed video...

I guess it means I do not send the time correctly, how can I make sure i do?

disa6302, 2020-01-26:
Hello @indoorrobot , 

I will have someone look at this for you. Thank you for your patience.

zhiyua-git, 2020-01-26:
Hi @indoorrobot ,

It is not clear why this time sync issue happen, it might be caused by the media pipeline not producing frames at same speed. But the weird timed video in producer timestamp mode should not happen at all. We can debug both by enabling absolute timestamp, can you do these and share the log with us?

1. Make below code change to [gstkvssink.cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp).

```
diff --git a/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp b/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp
index 046f5fa..3f2b095 100644
--- a/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp
+++ b/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp
@@ -1080,6 +1080,10 @@ gst_kvs_sink_handle_buffer (GstCollectPads * pads,
         buf->dts = data->last_dts + DEFAULT_FRAME_DURATION_MS * HUNDREDS_OF_NANOS_IN_A_MILLISECOND * DEFAULT_TIME_UNIT_IN_NANOS;
     }
 
+    if (!IS_OFFLINE_STREAMING_MODE(kvssink->streaming_type) && data->pts_base == 0) {
+        data->pts_base = buf->pts;
+    }
+
     data->last_dts = buf->dts;
     track_id = kvs_sink_track_data->track_id;
 
@@ -1108,7 +1112,8 @@ gst_kvs_sink_handle_buffer (GstCollectPads * pads,
     }
 
     put_frame(data->kinesis_video_stream, info.data, info.size,
-              std::chrono::nanoseconds(buf->pts),
+              IS_OFFLINE_STREAMING_MODE(kvssink->streaming_type) ? std::chrono::nanoseconds(buf->pts)
+              : std::chrono::nanoseconds(buf->pts - data->pts_base + chrono::duration_cast<microseconds>(systemCurrentTime().time_since_epoch()).count()),
               std::chrono::nanoseconds(buf->dts), kinesis_video_flags, track_id, data->frame_count);
     data->frame_count++;
```
2. Enable dumping absolute timestamp by running **DEBUG_DUMP_FRAME_INFO=1** gst-launch-1.0 -v v4l2src device=/dev/video1 do-timestamp=true ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1280,height=720,framerate=15/1 ! kvssink name=sink stream-name=${ROBOT_CLOUD_ID}_rc access-key=$AWS_ACCESS_KEY_ID secret-key=$AWS_SECRET_ACCESS_KEY **absolute-fragment-times=true**

Thanks.

MushMal, 2020-01-27:
Could you also please ensure you are not streaming into the same stream causing ingestion to "splice" the fragments into the same timeline. I doubt it though as the backend would result in an error in case the timestamp drifts back.

Also, could you please send the logs to kinesis-video-support@amazon.com

It is possible that the producer-side streaming results in errors - either in KVS sink or prior to that in the encoder, causing re-streaming/odd timestamps.

indoorrobot, 2020-01-27:
Hi,

I am not streaming to the same stream.
I moved back to release 2.0.2 and it looks like it is working fine now.

disa6302, 2020-01-27:
@indoorrobot , glad to know it is working for you now with a previous release. 
However, we would still like to understand what is going on in the current release version. It would be really helpful if you could send the logs to kinesis-video-support@amazon.com before and after making the code change suggested by @zhiyua-git. It will help in figuring out what is affecting the current version.

chehefen, 2020-02-06:
Hi @indoorrobot, can you run `gst-launch-1.0 -v v4l2src device=/dev/video1 ! fakesink silent=FALSE` for a minute and send us the log? Note that `do-timestamp=TRUE` is not included because we wanna use timestamp produced directly by the encoder on the device. The log should give us some idea how your frame timestamps look like. Also note that when viewing the stream in AWS console, it's always preferable to use PRODUCER_TIMESTAMP mode because it ensures stream continuity despite any network fluctuations.

indoorrobot, 2020-02-07:
see attached.
Using release 2.0.2

chehefen, 2020-02-07:
@indoorrobot I can't see the file. you can also send it to kinesis-video-support@amazon.com. Thanks.

MushMal, 2020-03-18:
Is there any update on this matter?

MushMal, 2020-03-21:
Resolving this. Please re-open or cut another issue if need further help

closed 2020-03-21 by MushMal