awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #331: Can you install the Amazon kinesis Video Streams Producer SDK directly in IP Camera?

Moriuks, 2020-01-28:
**this is my current set up.**
[image]

i have the SDK install on a Ubuntu t2.Medium EC2 instance and i was wondering if you can install the SDK directly on the IP camera and how can i implement it(im using HIKVISION DS-2CD2143G0-I(S) Fixed DomeNetwork Camera) or the need for the SDK to run in an instance/mac/Raspberry is necessary.

Everything works fine,but the main reason for this is to reduce latency.

MushMal, 2020-01-28:
@Moriuks Very cool setup indeed. There are couple of aspects you are touching. First, the SDK can be installed on pretty much any platform as long as you have access to the firmware/software image. You simply cross-compile it and use directly on the device. The second aspect is the latency. Your camera assuming streams out RTSP stream to your instance and the instance using KVS producer SDK would also achieve very low latency. However, I am not sure what Reconocimento Video box is - seems it's Amazon Rekognition service which is using fragmented stream. This will produce inherent latency of at least a fragment duration. So, if your fragment is 2 seconds long, it will be at least 2 seconds delay.

disa6302, 2020-02-28:
Closing this for now. Feel free to reopen or create a new ticket in the future.

closed 2020-02-28 by disa6302

souravssp, 2020-09-11:
Hello Kinesis team. Wanted to reach out for some building of new Architecture on AWS KINESIS. FYI We are an AWS Activate Startup. Please do reach out to me at sourav@oureye.ai