awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #332: Illegal instruction in getSamplingFreqAndChannelFromAacCpd function under MIPS platform

henryChmn, 2020-01-31:
We had problem when try to run KvsAacAudioVideoStreamingSample example code under our camera with MIPS platform, Illegal instruction as below,
seems relative the parameter assign address as below,
[image]

Illegal instruction:
*pSamplingFrequency = gMkvAACSamplingFrequencies[samplingRateIdx];

the correct address as below,
[image]

The sample code run under Ubuntu are well.

henryChmn, 2020-01-31:
fixed by modify mkvgenAdaptCodecPrivateData function call

closed 2020-01-31 by henryChmn

MushMal, 2020-01-31:
@henryChmn could you please give us a little more information? Is this still an issue on MIPS?

henryChmn, 2020-02-03:
@MushMal  It's an issue on MIPS for current example code on the git.
We fixed it by modify PIC library function call to access the parameter.

it's seems relative memory allocate as below, 
allocationSize = SIZEOF(KinesisVideoStream) + tagsSize + trackInfoSize + MKV_SEGMENT_UUID_LEN;
pKinesisVideoStream = (PKinesisVideoStream) MEMCALLOC(1, allocationSize);

MushMal, 2020-02-03:
@henryChmn thanks for the info but I am still unsure how this breaks as both of the locations you've provided would work just fine. Are you fixing up for alignment memory/non-tight packed build?

henryChmn, 2020-02-03:
@MushMal  it's working in Ubuntu but not MIPS. 
Inside the function call can't give the parameter value, only outside the function call could do that.
I'm not fixed alignment memory/non-tight packed build just avoid assign value this parameter inside the function call.

MushMal, 2020-02-03:
@henryChmn thanks for bearing with me here. The inspection of the code looks good - we do provide a valid pointer to DOUBLE which gets filled by the function. Perhaps we see an issue with the compiler handling double out of alignment.