awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #333: How to start and stop streaming from consumer side of kinesis video streams while using cpp producer sdk?

suhailmoideenv, 2020-02-02:


MushMal, 2020-02-02:
@suhailmoideenv I am not entirely sure I understand the question so let me reflect it back. You are attempting to use a lambda which should trigger a producer start and will handle the stream? 
 
Just a word about the lambda - those have time limits on the execution so ensure that your application run time is within the lambda limit.

Generally, if you actually want to trigger the start of the producer, you need some sort of agent installed on the producer side that will listen to events from your lambda application and start streaming. IoT might be a good start. Can you provide more details on your application so perhaps we can suggest a solution architecture that might work best for you.

suhailmoideenv, 2020-02-03:
@MushMal thanks for the reply.
My proposed poc is for streaming from a usb cam inside a moving car to kinesis by making use of a in car processor (preferring nvidia jetson) and then from Kinesis to app using a lambda function which wil be consumed at client side
But trigger of on and off should be on app side.

MushMal, 2020-02-03:
@suhailmoideenv very cool scenario. As I mentioned, be mindful of the lambda execution times: https://docs.aws.amazon.com/lambda/latest/dg/limits.html it's 15 mins.

You really need to create the triggering mechanism for your producer. There are multiple solutions which you could use, including IoT mechanisms. 

Your application could simply be executed on the trigger. It could create the producer object, create a stream object, put frames into the stream object for the duration of the execution, stop the stream (sync) and then free the producer. Repeat on new trigger.

It's hard to suggest an appropriate solution architecture. Is your app going to be based on gstreamer? Is it going to use kvs plugin? Are you planning on using some custom media pipeline?

suhailmoideenv, 2020-02-03:
@MushMal Awesome..
iam planning to use gstreamer and kvs plugin with out the any custom media pipeline.
trying to make it work end to end with existing libraries on aws.(newbee on any video streams)

MushMal, 2020-02-03:
Sounds great. Let us know how we can help after you put together the application. Do you mind resolving this issue and opening a more targeted ones once they arise?

suhailmoideenv, 2020-02-03:
> Sounds great. Let us know how we can help after you put together the application. Do you mind resolving this issue and opening a more targeted ones once they arise?

closed 2020-02-03 by suhailmoideenv