awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #334: Codec changed without reporting the current codec

ib1368, 2020-02-03:
Hello,

I am using this SDK v2.1.0 to stream video to kinesis video stream. It was working for long time, but suddenly it reported codec changed error without telling more about the new codec. I didn't face this error while I was using v 1.7.9.
Any one faced similar problem?

[image]

Thanks in advance.

MushMal, 2020-02-03:
@ib1368 this issue is generated on the backend playback. This generally means that between the fragments there is a difference in codec private data bits. Either resolution changed or even if the fragment is missing the CPD. A KVS stream is composed of multiple KVS stream sessions (PutMedia). When using SDK, it outputs the CPD as part of the MKV header once at the beginning of the streaming session. The backend, however, prepends the MKV header to every fragment. 

It is possible that your encoder got re-configured - perhaps a resolution change or encoder settings change which might change the CPD. Do you happen to have the logs?

ib1368, 2020-02-03:
Thanks. So why doesn't it report the new codec type. When I searched for similar error, I saw the new codec is reported in the error message as well.
Can dynamcic bit rate cause the issue?
Which logs do you mean? This doesn't happen frequently.

MushMal, 2020-02-03:
Those are different conditions I believe for the console playback application.

Could you elaborate a little what you mean by dynamic bit rate? Are you using different streaming sessions with different CPDs (bit rates/resolutions) or your encoder is able to dynamically change the resolution?

In the latter case, I assume the encoder produces an Annex-B format frames where the I-frame is also pre-pended the SPS/PPS NALus. If they change, dynamically the playback will be broken.

ib1368, 2020-02-04:
Thanks for the information. During my investigations, I understood that the fragment which is the source of codec changed error is related to next video session. If you look at the attached image, the error happens at 9:51:31, but the reported fragment is for 10:35:54. Does it make sense? Actually the first video session is terminated due to some unknown errors(that could be potentially codec changed error due the the error message on the player) and the video session is restarted at some point of time which is the fragment that is reported as source of bug, but it is a fragment from next session. 
[image]

ib1368, 2020-02-04:
I also tried to investigate 3 video fragments. The one that is reported as source of "codec changed" error. One from 1st video session and one from 2nd video session. All the 3 fragments had same codec type, resolution, ...

ib1368, 2020-02-04:
I tried to check the logs. This error with error code 0x5200005F happens at the same time and the process get killed. Based on [error code page](https://github.com/awsdocs/AWS-Kinesis-Video-Documentation/blob/master/doc_source/producer-sdk-errors.md), it shouldn't happen: "Kinesis Video Streams enforces the max connection duration as specified in the Producer SDK Limits. The Producer SDK automatically rotates the stream/token before the maximum is reached, and so clients using the SDK should not receive this error."

So I have 2 questions:

- why should this happen despite the document says it shouldn't happen?
- And why do we have codec change error which is unrelated?

MushMal, 2020-02-04:
Sorry for the delay. STATUS_ACK_ERR_CONNECTION_DURATION_REACHED is generated by the backend when it sees fragments which are longer than max supported duration which is defined as 10 seconds: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/limits.html

The documentation is slightly misleading - the SDK will enforce the theoretical limit which is signed 16bit timestamp field in the MKV (at the default time scale of 1ms it would evaluate to ~32 secs).

Could you please check whether your encoder is producing I-frames at regular intervals (GoP size should be set from 1-10 seconds). This could also theoretically happen in case if an I-frame gets dropped on buffer pressure from the producer side and the next fragment P-frames just happen to fall on a whole frame boundary.

MushMal, 2020-02-04:
Actually, I am wrong. This particular error is another limit which is the overall PutMedia streaming session duration. The SDK has an enforced limit of 40 minutes at which a new session is generated.  The 40 minutes is smaller than the default session limit of 45 minutes so the new session should take over. However, it is possible that the previous session runs longer than the max duration due to very slow network which still is enough that it doesn't trigger the slow connection limit trigger. Could you please retry and capture the logs which might help to properly diagnose the issue?

MushMal, 2020-03-17:
@ib1368 Any further updates on this issue? Please resolve if satisfied, otherwise, please provide more update so we can help.

MushMal, 2020-03-21:
Resolving this as stale. Please re-open or cut another issue.

closed 2020-03-21 by MushMal

ib1368, 2020-04-01:
This error happens a lot and also seems to be unrecoverable.
I don't know why it is not recoverable. https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/318371e2b71938a95c33cdd8ec866b82cff6e2e7/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L188

How can the logs help? what should be the log level to help in this case?

ib1368, 2020-04-01:
This is not the only error that I face and shouldn't happen based on the error code page. I see also some STATUS_ACK_ERR_FRAGMENT_TIMECODE_NOT_MONOTONIC errors. Not sure why. is it possible to use the SDK in a wrong way?

disa6302, 2020-04-01:
Hello @ib1368 ,

For the STATUS_ACK_ERR_FRAGMENT_TIMECODE_NOT_MONOTONIC error, there is an active PR addressing this. The error is generated when the timestamp of the last frame of previous fragment and the key frame of next fragment are same. For more information, you can look at the current issue: 

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/386

> This error happens a lot and also seems to be unrecoverable.
> I don't know why it is not recoverable. [...]

Do you mean INVALID_MKV_DATA? If yes, it is recoverable and if your app is failing, it might be something else. Please attach logs with VERBOSE level in a new issue so that we can see what is going on. Thanks!

ib1368, 2020-04-01:
Hello @disa6302 
Thanks for the information.
Which PR fixes STATUS_ACK_ERR_FRAGMENT_TIMECODE_NOT_MONOTONIC error? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/316 ?

> Do you mean INVALID_MKV_DATA? If yes, it is recoverable and if your app is failing, it might be something else. Please attach logs with VERBOSE level in a new issue so that we can see what is going on. Thanks!

I mean STATUS_ACK_ERR_CONNECTION_DURATION_REACHED error

MushMal, 2020-04-01:
@ib1368 STATUS_ACK_ERR_CONNECTION_DURATION_REACHED - one of the reasons we might hit this is due to stale credentials. We pushed a change last week. Do you mind updating your codebase?