awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #335: x264enc and bframes introduces time shift in ingested video

ib1368, 2020-02-03:
Hello,

I am using X264enc gstreamer element to encode the video before streaming it to kinesis video backend. As it is specified in [this doc](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html), we need to set bframes and key-int-max. by default bframes is 0 and I didn't set it to 0 in my code. It was working with this configuration when I was using SDK v1.7.9. After switching to SDK v2.1.0, I saw that we have time shift around 1 min in the first 5 mins and after a few hours, it was more than half an hour. I tried to investigate it and seemed that while the video stream is smooth the producer timestamp jump in HLS player(in aws console). After some investigation I figured out that the problem pts and dts are not equal for x264enc output. after setting b-adapt to false and also explicitly setting bframes to 0(which seems to be a bug, because it should be by default 0) the problem didn't happen again. so I have some questions:

- Why did this happen after switching from v1.7.9 to v2.1.0?
- How does the example in the above mentioned link works for you without setting b-adapt? Did you check for time shift problem for the new version?
- Why isn't this SDK able to handle b-frames?

Thanks in advance

MushMal, 2020-02-03:
@ib1368 thanks for reporting. There are few major updates between these versions but not a lot when it comes to the GStreamer interface. We have introduced a GStreamer filter that would take care of different timestamps from different sources (as some sources are using relative while others are absolute). This might be causing issues on our side. We need to investigate this further.

To answer your question about the b-frames. B-frames reference frames that are following them - aka, frames in the future. The encoder should have longer reference time in order to optimize the encoding and introduce b-frames which are encoded against later in the timeline. This won't work with low-latency encoders as the current frame bits should be produced out in order to not impact the latency. 

This said, the SDK itself is able to handle b-frames as long as they are not the first frame in the fragment.

chehefen, 2020-02-06:
@ib1368 thanks for reporting. Can you give us the **exact** step on how to reproduce the time shift issue? 
Here is what I tried to reproduce the issue myself:
I used this gstreamer pipeline to produce the stream:
```
gst-launch-1.0 videotestsrc is-live=TRUE ! clockoverlay ! videoconvert ! x264enc key-int-max=30 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name=my-stream access-key="" secret-key=""
```
Like you described, I didn't set `b-adapt` and `bframes` param for x264enc. 
Then I viewed the stream using the player in AWS console for 5 mins. 
I didn't observe any time shift. The clock overlay displayed in video plus latency were always roughly current clock time. However, occasionally I did observe the player stalled for a few seconds possibly due to network fluctuation and when the player recovered it recovered at the last played frame, thus falling behind the latest stream. But you can jump to latest by clicking the fast forward button and everything played ok.
I also tried explicitly setting `b-adapt=TRUE` and `bframes=5`, no time shift was observed either.

MushMal, 2020-03-17:
I am resolving this as stale. Please re-open if you have more info

closed 2020-03-17 by MushMal