awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #336: Send metadata along media to KVS

matrosov-nikita, 2020-02-05:
Hi guys,
I'm using gst-launch-1.0 CLI tool with kvssink plugin to export video from camera to Amazon KVS, there is a pipeline:

`gst-launch-1.0  rtspsrc location=<CAMERA_ADDRESS> short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=my-stream storage-size=512 retention-period=10 access-key=<ACCESS_KEY> secret-key=<SECRET_KEY>`

It works perfect, but now I need to send some metadata along video fragments (e.g. some info about my camera). According to [docs](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-meta.html) I can do it using SDK, but is there way to send metadata using gst-launch-1.0?

I see in the code, that a kvssink plugin listens to  event: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/0888c666efbe7f20891a86c9e64a677f73e33b17/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp#L947), but I could not find a way to send it with gst-launch-1.0 CLI tool.

Let me know if I need to provide more details. Thanks.

MushMal, 2020-02-05:
@matrosov-nikita thanks for the question. The KVS sink itself supports the fragment metadata via custom events gst_element_send_event. However, I am not aware of ways custom events could be constructed/sent when using gst-launch-1.0.

matrosov-nikita, 2020-02-06:
@MushMal thanks for your quick response.
Probably I don't need to send custom metadata at all. Let me explain my case, I'm streaming to KVS using [this sample](
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/demo/kvs_producer_plugin_rtsp_demo.cpp) periodically when some internal event happens. 

Then I need to seek fragments related to desired event. My guess is to save producer timestamp(or fragment number) once streaming started and then use it with GetMedia request. But I see no way to pass (or get) this timestamp.

MushMal, 2020-02-06:
@matrosov-nikita I understand your scenario. Indeed, having embedded metadata in the stream itself will not help. Even with an ability to query for metadata is implemented by the KVS, still, you won't know the timestamps in this case to query. 

I believe in your case you need to do some solution architecting. While you can choose many alternative solutions but perhaps having a kinesis data stream which gets the necessary info on when the KVS streaming session happened (timestamps/durations/other info). On that end, you could trigger a lambda from KDS to perform actions.

We are contemplating on building an ability to trigger lambda on certain events - that would have been helpful for your case.

matrosov-nikita, 2020-02-06:
@MushMal 
Yeah, probably using KDS with lambdas is a good choice, but I wonder if there is way to get PRODUCER_TIMESTAMP? It should be sending with each fragment according to PutMedia docs; fragment's acknowledgement message includes a FragmentTimecode as well.
Just need the easiest solution to retrieve specific fragments from stream.

MushMal, 2020-02-06:
The easiest would be to ensure you are using Absolute timestamps in the stream first of all to make sure those map directly into the stream index. The default setting (which needs to be changed) for GST KVS plugin is Relative. You can specify https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp#L499 property that controls the selection of producer timestamps.

This said, the relative timestamps will be translated to absolute in the backend taking the PutMedia stream start timestamp as the base but that can skew the clock a little.

The main issue then becomes knowing the start timestamp. The real-time clock on the system might not be the one that's used by the media pipeline encoder. It's good to simply record the system time at the event and attempt to load the stream at that time. The consumer APIs (ListFragment, GetMedia, etc) use open (inclusive) periods so if you specify the start timestamp the producer time then your first fragment will be included.

If, however, your media pipeline uses other clocks (high precision media clock for example on some devices) then you might have some difficulty on knowing the producer timestamp on the device unless you are willing to write an application similar to GStreamer sample app we have instead of using gst-launch-1.0 utility. The application passing the frames to the plugin can then record the producer timestamp.

matrosov-nikita, 2020-02-06:
@MushMal thank you, sounds good to me. Could you please provide a link to code in Gstreamer sample app where I can override frame timestamps? (I see a bunch of examples, just want to ensure that we are on the same page).

MushMal, 2020-02-06:
Sure.

Set it to TRUE here
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L37

Have a flag if you are just starting then save the frame timestamp that's put into KVS here
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L392

MushMal, 2020-02-06:
NOTE: This is a video only sample. There are similar samples in the same directory with multi-stream and multi-track

matrosov-nikita, 2020-02-12:
@MushMal, thanks for your support. Have not worked with Cmake files before, it would be nice to have Cmake file just for kinesis_video_gstreamer_sample_app.
I'm trying to change Cmake I used for kvs_producer_plugin_rtsp_demo - https://gist.github.com/matrosov-nikita/6e95d6cf3a22c89909226a03b31c8c4b, but w\o any luck, get list of errors like:

'gst-exporter.cpp:(.text+0x1e5): undefined reference to `log4cplus::Logger::isEnabledFor(int) const'
gst-exporter.cpp:(.text+0x1f7): undefined reference to `log4cplus::detail::get_macro_body_oss[abi:cxx11]()'
gst-exporter.cpp:(.text+0x25c): undefined reference to `log4cplus::detail::macro_forced_log(log4cplus::Logger const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
....
`

MushMal, 2020-02-13:
@matrosov-nikita yes indeed, we are looking at separating the samples which would then have simple cmake files. Working with CMake files is not my forte either - I just don't fundamentally understand them perhaps.

Are you able to compile the existing repository? If so, I would simply start off with the sample app modification. 

I am not familiar with gst-exporter.cpp - perhaps this is part of the log4cplus? 

@chehefen any pointers?

matrosov-nikita, 2020-02-13:
@MushMal  I just copied kinesis_video_gstreamer_sample_app.cpp to my custom gst-exporter.cpp file and tried to create separate CMake.

MushMal, 2020-02-13:
You need to include log4cplus. The install script pulls down the sources and builds it from scratch: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/install-script#L141

Perhaps the easiest would be to install it and use it on your system using standard packaging tools. For example on mac osx you could run `brew install log4cplus`

matrosov-nikita, 2020-02-13:
I'm using Dockerfile mentioned above, seems it aleady includes install-script instruction. My guess is that's probably Cmake issue.

MushMal, 2020-02-14:
Could you make sure you have an instruction in your CMake to validate if log4cplus is installed like so: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/CMakeLists.txt#L136

You should ensure the log4cplus libraries are linked within your targets target_link_libraries command by adding ${LOG4CPLUS_LIBRARIES}

Then, if you are on windows, ensure you link ${KINESIS_VIDEO_OPEN_SOURCE_SRC}/log4cplus-1.2.1/msvc10/${KINESIS_VIDEO_BUILD_BITNESS_TYPE}/bin.Release/log4cplusS.lib

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/CMakeLists.txt#L442

matrosov-nikita, 2020-02-16:
@MushMal thank you, will try to fix. Close this issue now.

closed 2020-02-16 by matrosov-nikita