awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #339: CPD data invalid

Subburaj47, 2020-02-10:
Hi Team, 

We have a scenario, where we pull the frames out of propreitary platform API and have it as transport streams. This is fed to gstreamer pipeline and the frames are pushed to KVS SDK. This is our working model. We are decoupling gstreamer and pushing the H.264 frames fetched from the platform directly to KVS. We get a Invalid codec private data exception.
Meanwhile, we had a quick experiement to compare the frames from the platform APIs and the frames from gstreamer (for the same playback). The header data for the frames from the platform were of the format NALu||SPS||NALu||PPS. From Gstreamer, this looked different. When we tried to replace the gstreamer header to the frames obtained from the platform, we had the playback but with macro blocking.

Please advise here the right CPD data or any dedicated parameters if any during the stream definition. Attaching the sample frames (I-frame and P-frame) obtained out the platform API and the gstreamer implementation.
[frames.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4178960/frames.zip)

Thanks for your help in advance!!

MushMal, 2020-02-10:
@Subburaj47 your Hydra frames are in Annex-B format and can be directly produced into KVS using Producer SDK. Moreover, please specify Adapt CPD NALus and Adapt Frame NALus and the CPD will be extracted automatically and the frames will be converted automatically to AcCC format which would then be playable in the KVS console (or you can get HLS/MPEG-DASH stream).

This sample does exactly what you are doing/need:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/samples/KvsVideoOnlyStreamingSample.c

The default stream info provider will set it by default:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/src/source/StreamInfoProvider.c#L115

Subburaj47, 2020-02-11:
Thanks @MushMal . Earlier we used to supply the CPD while we call the start() function before pushing the streams. From what you have advised us above, are we supposed to use the overridden version of the start function (without arguments) ?
I went through the KinesisVideoStream.h header where I saw the start() to be mandated in the comments.

MushMal, 2020-02-11:
@Subburaj47 the start() API in the C++ layer supplies the CPD at the beginning of the streaming session. This should be called before any frames are produced, otherwise, an invalid state error will happen. This is done so we can package the CPD as part of the header before any frames are produced. In your case, the SDK can take care of the CPD as it's available as part of your H264 Annex-B frames. You can skip the call to start() all together as it's a no-op for now for the parameterless override.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-producer/src/KinesisVideoStream.cpp#L109

chehefen, 2020-03-03:
Closing the issue as it's stale. Please create new issue if there is any other problem.

closed 2020-03-03 by chehefen