awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #340: How to reduce starting video stream time on Raspberrypi?

kamlesh-solution, 2020-02-11:
Hello,

I have installed AWS kinesis video stream producer sdk on my raspberry pi and i am successfully streaming from my raspberrypi camera to kinesis cloud. 

But when i enter command for start streaming on my raspberrypi it takes 30 to 60 seconds to start streaming on kinesis and i can show hls playback after this delay time. I am using kinesis_video_gstreamer_sample_app for start streaming.

How can i reduce this starting streaming time? so i can see live streaming instantly with hls playback.

Thanks.

chehefen, 2020-02-11:
Hi, sdk shouldn't take 30 to 60 seconds to start streaming, normally it's just 1 to 2 seconds. Are you using the correct AWS region? By default sdk uses us-west-2. Can you also show us the sdk log? It can give us some idea what is taking time. Thanks.

MushMal, 2020-02-11:
@kamlesh-solution as @chehefen mentions, this delay points to something else that's going on. Also, how are you measuring the delay? Do you see the video from 30-60 seconds ago or you see delay of a couple of seconds as the streaming starts?

kamlesh-solution, 2020-02-12:
Hi @chehefen , @MushMal  

Thanks for your reply , 

Yes i m using default sdk region us-west-2. After run kinesis_video_gstreamer_sample_app it taking 30 to 60 second and after this delay time i can see playback of streaming on https://aws-samples.github.io/amazon-kinesis-video-streams-media-viewer/ this hls playback viewer app.

This following is my sdk log after running kinesis_video_gstreamer_sample_app for start streaming.

[INFO ] [12-02-2020 05:15:27:934.470 GMT] Using region: us-west-2
[INFO ] [12-02-2020 05:15:27:934.541 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [12-02-2020 05:15:27:934.557 GMT] No session token was detected.
[INFO ] [12-02-2020 05:15:27:938.757 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [12-02-2020 05:15:27:938.897 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [12-02-2020 05:15:27:938.914 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [12-02-2020 05:15:27:938.988 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [12-02-2020 05:15:27:939.029 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [12-02-2020 05:15:27:939.054 GMT] getSecurityTokenHandler invoked
[DEBUG] [12-02-2020 05:15:27:939.146 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1581484527939076534 Expiration: 0
[INFO ] [12-02-2020 05:15:27:939.177 GMT] New credentials expiration is 1581486927
[DEBUG] [12-02-2020 05:15:27:939.211 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [12-02-2020 05:15:27:939.230 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [12-02-2020 05:15:27:939.252 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [12-02-2020 05:15:27:939.268 GMT] clientReadyHandler invoked
[DEBUG] [12-02-2020 05:15:27:939.288 GMT] Client is ready
[INFO ] [12-02-2020 05:15:27:939.344 GMT] Creating Kinesis Video Stream dca6326a8c62
[INFO ] [12-02-2020 05:15:27:939.368 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [12-02-2020 05:15:27:939.549 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [12-02-2020 05:15:31:208.216 GMT] writeHeaderCallback(): RequestId: 4aa74af9-e7f8-4232-86aa-bdb8b644145a
[DEBUG] [12-02-2020 05:15:31:208.430 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.581050738136E9,"DataRetentionInHours":1,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-west-2:535693658748:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:535693658748:stream/dca6326a8c62/1581050738136","StreamName":"dca6326a8c62","Version":"FUA2zLBUxkj8g0MtTn2F"}}
[INFO ] [12-02-2020 05:15:31:209.258 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [12-02-2020 05:15:31:209.329 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [12-02-2020 05:15:31:209.359 GMT] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [12-02-2020 05:15:31:209.400 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[WARN ] [12-02-2020 05:15:36:240.788 GMT] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/getDataEndpoint with result Timeout was reached: Resolving timed out after 5000 milliseconds
[WARN ] [12-02-2020 05:15:36:240.930 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/getDataEndpoint
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAXZOOURJ6NVDWRP7K/20200212/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=b56af0be887053bdf5c99e814066f3203e6cdbd6a7e39b73f61ccfcf163473a2
    content-length: 58
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/2.1
[DEBUG] [12-02-2020 05:15:36:240.969 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: 
[INFO ] [12-02-2020 05:15:36:241.088 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[WARN ] [12-02-2020 05:15:41:472.286 GMT] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/getDataEndpoint with result Timeout was reached: Resolving timed out after 5000 milliseconds
[WARN ] [12-02-2020 05:15:41:472.604 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/getDataEndpoint
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAXZOOURJ6NVDWRP7K/20200212/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=042adb2a5e9f88f1bab857a6d2417a638d6addcbb75e6c55e7d0ef8b363189fd
    content-length: 58
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/2.1
[DEBUG] [12-02-2020 05:15:41:472.648 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: 
[... cut ...]
[DEBUG] [12-02-2020 05:16:30:879.980 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG] [12-02-2020 05:16:30:919.937 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG] [12-02-2020 05:16:31:225.015 GMT] postWriteCallback(): Curl post body write function for stream with handle: dca6326a8c62 and upload handle: 1 returned: {"EventType":"RECEIVED","FragmentTimecode":7987,"FragmentNumber":"91343852333181625531135231774324715573222763666"}

chehefen, 2020-02-12:
can you tell us where are you running the raspberry pi?

kamlesh-solution, 2020-02-12:
I m running raspberrypi in india.

MushMal, 2020-02-12:
Makes sense now. You have multiple timeouts trying to reach backend. For example: WARN ] [12-02-2020 05:15:36:240.788 GMT] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/getDataEndpoint with result Timeout was reached: Resolving timed out after 5000 milliseconds

While the SDK retries the connection/API, the buffer starts accumulating the frames:

> Overall storage byte size: 134217728
> Available storage byte size: 131573268
> Allocated storage byte size: 2644460
> Total view allocation byte size: 144064
> Total streams frame rate (fps): 30
> Total [...]

Here, you have 27095 ms or 27 seconds in the buffer - meaning that your playback will be at least that much behind.

Could you keep streaming and see if this can catch up (if you have good throughput). Then, in the console/HLS - hit the button to move to the head to playback from the head.

Your observed throughput is 18 Mbps

MushMal, 2020-02-12:
One more suggestion - try to select a region closest to your location

chehefen, 2020-02-12:
you can try change the region to ap-south-1 by doing `export AWS_DEFAULT_REGION=ap-south-1`

kamlesh-solution, 2020-02-13:
Thanks @MushMal  and @chehefen After changing region to ap-south-1 my streaming start very fast and i can show my video playback after 4 to 5 second using hls.

can you tell me how can i change region on sdk code without set environment variable?

MushMal, 2020-02-13:
Depends on the asset you are using. The sample, for example, is using

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/samples/KvsVideoOnlyStreamingSample.c#L98

Please resolve this issue if you have no further questions

closed 2020-02-14 by kamlesh-solution