awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #341: Fix documentation for frame_timecodes

sharadrj, 2020-02-11:
Documentation for frame_timecodes doesn't look right. (https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-data.html#how-data-header-streamdefinition)
 
It says "If true, Kinesis Video Streams stamps the frames when they are received. If false, Kinesis Video Streams uses the decode time of the received frames." However, looking at the code, it seems like the reverse. 

The documentation should probably say something to this effect: 
"if true, Kinesis Video Streams uses the decode time (dts and pts received from encoder) of the received frames. If false, Kinesis Video Streams stamps the frames when they are received with system-generated time values."

disa6302, 2020-09-17:
Closing. Tagged this in issue #423 to keep all documentation related items at one place

hassanctech, 2022-09-22:
Related issues were all closed but this was never fixed.  Re-opening.

sirknightj, 2023-11-02:
Tracking internally.

closed 2023-11-02 by sirknightj