awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #342: RPI Install problem

istranic, 2020-02-13:
I'm having an issue installing the produced SDK on my RPI4. I followed the instructions in the "Install Steps for Ubuntu 17.x and Raspbian Stretch using apt-get" and used "./min-install-script". 

I'm able to create a stream using "kinesis_video_gstreamer_sample_app ...", but I can't create a stream using "gst-launch-1.0.........". I get the error "WARNING: erroneous pipeline: no element "kvssink""

I assume my gstreamer plugin cannot be found, and when I check the "/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib" folder, there's no gstreamer related files there.

It's not clear to me where the gstreamer plugin was installed, given that I used the mini-install-script. How can I check this? Is there a way for force install-script to work on my RPI, instead of having to use mini-install-script.

I'd greatly appreciate your help to debug this problem.

closed 2020-02-13 by istranic

perronj2, 2020-04-09:
Did you find a solution to this? I am also having the same problem.

MushMal, 2020-04-09:
@perronj2 the min-install-script does not "install" the plugin. It builds the binary but you need to follow the steps in the readme to set the paths to the plugin.

istranic, 2020-04-09:
@perronj2 If my memory serves me correctly, I used mini-install-script, and the issue I had was the setting of LD_LIBRARY_PATH and GST_PLUGIN_PATH

I used:
export LD_LIBRARY_PATH=<YourSdkFolderPath>/kinesis-video-native-build
export GST_PLUGIN_PATH=<YourSdkFolderPath>/kinesis-video-native-build

This is slightly different than the documentation. I feel like this suggestion was made by the AWS support team, but I can't find the GitHub issue or Email, so I'm not 100 sure. @MushMal does this sound right? I feel like you (or @chehefen ) and I discussed this at some point.

perronj2, 2020-04-09:
@istranic yup that fixed it. I thought I had set the environment variables correctly but I guess not. That finally worked. I've notice a some serious documentation flaws and spending considerable amount of time searching for solutions in random posts. Thanks for the help!

MushMal, 2020-04-09:
Thanks for the info guys. We absolutely would appreciate any contribution to the codebase, including the documentation fixes. Please feel free to submit PRs