awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #343: Streaming Thermal Camera Data

istranic, 2020-02-13:
I'm trying to stream thermal camera temperature data from a Rasperry Pi + USB camera (FLIR Lepton3.5 + PureThermal2) to Kinesis Video in GRAY8 or GRAY16_LE format. Could you please help me figure out how to structure the gstreamer command to make this work?

FYI, when I use the normal Raspberry pi camera, I am able to create a stream using either of the commands below. 
_gst-launch-1.0 -v v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=5/1 ! omxh264enc periodicty-idr=5 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName access-key="YourAccessKey" secret-key="YourSecretKey"_

_gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=5/1 ! x264enc  bframes=0 key-int-max=55 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"_

However, when I try to change the format (GRAY8 vs I420) and source (video1 vs video0), I can't get it to work. I suspect my command is no longer compatible with the encoder. In addition, I can successfully display the thermal camera feed when I run the command below, so I know my thermal camera is working.

_gst-launch-1.0 v4l2src device=/dev/video0 ! videoconvert ! ximagesink_

chehefen, 2020-02-13:
did you try running your second pipeline with the thermal camera? videoconvert should handle the conversion from GRAY8 to I420

MushMal, 2020-02-13:
@istranic looks like a cool usage case. Could you share a little more about your use case? Are you prototyping or you are building an app using GStreamer KVS plugin?

istranic, 2020-02-13:
@chehefen is this what you're referring to?

gst-launch-1.0 -v v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"

I get an error related to the internal data stream. I can post a full log if that helps?

If I run the command above with video0(non thermal) instead of video1, it works fine.

Can you pls help me understand the command above. Is I420 the format of the input video, or that is format at the output of the encoder?

istranic, 2020-02-13:
@MushMal I'm prototyping right now, but my goal is to build a piece of hardware that will take a visible and IR video and stream it to the cloud. I'd really appreciate any advice on how to do this. My approach was to use Kinesis Video and somehow figure out how to stream from multiple cameras. First, I'm figuring out how to stream from the visible and IR cameras separately, and then I'll have to figure out how to do it together.

chehefen, 2020-02-13:
can you do `export GST_DEBUG=4` and then try this? 
```
gst-launch-1.0 -v v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,width=640,height=480,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"
```
If it didn't work, you can send us the log to kinesis-video-support@amazon.com

In your second pipeline,  I420 is the format of input video to the encoder. however, your source video can be other format and videoconvert will handle the conversion to I420.

MushMal, 2020-02-13:
@istranic very cool scenario. There are two parts to your producer side. 1) How to extract and encode the frames and 2) How to stream to KVS.

#1 you could indeed use GStreamer to work with the source and the encoders, unless you have a hardware encoder or custom pipeline that can give you already encoded frames that you could then submit to KVS SDK directly, circumventing GStreamer all together. The color-space converter or encoder is something that can be resolved by looking at the documentation or dumping out the errors gstreamer produces.

#2 would be based on the choices on #1. If it's GStreamer then the recommended approach would be to use GStreamer based application and put frames directly into KVS SDK or use the the KVS GStreamer plugin either in a GStreamer application or have a daemon with gst-launch-1.0 tool and command line that integrates with kvs plugin. As higher you go up the tool chain you go the easier is the integration but less flexible in terms of configurability or edge case handling like buffer pressures.

You can certainly prototype using kvs plugin with gst-lunch-1.0 but I would recommend putting together an application that would construct the pipeline either with kvs plugin or without - in which case you would have some sort of an app sink that will feed the frames into kvs producer SDK (similar to KVS plugin or gstreamer sample application).

istranic, 2020-02-13:
@chehefen It didn't work and I send you the log via email

@MushMal Your comment make sense, though the implementation is a bit outside of my area of expertise. Would you be interested in helping with this. If so, we can connect via email.

MushMal, 2020-02-13:
@istranic we received your log and will look through it to see what could cause GStreamer error - most likely an issue with the unexpected format in the encoder.

Regarding the help. We would gladly help you by answering questions related to issues with KVS or the KVS SDK in general. It will be hard for us to help with your application though.

Recommend taking a look at the samples and see if you can replicate. If, however, you are not an expert in native code development then perhaps the GStreamer KVS plugin with gst-launch-1.0 tool might fit the bill.

chehefen, 2020-02-13:
in the log there was this line:
`
0:00:01.251925453  4123   0x812d80 INFO                    v4l2 gstv4l2object.c:3927:gst_v4l2_object_get_caps:<v4l2src0> probed caps: video/x-raw, format=(string)UYVY, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ 2:4:7:1 }, framerate=(fraction)9/1; video/x-raw, format=(string)BGR, width=(int)160, height=(int)122, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)BGR, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction)9/1; video/x-raw, format=(string)RGB16, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ sRGB }, framerate=(fraction)9/1; video/x-raw, format=(string)GRAY8, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ 2:4:7:1 }, framerate=(fraction)9/1; video/x-raw, format=(string)I420, width=(int)160, height=(int)122, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)I420, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction)9/1; video/x-raw, format=(string)YV12, width=(int)160, height=(int)122, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)YV12, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction)9/1; video/x-raw, format=(string)RGB, width=(int)160, height=(int)122, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction)[ 0/1, 2147483647/1 ]; video/x-raw, format=(string)RGB, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, framerate=(fraction)9/1; video/x-raw, format=(string)GRAY16_LE, width=(int)160, height=(int)122, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ 2:4:7:1 }, framerate=(fraction)9/1; video/x-raw, format=(string)GRAY16_LE, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string){ 2:4:7:1 }, framerate=(fraction)9/1
`

It stated the supported format of your camera. As you can see, the camera don't support resolution 640x480, you can either drop the resolution or you can add videoscale in the pipeline like such:

`gst-launch-1.0 -v v4l2src device=/dev/video1 ! videoconvert ! videoscale ! video/x-raw,width=640,height=480,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"
`

istranic, 2020-02-13:
Hi chehefen,

I forgot to mention previously that I tried 160x120, which is the
resolution of the thermal camera. Attached are the logs for the lower
resolution method or the scaling method.

-- 
Ivo Stranic
Stanford University

COMMAND
gst-launch-1.0 -v v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,width=160,height=120,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName3 storage-size=128 access-key=llllllll secret-key=llllllllllllllllll aws-region="us-east-1"

LOG
0:00:00.000492393  4430   0x33ee00 INFO                GST_INIT gst.c:510:init_pre: Initializing GStreamer Core Library version 1.10.4
0:00:00.000770621  4430   0x33ee00 INFO                GST_INIT gst.c:511:init_pre: Using library installed in /usr/lib/arm-linux-gnueabihf
0:00:00.000848381  4430   0x33ee00 INFO                GST_INIT gst.c:522:init_pre: Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l
0:00:00.002404935  4430   0x33ee00 INFO                GST_INIT gstmessage.c:126:_priv_gst_message_initialize: init messages
0:00:00.005490542  4430   0x33ee00 INFO                GST_INIT gstcontext.c:83:_priv_gst_context_initialize: init contexts
0:00:00.007645426  4430   0x33ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:316:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.008263235  4430   0x33ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:224:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.008341464  4430   0x33ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.008516567  4430   0x33ee00 INFO            GST_REGISTRY gstregistry.c:1738:ensure_current_registry: reading registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.120280368  4430   0x33ee00 INFO            GST_REGISTRY gstregistrybinary.c:619:priv_gst_registry_binary_read_cache: loaded /home/pi/.cache/gstreamer-1.0/registry.arm.bin in 0.111650 seconds
0:00:00.120565367  4430   0x33ee00 INFO            GST_REGISTRY gstregistry.c:1594:scan_and_update_registry: Validating plugins from registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.133202379  4430   0x33ee00 INFO            GST_REGISTRY gstregistry.c:1696:scan_and_update_registry: Registry cache has not changed
0:00:00.133266025  4430   0x33ee00 INFO            GST_REGISTRY gstregistry.c:1773:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.133299150  4430   0x33ee00 INFO                GST_INIT gst.c:720:init_post: GLib runtime version: 2.50.3
0:00:00.133332274  4430   0x33ee00 INFO                GST_INIT gst.c:722:init_post: GLib headers version: 2.50.3
0:00:00.133360347  4430   0x33ee00 INFO                GST_INIT gst.c:723:init_post: initialized GStreamer successfully
0:00:00.133490503  4430   0x33ee00 INFO            GST_PIPELINE gstparse.c:325:gst_parse_launch_full: parsing pipeline description 'v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,width=160,height=120,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName3 storage-size=128 access-key=lllllllllllllllll secret-key=llllllllllllllllllllllll aws-region=us-east-1 '
0:00:00.144581274  4430   0x33ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideo4linux2.so" loaded
0:00:00.144665701  4430   0x33ee00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "v4l2src"
0:00:00.147910422  4430   0x33ee00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseSrc@0x4a60a0> adding pad 'src'
0:00:00.148898906  4430   0x33ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideoconvert.so" loaded
0:00:00.148988801  4430   0x33ee00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "videoconvert"
0:00:00.149765516  4430   0x33ee00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseTransform@0x4ae260> adding pad 'sink'
0:00:00.149834838  4430   0x33ee00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseTransform@0x4ae260> adding pad 'src'
0:00:00.154648509  4430   0x33ee00 INFO                 x264enc gstx264enc.c:2741:plugin_init: x264 build: 148
0:00:00.155046840  4430   0x33ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstx264.so" loaded
0:00:00.155113715  4430   0x33ee00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "x264enc"
0:00:00.155978398  4430   0x33ee00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x4b4200> adding pad 'sink'
0:00:00.156108970  4430   0x33ee00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x4b4200> adding pad 'src'
0:00:00.232034065  4430   0x33ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/home/pi/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libgst [... cut ...]
0:00:00.658355567  4477   0xba4e00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<videoscale0> current NULL pending VOID_PENDING, desired next READY
0:00:00.658389994  4477   0xba4e00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<videoscale0> completed state change to READY
0:00:00.658417494  4477   0xba4e00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<videoscale0> notifying about state-changed NULL to READY (VOID_PENDING pending)
0:00:00.658457025  4477   0xba4e00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'videosca

chehefen, 2020-02-13:
log says it couldn't identify /dev/video1
`0:00:00.560516079 4430 0x33ee00 INFO GST_ERROR_SYSTEM gstelement.c:1890:gst_element_message_full_with_details:<v4l2src0> posting message: Cannot identify device '/dev/video1'.
`
can you try run `gst-device-monitor-1.0` and see what it prints?

istranic, 2020-02-13:
Hi,

I'm so sorry for the mistake. I accidentally unplugged the camera. This
time it's plugged in, and i verified that the came works works by
displaying a feed on my screen using gstreamer.

Attached are the error logs for when I try to stream to KVS using the
commands discussed previously.

-- 
Ivo Stranic
Stanford University

COMMAND
gst-launch-1.0 -v v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,width=160,height=120,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName3 storage-size=128 access-key=lllllllllllllllll secret-key=lllllllllllllllllll aws-region="us-east-1"

LOG
0:00:00.000404267  4813   0xed4e00 INFO                GST_INIT gst.c:510:init_pre: Initializing GStreamer Core Library version 1.10.4
0:00:00.000725202  4813   0xed4e00 INFO                GST_INIT gst.c:511:init_pre: Using library installed in /usr/lib/arm-linux-gnueabihf
0:00:00.000810879  4813   0xed4e00 INFO                GST_INIT gst.c:522:init_pre: Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l
0:00:00.002469044  4813   0xed4e00 INFO                GST_INIT gstmessage.c:126:_priv_gst_message_initialize: init messages
0:00:00.005530844  4813   0xed4e00 INFO                GST_INIT gstcontext.c:83:_priv_gst_context_initialize: init contexts
0:00:00.007714109  4813   0xed4e00 INFO      GST_PLUGIN_LOADING gstplugin.c:316:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.008426656  4813   0xed4e00 INFO      GST_PLUGIN_LOADING gstplugin.c:224:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.008507906  4813   0xed4e00 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.008668321  4813   0xed4e00 INFO            GST_REGISTRY gstregistry.c:1738:ensure_current_registry: reading registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.150725087  4813   0xed4e00 INFO            GST_REGISTRY gstregistrybinary.c:619:priv_gst_registry_binary_read_cache: loaded /home/pi/.cache/gstreamer-1.0/registry.arm.bin in 0.141943 seconds
0:00:00.150988783  4813   0xed4e00 INFO            GST_REGISTRY gstregistry.c:1594:scan_and_update_registry: Validating plugins from registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.163471347  4813   0xed4e00 INFO            GST_REGISTRY gstregistry.c:1696:scan_and_update_registry: Registry cache has not changed
0:00:00.163543222  4813   0xed4e00 INFO            GST_REGISTRY gstregistry.c:1773:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.163578742  4813   0xed4e00 INFO                GST_INIT gst.c:720:init_post: GLib runtime version: 2.50.3
0:00:00.163612701  4813   0xed4e00 INFO                GST_INIT gst.c:722:init_post: GLib headers version: 2.50.3
0:00:00.163641971  4813   0xed4e00 INFO                GST_INIT gst.c:723:init_post: initialized GStreamer successfully
0:00:00.163787908  4813   0xed4e00 INFO            GST_PIPELINE gstparse.c:325:gst_parse_launch_full: parsing pipeline description 'v4l2src device=/dev/video1 ! videoconvert ! video/x-raw,width=160,height=120,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName3 storage-size=128 access-key=lllllllllll secret-key=lllllllllllllllllllll aws-region=us-east-1 '
0:00:00.350634917  4813   0xed4e00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideo4linux2.so" loaded
0:00:00.350721062  4813   0xed4e00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "v4l2src"
0:00:00.353788331  4813   0xed4e00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseSrc@0x10400a0> adding pad 'src'
0:00:00.354688689  4813   0xed4e00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideoconvert.so" loaded
0:00:00.354744418  4813   0xed4e00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "videoconvert"
0:00:00.355524776  4813   0xed4e00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseTransform@0x1046260> adding pad 'sink'
0:00:00.355591026  4813   0xed4e00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseTransform@0x1046260> adding pad 'src'
0:00:00.358998605  4813   0xed4e00 INFO                 x264enc gstx264enc.c:2741:plugin_init: x264 build: 148
0:00:00.359191364  4813   0xed4e00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstx264.so" loaded
0:00:00.359236832  4813   0xed4e00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "x264enc"
0:00:00.359966515  4813   0xed4e00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x104b008> adding pad 'sink'
0:00:00.360033493  4813   0xed4e00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x104b008> adding pad 'src'
[... cut ...]
0:00:00.405843981  4846  0x129de00 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<videoconvert0:src> pad has no peer
0:00:00.408728288  4846  0x129de00 INFO                GST_PADS gstutils.c:1447:prepare_link_maybe_ghosting: v4l2src0 and videoconvert0 in same bin, no need for ghost pads
0:00:00.408803652  4846  0x129de00 INFO                GST_PADS gstpad.c:2315:gst_pad_link_prepare: trying to link v4l2src0:src and videoconvert0:sink
0:00:00.408864850  4846  0x129de00 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<videoc

istranic, 2020-02-14:
Hi chehefen,

Any chance you've had an opportunity to look at the logs from my previous
note.

Thanks,
Ivo

-- 
Ivo Stranic
Stanford University

COMMAND
gst-launch-1.0 -v v4l2src device=/dev/video1 ! videoconvert ! videoscale ! video/x-raw,width=640,height=480,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName3 storage-size=128 access-key=lllllllllllllll secret-key=llllllll aws-region="us-east-1"

PARTIAL LOG (I couldn't see the start of the log in the terminal)
0:00:00.329321681  4846  0x129de00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "x264enc"
0:00:00.330073552  4846  0x129de00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x1418798> adding pad 'sink'
0:00:00.330135843  4846  0x129de00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x1418798> adding pad 'src'
0:00:00.404396020  4846  0x129de00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/home/pi/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libgstkvssink.so" loaded
0:00:00.404488102  4846  0x129de00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "kvssink"
0:00:00.405328306  4846  0x129de00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "pipeline"
0:00:00.405671898  4846  0x129de00 INFO            GST_PIPELINE grammar.y:596:gst_parse_perform_link: linking some pad of GstV4l2Src named v4l2src0 to some pad of GstVideoConvert named videoconvert0 (0/0) with caps "(NULL)"
0:00:00.405728252  4846  0x129de00 INFO        GST_ELEMENT_PADS gstutils.c:1633:gst_element_link_pads_full: trying to link element v4l2src0:(any) to element videoconvert0:(any)
0:00:00.405771116  4846  0x129de00 INFO                GST_PADS gstutils.c:933:gst_pad_check_link: trying to link v4l2src0:src and videoconvert0:sink
0:00:00.405843981  4846  0x129de00 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<videoconvert0:src> pad has no peer
0:00:00.408728288  4846  0x129de00 INFO                GST_PADS gstutils.c:1447:prepare_link_maybe_ghosting: v4l2src0 and videoconvert0 in same bin, no need for ghost pads
0:00:00.408803652  4846  0x129de00 INFO                GST_PADS gstpad.c:2315:gst_pad_link_prepare: trying to link v4l2src0:src and videoconvert0:sink
0:00:00.408864850  4846  0x129de00 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<videoconvert0:src> pad has no peer
0:00:00.411590148  4846  0x129de00 INFO                GST_PADS gstpad.c:2523:gst_pad_link_full: linked v4l2src0:src and videoconvert0:sink, successful
0:00:00.411633221  4846  0x129de00 INFO               GST_EVENT gstevent.c:1511:gst_event_new_reconfigure: creating reconfigure event
0:00:00.411665200  4846  0x129de00 INFO               GST_EVENT gstpad.c:5652:gst_pad_send_event_unchecked:<v4l2src0:src> Received event on flushing pad. Discarding
0:00:00.411717335  4846  0x129de00 INFO            GST_PIPELINE grammar.y:596:gst_parse_perform_link: linking some pad of GstVideoConvert named videoconvert0 to some pad of GstVideoScale named videoscale0 (0/0) with caps "(NULL)"
0:00:00.411769730  4846  0x129de00 INFO        GST_ELEMENT_PADS gstutils.c:1633:gst_element_link_pads_full: trying to link element videoconvert0:(any) to element videoscale0:(any)
0:00:00.411813168  4846  0x129de00 INFO                GST_PADS gstutils.c:933:gst_pad_check_link: trying to link videoconvert0:src and videoscale0:sink
0:00:00.413587846  4846  0x129de00 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<videoscale0:src> pad has no peer
0:00:00.418515423  4846  0x129de00 INFO                GST_PADS gstutils.c:1447:prepare_link_maybe_ghosting: videoconvert0 and videoscale0 in same bin, no need for ghost pads
0:00:00.418658808  4846  0x129de00 INFO                GST_PADS gstpad.c:2315:gst_pad_link_prepare: trying to link videoconvert0:src and videoscale0:sink
0:00:00.420594214  4846  0x129de00 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<videoscale0:src> pad has no peer
0:00:00.425453355  4846  0x129de00 INFO                GST_PADS gstpad.c:2523:gst_pad_link_full: linked videoconvert0:src and videoscale0:sink, successful
0:00:00.425511844  4846  0x129de00 INFO               GST_EVENT gstevent.c:1511:gst_event_new_reconfigure: creating reconfigure event
0:00:00.425545334  4846  0x129de00 INFO               GST_EVENT gstpad.c:5652:gst_pad_send_event_unchecked:<videoconvert0:src> Received event on flushing pad. Discarding
0:00:00.425609865  4846  0x129de00 INFO            GST_PIPELINE grammar.y:596:gst_parse_perform_link: linking some pad of GstVideoScale named videoscale0 to some pad of GstX264Enc named x264enc0 (0/0) with caps "video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)5/1"
0:00:00.427598031  4846  0x129de00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstcoreelements.so" loaded
0:00:00.427671000  4846  0x129de00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "capsfilter"
[... cut ...]
0:00:01.747527236  4846  0x129de00 INFO        GST_ELEMENT_PADS gstelement.c:785:gst_element_remove_pad:<videoconvert0> removing pad 'src'
0:00:01.747564267  4846  0x129de00 INFO         GST_REFCOUNTING gstelement.c:3040:gst_element_dispose:<videoconvert0> 0x140e260 parent class dispose
0:00:01.747597288  4846  0x129de00 INFO         GST_REFCOUNTING gstelement.c:3071:gst_element_finalize:<videoconvert0> 0x140e260 finalize
0:00:01.747626403  4846  0x129de00 INFO         GST_REFCOUNTING gstelement.c:3077:gst_element_finalize:<videoconvert0> 0x140e260 finalize parent
0:00:01.747668590  4846  0x129de00 INFO

chehefen, 2020-02-14:
libv4l2: error set_fmt gave us a different result then try_fmt!
0:00:01.299837127  4846  0x13fe8f0 WARN                 default v4l2-utils.c:189:gst_v4l2_error:<v4l2src0> error: Device '/dev/video1' cannot capture in the specified format
0:00:01.299947595  4846  0x13fe8f0 WARN                 default v4l2-utils.c:192:gst_v4l2_error:<v4l2src0> error: Tried to capture in YU12, but device returned format UYVY
0:00:01.300093948  4846  0x13fe8f0 INFO        GST_ERROR_SYSTEM gstelement.c:1890:gst_element_message_full_with_details:<v4l2src0> posting message: Device '/dev/video1' cannot capture in the specified format

Not sure what the error is about, you can try force UYVY from device

gst-launch-1.0 -v v4l2src device=/dev/video1 ! video/x-raw,format=UYVY ! videoconvert ! videoscale ! video/x-raw,width=640,height=480,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"

istranic, 2020-02-14:
> v4l2-utils.c:189:gst_v4l2_error: error: Device '/dev/video1' cannot capture
> v4l2-utils.c:192:gst_v4l2_error: error: Tried to capture in YU12, but
> gstelement.c:1890:gst_element_message_full_with_details: posting message:

Hi,

That didn't work either. It looks like I'll have to do a deep-dive to solve
this issue.

I've attached the last log, but I don't think it tells us anything new.
Thank you for your help and responsiveness!

Ivo

-- 
Ivo Stranic
Stanford University

COMMAND:
gst-launch-1.0 -v v4l2src device=/dev/video1 ! video/x-raw,format=UYVY ! videoconvert ! video/x-raw,width=160,height=120,framerate=5/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName3 storage-size=128 access-key=lll secret-key=lll aws-region="us-east-1"

LOG (PARTIAL)

1:plugin_init: x264 build: 148
0:00:00.302738283  8887  0x1144e00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstx264.so" loaded
0:00:00.302781407  8887  0x1144e00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "x264enc"
0:00:00.303539577  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x12ba5e0> adding pad 'sink'
0:00:00.303605827  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstVideoEncoder@0x12ba5e0> adding pad 'src'
0:00:00.374814136  8887  0x1144e00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/home/pi/Kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libgstkvssink.so" loaded
0:00:00.374908042  8887  0x1144e00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "kvssink"
0:00:00.375767826  8887  0x1144e00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "pipeline"
0:00:00.376122926  8887  0x1144e00 INFO            GST_PIPELINE grammar.y:596:gst_parse_perform_link: linking some pad of GstV4l2Src named v4l2src0 to some pad of GstVideoConvert named videoconvert0 (0/0) with caps "video/x-raw, format=(string)UYVY"
0:00:00.377906035  8887  0x1144e00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstcoreelements.so" loaded
0:00:00.377968587  8887  0x1144e00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "capsfilter"
0:00:00.378210407  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseTransform@0x12ea148> adding pad 'sink'
0:00:00.378272438  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseTransform@0x12ea148> adding pad 'src'
0:00:00.378326448  8887  0x1144e00 INFO              GST_STATES gstbin.c:2110:gst_bin_get_state_func:<pipeline0> getting state
0:00:00.378408218  8887  0x1144e00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<capsfilter0> completed state change to NULL
0:00:00.378456186  8887  0x1144e00 INFO               GST_EVENT gstevent.c:1511:gst_event_new_reconfigure: creating reconfigure event
0:00:00.378492488  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstutils.c:1633:gst_element_link_pads_full: trying to link element v4l2src0:(any) to element capsfilter0:sink
0:00:00.378531602  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstelement.c:916:gst_element_get_static_pad: found pad capsfilter0:sink
0:00:00.378563581  8887  0x1144e00 INFO                GST_PADS gstutils.c:1447:prepare_link_maybe_ghosting: v4l2src0 and capsfilter0 in same bin, no need for ghost pads
0:00:00.378615248  8887  0x1144e00 INFO                GST_PADS gstpad.c:2315:gst_pad_link_prepare: trying to link v4l2src0:src and capsfilter0:sink
0:00:00.378679934  8887  0x1144e00 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<capsfilter0:src> pad has no peer
0:00:00.378734517  8887  0x1144e00 INFO                GST_PADS gstpad.c:2523:gst_pad_link_full: linked v4l2src0:src and capsfilter0:sink, successful
0:00:00.378768475  8887  0x1144e00 INFO               GST_EVENT gstevent.c:1511:gst_event_new_reconfigure: creating reconfigure event
0:00:00.378796027  8887  0x1144e00 INFO               GST_EVENT gstpad.c:5652:gst_pad_send_event_unchecked:<v4l2src0:src> Received event on flushing pad. Discarding
0:00:00.378840402  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstutils.c:1633:gst_element_link_pads_full: trying to link element capsfilter0:src to element videoconvert0:(any)
0:00:00.378875401  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstelement.c:916:gst_element_get_static_pad: found pad capsfilter0:src
0:00:00.378910141  8887  0x1144e00 INFO                GST_PADS gstutils.c:933:gst_pad_check_link: trying to link capsfilter0:src and videoconvert0:sink
0:00:00.378995817  8887  0x1144e00 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<videoconvert0:src> pad has no peer
0:00:00.381793760  8887  0x1144e00 INFO                GST_PADS gstutils.c:1447:prepare_link_maybe_ghosting: capsfilter0 and videoconvert0 in same bin, no need for ghost pads
0:00:00.381848551  8887  0x1144e00 INFO                GST_PADS gstpad.c:2315:gst_pad_link_prepare: trying to link capsfilter0:src and videoconvert0:sink
0:00:00.381928186  8887  0x1144e00 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<videoconvert0:src> pad has no peer
[... cut ...]
0:00:01.635857385  8887  0x1144e00 INFO         GST_REFCOUNTING gstelement.c:2994:gst_element_dispose:<videoconvert0> 0x12b4260 dispose
0:00:01.635926447  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstelement.c:785:gst_element_remove_pad:<videoconvert0> removing pad 'sink'
0:00:01.636010144  8887  0x1144e00 INFO        GST_ELEMENT_PADS gstelement.c:785:gst_element_remove_pad:<videoconvert0> removing pad 'src'
0:00:01.636093269  8887  0x1144e00 INFO         GST_REFCOUNTING gstelement.c:3040:gst_element_dispose:<videoconvert0> 0x12b4260 parent class dispose
0:00:01.636168945  88

MushMal, 2020-02-14:
It might be something to do with the color format. From the gstreamer log I see
0:00:01.204509337  8887  0x12a5c90 INFO                 x264enc gstx264enc.c:561:gst_x264_enc_add_x264_chroma_format: This x264 build supports 8-bit depth

Perhaps GStreamer forums are a better choice to get help on encoder/media source/color format issues.

I am going to resolve this issue for now. Please open a separate issue if you have more questions regarding KVS.

closed 2020-02-14 by MushMal

istranic, 2020-02-15:
@chehefen @MushMal 

I am able to create the stream when I delete the "video/x-raw, format...,width...., framerate..." after "videoconvert !" :

gst-launch-1.0 -v v4l2src device=/dev/video1 ! video/x-raw,format=GRAY8 ! videoconvert ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"

The command above works with either GRAY8, GRAY16_LE, or UYVY as the output format from the device.

However, that means I'm not controlling the framerate or output format. Does this shed light on the issue. I'd like to be able to set the framerate.

istranic, 2020-02-15:
Update,

I'm able to run gst-launch-1.0 -v v4l2src device=/dev/video1 ! video/x-raw,format=GRAY8 ! videoconvert ! video/x-raw, framerate=9/1 ! x264enc bframes=0 key-int-max=5 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"

I think the issue was that the camera output framerate is 9 FPS, so it must be set to that number. However, I thought that videoconvert would allow me to convert to a number that is lower.

istranic, 2020-02-15:
Final update. Adding ! videorate ! enables me to change the framerate.

Aman-Haris, 2022-01-10:
Hi @istranic, I came across this comment thread as I was researching for a way to combine RGB and IR camera feed into one single feed by converting them to a grayscale format. I am doing my bachelor's final year project where I'm using YOLOR object detection algorithm to detect vehicles during bad weather conditions. I'm interested to know whether you have completed your objective and if yes, I would be happy if you are ready to help me out.

istranic, 2022-01-10:
Hey @Aman-Haris. I got this to work, so I'm happy to chat. Can you ping me on [LinkedIn](https://www.linkedin.com/in/ivo-stranic-07b73b28) and we can go from there.