awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #344: Cannot install on Raspberry Pi 4

daaru00, 2020-02-15:
Hello,

I'm following the documentation at https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-cpp-rpi.html, but when I try to execute the step:
```bash
./install-script
```
script ends with message:
```
install-script takes more than 10 hours to complete on raspberry pi. Please use min-install-script instead.
```

So I try to execute script `./min-install-script` but lot of dependencies are missing during cmake check, for example libgstreamer-plugins-bad1.0-dev, libgstreamer1.0-dev, libglib2.0-dev, libcurl4-openssl-dev and so on. 

Now I'm stuck on this error:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LOG4CPLUS_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
   used as include directory in directory /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build

-- Configuring incomplete, errors occurred!
See also "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/CMakeFiles/CMakeError.log".
```

In `CMakeError.log` log file I found a compile error:
```
Run Build Command:"/usr/bin/make" "cmTC_f5e30/fast"
/usr/bin/make -f CMakeFiles/cmTC_f5e30.dir/build.make CMakeFiles/cmTC_f5e30.dir/build
make[1]: Entering directory '/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_f5e30.dir/CheckSymbolExists.c.o
/usr/bin/cc   -fPIC    -o CMakeFiles/cmTC_f5e30.dir/CheckSymbolExists.c.o   -c /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_f5e30
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_f5e30.dir/link.txt --verbose=1
/usr/bin/cc  -fPIC     -rdynamic CMakeFiles/cmTC_f5e30.dir/CheckSymbolExists.c.o  -o cmTC_f5e30 
/usr/bin/ld: CMakeFiles/cmTC_f5e30.dir/CheckSymbolExists.c.o: in function `main':
CheckSymbolExists.c:(.text+0x48): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_f5e30.dir/build.make:87: cmTC_f5e30] Error 1
make[1]: Leaving directory '/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_f5e30/fast] Error 2
```

How can I complete the install steps?

chehefen, 2020-02-17:
sorry the doc there is outdated, can you follow the steps here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#install-steps-for-ubuntu-17x-and-raspbian-stretch-using-apt-get. thanks

daaru00, 2020-02-23:
Hi @chehefen,

thank you, this help me a lot, I complete the installation using `./min-install-script` script and it ends succesfully:
```
[100%] Linking CXX executable kinesis_video_gstreamer_sample_multistream_app
[100%] Built target kinesis_video_gstreamer_sample_multistream_app
**********************************************************
Success!!!
**********************************************************
```

I still have a problem running `gst-launch-1.0`:
```
$ gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,width=640,height=480,framerate=30/1,format=I420 ! omxh264enc periodicty-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name="RaspberryPiCamera" access-key="xxxxxxxxxxxxxxxxxxxx" secret-key="xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx" alsasrc device=hw:1,0 ! audioconvert ! avenc_aac ! queue ! sink
WARNING: erroneous pipeline: no element "kvssink"
```
seems `kvssink` plugin is not installed/compiled/downloaded.

> Note that the install-script also builds the Kinesis Video Streams producer SDK as a GStreamer plugin (kvssink).

I executed `./min-install-script` and not `./install-script` (because I cannot execute the last one on RaspberryPi), maybe some plugins are missing.

Don't know if related but inside `amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local` I cannot find a `bin` directory, just `include` and `lib`.

zhiyua-git, 2020-02-23:
Hi @daaru00 ,

Have you set the GST_PLUGIN_PATH and LD_LIBRARY_PATH as mentioned in readme?

daaru00, 2020-02-23:
Hi @zhiyua-git,

yep, set both environment variables:
```
export GST_PLUGIN_PATH=/opt/awssdk/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib:$GST_PLUGIN_PATH
export LD_LIBRARY_PATH=/opt/awssdk/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib:$LD_LIBRARY_PATH
```
(I downloaded the repository in `/opt/awssdk/`).

I think the problem is the missing `kinesis-video-native-build/downloads/local/lib` directory.

---

Running `./kinesis_video_gstreamer_sample_app` mentioned [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#running-the-gstreamer-webcam-sample-application) it start working!! I receiving the stream and I can see the Media playback from the console! :tada:

zhiyua-git, 2020-02-23:
Hi @daaru00 ,

Can you set it to below? These should be correct path for min-install-script.

export GST_PLUGIN_PATH=/opt/awssdk/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build:$GST_PLUGIN_PATH
export LD_LIBRARY_PATH=/opt/awssdk/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build:$LD_LIBRARY_PATH

daaru00, 2020-02-23:
Hi @zhiyua-git,

ok, setting those path the `kvssink` warning disappear.

Now I have an other issue:
```
WARNING: erroneous pipeline: no element "alsasrc"
```

I tried the solution posted in [doc](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#when-kinesis_video_gstreamer_audio_video_sample_app-failed-with-warning-erroneous-pipeline-no-element-alsasrc):

- installed missing package `libasound2-dev`
- skipped deletion of file `downloads/local/lib/libgstvideo-1.0.so` because it does not exist (only one file is present: `libjsmn.a`)
- executed `./min-install-script` script again (documentation refers to `install-script` so I supposed to use `min-install-script` on raspberry) 

does fix it, warning is still there.

chehefen, 2020-03-03:
Hi @daaru00, sorry the README is outdated, can you run 
`sudo apt-get reinstall libgstreamer-plugins-base1.0-dev` and then do `gst-inspect-1.0 | grep alsasrc` to see if alsasrc get installed?

zhiyua-git, 2020-03-16:
Closing as the issue is fixed, need to update documents are action item.

closed 2020-03-16 by zhiyua-git