awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #349: how to set resolution width & height when using ./kinesis_video_gstreamer_audio_video_sample_app

mistersingh179, 2020-02-20:
Hi, how can i specify the width, height &  fps at which i want the video to be recorded when I am using the sample app `kinesis_video_gstreamer_audio_video_sample_app`.

I tried passing in the -w, -h & -f options, but they didnt work.

I also tried doing `v4l2-ctl --set-fmt-video=width=1280,height=720` to change the cameras default, but even that didn't change the resolution at which the sample app records. 

The sample app always records at 800x600.

zhiyua-git, 2020-02-20:
There is no -w -h -f built-in for audio video gstreamer sample now, you can set frame rate through https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_audio_video_sample_app.cpp#L826 and add width/height constraint to it similar to https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_audio_video_sample_app.cpp#L876. Thanks.

mistersingh179, 2020-02-20:
hi @zhiyua-git I am running on a raspberry-pi, and i see that 876 is for mac. would that still help me when i am running from a pi?
also i have been using the files under `kinesis-video-native-build` folder. After I make changes to the cpp file in `kinesis-video-gst-demo`, then how would i build the executable?

thanks

zhiyua-git, 2020-02-20:
Hi @mistersingh179 ,

You can modify L826 to video_caps_string = "video/x-raw, framerate=(fraction) [ 20/1, 35/1 ], format=I420, width=(int) yourWidth, height=(int) yourHeight"; if you are on RaspberryPi. After modifying the file in kinesis-video-gst-demo, run ./min-install-script or make under kinesis-video-native-build can help you build with the change. Thanks.

mistersingh179, 2020-02-20:
Hi @zhiyua-git 

I changes it to:
```
video_caps_string = "video/x-raw, framerate=(fraction) [ 10/1, 15/2, 5/1 ], format=I420, width=(int) 1280, height=(int) 720";
```

Then i ran `./min-install-script` followed by `./kinesis_video_gstreamer_audio_video_sample_app test`

Now, it fails to stream the video. The error it prints:

> (kinesis_video_gstreamer_audio_video_sample_app:7137): GStreamer-CRITICAL **: 21:39:01.337: gst_mini_object_unref: assertion 'mini_object != NULL' failed

Also, I confirmed with `gst-device-monitor-1.0` that the resolution I put in of 1280x720 is supported by my camera

mistersingh179, 2020-02-20:
@zhiyua-git 

interesting that even reverting the code and rebuilding breaks it and it doesnt stream the video any more. it just hangs when i run `./kinesis_video_gstreamer_audio_video_sample_app test`

This would mean that the issue is with my build/make process.

chehefen, 2020-02-20:
you video_caps_string should be video_caps_string = "video/x-raw,framerate=15/2, format=I420, width=1280, height=720"

mistersingh179, 2020-02-20:
thank you @chehefen & @zhiyua-git – this fixed the issue.

` video_caps_string = "video/x-raw,framerate=15/2, format=I420, width=1280, height=720";`

closed 2020-02-20 by mistersingh179