awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #350: Erroneous pipeline: no element "kvssink"

wrigsa, 2020-02-20:
Hello,

I'm fairly certain that I've properly set the environment variables, but when I try to run the following command,

gst-launch-1.0 ksvideosrc ! videoconvert ! video/x-raw,format=I420,width=640,height=480 ! x264enc bframes=0 key-int-max=45 bitrate=512 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="{myStreamName}" storage-size=512 access-key="{myAccessKey}" secret-key="{mySecretKey}" aws-region="us-east-1"

all I get back is: WARNING: erroneous pipeline: no element "kvssink"

any help would be much appreciated.

disa6302, 2020-02-20:
Hello @wrigsa , 

1. Do you mind sharing the environment variables you set? 
2. Are you using min-install-script or install-script?

wrigsa, 2020-02-20:
1. I used export GST_PLUGIN_PATH=PWD and export LD_LIBRARY_PATH=PWD:$LD_LIBRARY_PATH from inside kinesis-video-native-build 

2. I'm using min-install-script

disa6302, 2020-02-20:
Is libgstkvssink.so located in the native build directory? Because, the error you are seeing pops up when it cannot locate this specific library.

disa6302, 2020-02-20:
Can you verify libgstkvssink.so is in the native-build directory? 

Because, the error you are seeing happens when this library is not located.

wrigsa, 2020-02-20:
Well, that would be it. Thank you!

wrigsa, 2020-02-20:
I can't find that file anywhere. How can I get it?

disa6302, 2020-02-20:
@wrigsa ,

min-install-script would install it.

To help you out a little, ensure you have done the following:
1. Installed pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav (either using sudo apt-get or brew or in mungw based on platform)
2. Run ./min-install-script in native build directory
3. export GST_PLUGIN_PATH=/path/to/libgstkvssink.so

At the end of the min-install-script run, you should see the following:

[ 98%] Linking CXX shared module **libgstkvssink.so**
[ 98%] Built target gstkvssink
[100%] Linking CXX executable kinesis_video_gstreamer_sample_app
[100%] Built target kinesis_video_gstreamer_sample_app
**********************************************************
Success!!!
**********************************************************

zhiyua-git, 2020-02-20:
Hi @wrigsa ,

Can you check if you are using wrong environment variables? You need export GST_PLUGIN_PATH=$PWD not PWD. Can you paste the output of echo $GST_PLUGIN_PATH so we can tell if that is the issue? Thanks.

wrigsa, 2020-02-20:
@zhiyua-git this is the output: C:/msys64/home/sarah/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build 

I think I'm just having trouble installing gstreamer

disa6302, 2020-02-20:
@wrigsa , once your gstreamer is installed correctly, you should be able to see the library after running ./min-install-script. Let us know if you have any other questions!

chehefen, 2020-02-20:
@wrigsa do you mind changing the repo name from amazon-kinesis-video-streams-producer-sdk-cpp to something shorter like kvssdk. On msys2 there were issue about path name being too long.

wrigsa, 2020-02-20:
I've changed the path name and made sure everything is installed, but it still quits at 85%. These are the last few lines after I run ./min-install-script  

[ 85%] Linking CXX shared library libproducer.dll
make[2]: Leaving directory '/home/sarah/kvssdk/kinesis-video-native-build'
[ 85%] Built target producer
make[1]: Leaving directory '/home/sarah/kvssdk/kinesis-video-native-build'
make: *** [Makefile:84: all] Error 2

Thank you guys for all of your help

MushMal, 2020-02-20:
Could you please include more info? It doesn't say anything about the error it encountered (assuming this starts off from a clean repo)

francoisdutoit, 2020-02-29:
@MushMal 

I'm getting the same error - although my build succeeds...

@wrigsa - any luck on your side?

[image]
[image]

pi@raspberrypi:~/Documents/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ **export LD_LIBRARY_PATH=/home/pi/Documents/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build:$LD_LIBRARY_PATH**

pi@raspberrypi:~/Documents/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ **export GST_PLUGIN_PATH=/home/pi/Documents/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build:$GST_PLUGIN_PATH**

pi@raspberrypi:~/Documents/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ gst-launch-1.0 rtspsrc location="rtsp://192.168.88.108:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif" short-header=FALSE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=streamone storage-size=512 aws-region="us-east-1"
**WARNING: erroneous pipeline: no element "kvssink"**

zhiyua-git, 2020-02-29:
Hi @francoisdutoit ,

You GST_PLUGIN_PATH is pointing to wrong directory amazon-kinesis-video-streams-producer-sdk-cpp **-master**, run export GST_PLUGIN_PATH=/home/pi/Documents/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build:$GST_PLUGIN_PATH and see if it works. Thanks.

francoisdutoit, 2020-03-01:
@zhiyua-git - thanks for the reply - but setting the GST_PLUGIN_PATH to kinesis-video-native-build - did not work. What I did to get it to work was to copy libcproducer.so, libproducer.so and libgstkvssink.so to /usr/lib/arm-linux-gnueabihf/gstreamer-1.0 and then setting my paths as below...

export LD_LIBRARY_PATH=/usr/lib/arm-linux-gnueabihf/gstreamer-1.0:$LD_LIBRARY_PATH
export GST_PLUGIN_PATH=/usr/lib/arm-linux-gnueabihf/gstreamer-1.0:$GST_PLUGIN_PATH

I'm able to stream h264 now without any issues. :)

disa6302, 2020-03-04:
@francoisdutoit , 

Glad it worked! 
Closing this ticket based on your response. Feel free to open a new issue if you face any issues in the future.

closed 2020-03-04 by disa6302