awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #351: Clarification for gstreamer + IOT Credentials

istranic, 2020-02-20:
For connecting to kvs using IOT credentials, can someone please clarify the elements of the iot-certificate?

gst-launch-1.0 -v rtspsrc location="rtsp://YourCameraRtspUrl" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au !
 h264parse ! kvssink stream-name="iot-stream" iot-certificate="iot-certificate,endpoint=endpoint,cert-path=/path/to/certificate,key-path=/path/to/private/key,ca-path=/path/to/ca-cert,role-aliases=role-aliases"

1. What is the "endpoint". Is the ARN of the IOT Thing?
2. What is "role-aliases"?

zhiyua-git, 2020-02-20:
Hi @istranic ,

You can refer to https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html for the information. Please note the endpoint is authentication endpoint for IoT, not the IoT endpoint on AWS Console.

MushMal, 2020-02-20:
Resolving as it seems answered. Please re-open if needed

istranic, 2020-03-19:
Hi @MushMal. I've tried using the instructions you provided, but I'm having some issues (Running Windows 10).

When I 
aws iam put-role-policy --role-name KVSCameraCertificateBasedIAMRole --policy-name KVSCameraIAMPolicy --policy-document file://iam-permission-document.json

I get the error "An error occurred (MalformedPolicyDocument) when calling the PutRolePolicy operation: Syntax errors in policy."

I don't think it's caused by the formating, quotes, etc.,, because the previous step work just fine:

aws iam create-role --role-name KVSCameraCertificateBasedIAMRole --assume-role-policy-document file://iam-policy-document.json > iam-role.json

iam-permission-document.json looks like this:

{
"Version": "2012-10-17",
"Statement": [
{
"Effect": "Allow",
"Action": [
"kinesisvideo:DescribeStream",
"kinesisvideo:PutMedia",
"kinesisvideo:TagStream",
"kinesisvideo:GetDataEndpoint"
],
"Resource": "arn:aws:kinesisvideo:::stream/${credentials-iot:ThingName}/*"
}
]
}

When I try to create the permission in the AWS web interface, it says that the following line is a bad string:
"Resource": "arn:aws:kinesisvideo:::stream/${credentials-iot:ThingName}/*"

Is there a problem with the policy document itself?

istranic, 2020-03-19:
@MushMal Quick update. I tried to run the same command on an ubuntu machine, and I get the same error:

**Command**
aws --profile default iam put-role-policy --role-name KVSCameraCertificateBasedIAMRole --policy-name KVSCameraIAMPolicy --policy-document 'file://iam-permission-document.json'

MushMal, 2020-03-19:
@istranic I see your Resource line is a little different:

This is in your file:

`"Resource": "arn:aws:kinesisvideo:::stream/${credentials-iot:ThingName}/*"`

VS what's in the doc:

`"Resource": "arn:aws:kinesisvideo:*:*:stream/\${credentials-iot:ThingName}/*"`

istranic, 2020-03-19:
@MushMal Thx for pointing that out. I think that was a typo in my old post a month ago. I just tried everything again, on Windows and Ubuntu, using the correct format. Here's the document I'm using right now.

{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": [
                "kinesisvideo:DescribeStream",
                "kinesisvideo:PutMedia",
                "kinesisvideo:TagStream",
                "kinesisvideo:GetDataEndpoint"
            ],
            "Resource": "arn:aws:kinesisvideo:*:*:stream/\${credentials-iot:ThingName}/*"
        }
    ]
}

istranic, 2020-03-19:
Oh, wow, when I paste the document in github, some of the characters change. In any case, the line is correct in my actual file.

istranic, 2020-03-20:
@MushMal Do you have any further tips for what I should try?

MushMal, 2020-03-20:
Are you replacing the placeholder in the json template?

Note that this policy authorizes the specified actions only on a video stream (AWS resource) that is specified by the placeholder (${credentials-iot:ThingName}). This placeholder takes on the value of the IoT thing attribute ThingName when the IoT credentials provider sends the video stream name in the request.

istranic, 2020-03-20:
No I didn't replace it. What would the json look like if the thing is called "ABC". It's not clear to me whether to replace "ThingName" or the whole text "${credentials-iot:ThingName}", or something else.

What would I do it I have thousands of things in a production environment? Is there a way to do this for a a Thing Type?

MushMal, 2020-03-20:
I think the documentation needs to be a little clearer

Could you try something like this 

"Resource": "arn:aws:kinesisvideo:*:*:stream/ABC/*"

MushMal, 2020-03-20:
Sorry, could you please try the following instead:

{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": [
                "kinesisvideo:DescribeStream",
                "kinesisvideo:PutMedia",
                "kinesisvideo:TagStream",
                "kinesisvideo:GetDataEndpoint"
            ],
            "Resource": "arn:aws:kinesisvideo:*:*:stream/${credentials-iot:ThingName}/*"
        }
    ]
}

istranic, 2020-03-20:
@MushMal Amazing, that worked. Will work with multiple things, or do I need to replace "ThingName" by the name of my actual thing?

MushMal, 2020-03-20:
You could use the whildcarded version. Sorry for the trouble, we will fix the documentation shortly!

I am going to resolve the issue, please feel free to open again if you have any troubles with it

closed 2020-03-20 by MushMal

istranic, 2020-03-23:
@MushMal I successfully completed steps 1-4 in the tutorial below on my Ubuntu machine. I ran everything using a thing name and stream name of "myka_01", instead of the default stream name in the tutorial
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html

I then copied all the relevant credentials onto my raspberri pi and tried to complete step 5. However, the streams hangs and produces a warning:

2020-03-23 13:18:41 [1811932256] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"Message":"User: arn:aws:sts::132756671989:assumed-role/KVSCameraCertificateBasedIAMRole/eb9fbdd055689d470222e2c339a8572e38f5988b68d07d6c60e8bf6a2b11d05b is not authorized to perform: kinesisvideo:DescribeStream on resource: myka_01"}

Any ideas what the issue could be? I'm hiding the credentials endpoint just in case, but is that necessary?

Here's the full log:

**COMMAND**
gst-launch-1.0 -v v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! videoscale ! videoscale ! videorate ! video/x-raw,format=I420,width=640,height=480,framerate=5/1 ! omxh264enc periodicty-idr=5 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=myka_01 iot-certificate="iot-certificate,endpoint=hidden_endpoint.credentials.iot.us-east-1.amazonaws.com,cert-path=/home/pi/certificate.pem,key-path=/home/pi/private.pem.key,ca-path=/home/pi/cacert.pem,role-aliases=KvsCameraIoTRoleAlias"

**LOG**
0:00:00.000558538  2240  0x1338000 INFO                GST_INIT gst.c:586:init_pre: Initializing GStreamer Core Library version 1.14.4
0:00:00.000805411  2240  0x1338000 INFO                GST_INIT gst.c:587:init_pre: Using library installed in /usr/lib/arm-linux-gnueabihf
0:00:00.001036659  2240  0x1338000 INFO                GST_INIT gst.c:607:init_pre: Linux pi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l
0:00:00.002681752  2240  0x1338000 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.006206779  2240  0x1338000 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.008353691  2240  0x1338000 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.009115717  2240  0x1338000 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.009209154  2240  0x1338000 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.009355142  2240  0x1338000 INFO            GST_REGISTRY gstregistry.c:1727:ensure_current_registry: reading registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.120497016  2240  0x1338000 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /home/pi/.cache/gstreamer-1.0/registry.arm.bin in 0.110987 seconds
0:00:00.120791441  2240  0x1338000 INFO            GST_REGISTRY gstregistry.c:1583:scan_and_update_registry: Validating plugins from registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.134557959  2240  0x1338000 INFO            GST_REGISTRY gstregistry.c:1685:scan_and_update_registry: Registry cache has not changed
0:00:00.134651083  2240  0x1338000 INFO            GST_REGISTRY gstregistry.c:1762:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.134680458  2240  0x1338000 INFO                GST_INIT gst.c:807:init_post: GLib runtime version: 2.58.3
0:00:00.134714729  2240  0x1338000 INFO                GST_INIT gst.c:809:init_post: GLib headers version: 2.58.1
0:00:00.134738010  2240  0x1338000 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
0:00:00.134903061  2240  0x1338000 INFO            GST_PIPELINE gstparse.c:337:gst_parse_launch_full: parsing pipeline description 'v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! videoscale ! videoscale ! videorate ! video/x-raw,format=I420,width=640,height=480,framerate=5/1 ! omxh264enc periodicty-idr=5 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=myka_01 iot-certificate=iot-certificate,endpoint=hidden_endpoint.credentials.iot.us-east-1.amazonaws.com,cert-path=/home/pi/Authentication/certificate.pem,key-path=/home/pi/Authentication/private.pem.key,ca-path=/home/pi/Authentication/cacert.pem,role-aliases=KvsCameraIoTRoleAlias '
0:00:00.156624158  2240  0x1338000 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideo4linux2.so" loaded
0:00:00.156693897  2240  0x1338000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "v4l2src"
0:00:00.159864708  2240  0x1338000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseSrc@0x14fe0f0> adding pad 'src'
0:00:00.160909909  2240  0x1338000 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideoconvert.so" loaded
0:00:00.160971575  2240  0x1338000 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "videoconvert"
0:00:00.161873079  2240  0x1338000 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x1506090> adding pad 'sink'
[... cut ...]
2020-03-23 13:18:41 [1811932256] DEBUG - defaultStreamErrorReportCallback(): Reported streamError callback for stream handle 22256013. Upload handle 18446744073709551615. Fragment timecode in 100ns: 0. Error status: 0x52000010
2020-03-23 13:18:41 [1811932256] DEBUG - streamErrorHandler invoked
2020-03-23 13:18:41 [1811932256] ERROR - Reported stream error. Errored timecode: 0 Status: 0x52000010
2020-03-23 13:18:41 [1811932256] WARN - continuousRetryStreamErrorReportHandler(): Reporting stream error. Errored timecode: 0 Status: 0x7585fad852000010

chehefen, 2020-03-23:
Hi @istranic can you show us the iam policy for the role that you assume? Thanks. You can also try this utility script to setup iot: 
[IotProvisioningSampleScripts.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4372265/IotProvisioningSampleScripts.zip)

istranic, 2020-03-23:
@chehefen Here's is the IAM policy.

`{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": [
                "kinesisvideo:DescribeStream",
                "kinesisvideo:PutMedia",
                "kinesisvideo:TagStream",
                "kinesisvideo:GetDataEndpoint"
            ],
            "Resource": "arn:aws:kinesisvideo:::stream/${credentials-iot:ThingName}/*"
        }
    ]
}`

What environment do you recommend for running the provisioning script?

chehefen, 2020-03-24:
mac or linux should be ok. Btw your resource policy is missing the * between colons. They should look like this
```
arn:aws:kinesisvideo:*:*:stream/\${credentials-iot:ThingName}/*
```

istranic, 2020-03-24:
@chehefen The version of my policy document was based on @MushMal 's previous comment:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/351#issuecomment-601912659

If I use the version you reccomended, I get the errors described earlier in this thread.

istranic, 2020-03-24:
I'll try the provisioning script and get back to you.

MushMal, 2020-03-24:
Just a note that the documentation has been updated! https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html

istranic, 2020-03-25:
@chehefen @MushMal I was able to provision using provisioning script you provided.

However, I also tried to go through the steps in the online documentation, and after successfully completing steps 1-4, I still get the same problem as in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/351#issuecomment-602749422

When comparing the jsons produced by the online instructions vs the provisioning script, they are not exactly the same. For example:

**iam-permission-document - PROVISIONING** 
```
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": [
                "kinesisvideo:ListStreams",
                "kinesisvideo:CreateStream"
            ],
            "Resource": "*"
        },
        {
            "Effect": "Allow",
            "Action": [
                "kinesisvideo:DescribeStream",
                "kinesisvideo:PutMedia",
                "kinesisvideo:TagStream",
                "kinesisvideo:GetDataEndpoint"
            ],
            "Resource": "arn:aws:kinesisvideo:*:*:stream/${credentials-iot:ThingName}/*"
        }
    ]
}
```

**iam-permission-document - ONLINE** 
```
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Effect": "Allow",
            "Action": [
                "kinesisvideo:DescribeStream",
                "kinesisvideo:PutMedia",
                "kinesisvideo:TagStream",
                "kinesisvideo:GetDataEndpoint"
            ],
            "Resource": "arn:aws:kinesisvideo:*:*:stream/${credentials-iot:ThingName}/*"
        }
    ]
}
```
            
I'll continue using the provisioning script, but it still bugs me that I can't get it to work with just the online documentation

istranic, 2020-03-25:
One final question. Since the iot policy is attached to a certificate, if I provision once, as long as I use the same certificates for other iot things, I shouldn't need to provision each thing separately, correct?

chehefen, 2020-03-25:
you can reuse the certificate but it is not recommended. You can create new certificates and attach the same policy such that they all assume the same role, and each certificate can only access stream with the same name as the iot thing name because of the variable substitution.

MushMal, 2020-03-25:
@istranic To expand on the re-use point. Not only we are not recommending but we are strongly not recommending due to security concerns and the correctness. 

* Security concerns: You don't want to have multiple devices without any way of differentiating them. The IoT certificate is the RoT (root of trust) for the device and as such acts as the unique fingerprint for the device. This also helps with the blast radius. Say, one of your devices gets compromised, it won't affect other devices.
* Correctness: The IoT certificate contains the Thing Name which is mapped to the Stream Name. We enforce single producer into the same timeline on the same stream.

istranic, 2020-03-25:
Got it. Thanks. This can be closed.