awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #352: How to change stream producer Timezone?

kamlesh545, 2020-02-23:
Hi,

How to change stream producer Timezone in kinesis video stream sdk? 
currently my default timezone is GMT I want to change this into IST how can i change this timezone ?
This following is my debug log of kinesis sdk 
[INFO ] [23-02-2020 16:13:02:981.826 GMT] Using region: ap-northeast-1

MushMal, 2020-02-24:
@kamlesh545 there are a couple of issues here depending on the configuration of your application. 1) If you are using frame timestamp mode configuration (default) then your media pipeline would timestamp the frames. 2) If you are using SDK timestamp mode then the SDK would call default get current time API to timestamp of the frames. By default, it's the GETTIME() macro in PIC which will depend on the architecture and platform you are using. Check utils/Time.cpp

Next, if you are using the Relative mode then the PutMedia StreamStart parameter of the backend API will be set using the same get current time API.

In this case though, it seems that you simply want the logs to display the local TZ timestamp. If so, change GMTIME()

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/utils/src/Time.c#L19

to a local time using localtime CRT api.

kamlesh545, 2020-02-25:
Hi Mushmal,

Thanks for your reply can you tell me in utils/Time.cpp file after remove GMTIME() function what fucntion should i add to get local Timestamp ?

Thanks.

disa6302, 2020-02-25:
Hello @kamlesh545 ,

You can replace the line mentioned by @MushMal  by:

formattedStrLen = (UINT32) STRFTIME(pDestBuffer, destBufferLen, formatStr, LOCALTIME(&timestampSeconds))

kamlesh545, 2020-02-25:
Hello @disa6302 ,

Thanks for your reply when i changed GMT to LOCALTIME as you mentioned and after again compile all files with ./min-install-script command i m getting compile error at this following stage
[ 85%] Linking CXX shared library libproducer.so

disa6302, 2020-02-25:
Can you try localtime in lower case? The failure is probably because LOCALTIME is not defined in our pic CommonDefs.h file.

kamlesh545, 2020-02-25:
after changed with lowercase localtime compiled successfully but after starting stream still i am getting GMT timestamp why?

zhiyua-git, 2020-02-25:
Hi @kamlesh545 ,

Seems you are using gstreamer sample which uses log4cplus, you can change this https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/2.1.0/kinesis-video-native-build/kvs_log_configuration#L8 to [%D{%d-%m-%Y %H:%M:%S:%Q %Z}] to get local time. Thanks.

disa6302, 2020-03-04:
Hello @kamlesh545 , 

Can you close this ticket if the issue is resolved? 

Thanks!

disa6302, 2020-03-09:
Closing this for now. @kamlesh545 , feel free to create a new issue if you face any problems in the future.

closed 2020-03-09 by disa6302