awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #353: SSL connect error: error:04067084:rsa routines:rsa_ossl_public_decrypt:data too large for modulus

Remainfox, 2020-02-24:
T31X Mips Arch Embedded System , Linux Ingenic-uc1_1 3.10.14
[root@Ingenic-uc1_1:kvs]# ./kvsVideoOnlyStreamingSample my-stream 200
2020-02-24 07:10:40 INFO    heapInitialize(): Initializing native heap with limit size 3145728, spill ratio 0% and flags 0x00000001
2020-02-24 07:10:40 INFO    heapInitialize(): Creating AIV heap.
2020-02-24 07:10:40 INFO    heapInitialize(): Heap is initialized OK
2020-02-24 07:10:40 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-02-24 07:10:40 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2020-02-24 07:10:40 INFO    createDeviceResultEvent(): Create device result event.
2020-02-24 07:10:40 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-02-24 07:10:40 INFO    createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-02-24 07:10:40 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-02-24 07:10:40 WARN    curlCompleteSync(): curl perform failed for url https://kinesisvideo.ap-southeast-1.amazonaws.com/describeStream with result SSL connect error: error:04067084:rsa routines:rsa_ossl_public_decrypt:data too large for modulus
2020-02-24 07:10:40 WARN    curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.ap-southeast-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAXTTC7SKIRJTLJLXI/20200224/ap-southeast-1/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=c39bd6df9c94ca6561140f1ee172d70c9ce098caa894b8334d341752004a82e0
    content-length: 30
    content-type: application/json
    host: kinesisvideo.ap-southeast-1.amazonaws.com
    user-agent: AWS-SDK-KVS/2.1.0 GCC/4.7.2 Linux/3.10.14__isvp_swan_1.0__ mips
    X-Amz-Date: 20200224T071040Z
    x-amz-security-token: FwoGZXIvYXdzEHIaDGvtR39wm3LnrDWz7SKYAsCaD3zY3batzQsNcEWPa+iVNFKOYmG+4LYT4Wal/gUzmPeS/PDu5xcNtEkI/AKIpUzD/Bg4vTM0zHvA6w8qARQRFxMIAB8AqW1ryWGEp1zoOxeF6BXyHgx9nbqpSfLZjpRBNEx6uicq0m74GysHiSyxDVL0FBlB+eKC/67C/qqNlhpMXhS/EkxPGU/1J4+aP30C1o7nO5XUQRc/VifKOWjrs6FJU9SW882ZanYSOBXjicrAqgtbKrw+9JE0Ca2X1JQ2K0C3prTrSYRw72OnIBWQVhQ0/0DP661BMTPpu2Ikz1dRGBf+AHzdV+6+OGTggwLsJW39yJEFk6DtWOb9nLbxbjCiTF/2AGpcmCsNJurvsnO3oCC64joo5ai+8gUyLfLCS+7Z8rcri2/vIngD65mIbmsboan/B4F1sI4JWWFv6gqbgk+Vsn/EPL8UsA==
2020-02-24 07:10:40 DEBUG   describeStreamCurlHandler(): DescribeStream API response: 
2020-02-24 07:10:40 INFO    describeStreamResultEvent(): Describe stream result event.
2020-02-24 07:10:41 WARN    curlCompleteSync(): curl perform failed for url https://kinesisvideo.ap-southeast-1.amazonaws.com/describeStream with result SSL connect error: error:04067084:rsa routines:rsa_ossl_public_decrypt:data too large for modulus
2020-02-24 07:10:41 WARN    curlCompleteSync(): HTTP Error 0 : Response: (null)

MushMal, 2020-02-24:
Are you manually cross-compiling libCurl?

This is what came up in openssl forums: 

https://github.com/curl/curl/issues/2819
https://github.com/openssl/openssl/issues/6841

Remainfox, 2020-02-25:
Thanks for answer

Remainfox, 2020-02-25:
Thanks for answer, i cross-compiling libcurl  
	CROSS_COMPILE_PREFIX=/opt/mips-gcc472-glibc216-64bit/bin/mips-linux-gnu
	GCC=${CROSS_COMPILE_PREFIX}-gcc
	CPP=${CROSS_COMPILE_PREFIX}-g++
	HOST="--host=mips-linux"
./configure $HOST --prefix=$DOWNLOADS/local/ --enable-static=yes --enable-shared=no    --disable-rtsp --disable-ldap --without-zlib --enable-hidden-symbols --with-ssl=$DOWNLOADS/local  --with-ca-bundle=/media/mmcblk0p1/kvs/cert.pem

Remainfox, 2020-02-26:
Hi MushMal 
    The link you posted after the reference still won't solve my problem

MushMal, 2020-02-26:
@Remainfox I don't yet have anything off the top of my head. We need to look into enabling cross-compile and see if we hits this same issue. I see that you are using T31X Mips Arch - do you have the full name of the platform - perhaps we might have it and could try it locally.

Could you also try ldd <your kvs library> to see what libSsl, libCrypto is libCurl linked against - whether it's platform or your pre-built one.

closed 2020-02-28 by Remainfox