awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #354: sample app kinesis_video_gstreamer_audio_video_sample_app leads to Failed to submit frame to Kinesis Video client. status: 0x52000085

mistersingh179, 2020-02-24:
Hi, I am using the sample app to send audio and video to a KVS. 

```
./kinesis_video_gstreamer_audio_video_sample_app test
```

This seems to work for a while and then after some time it stops streaming. The logs have:

```
[2020-02-24 16:01:30]-WARN-putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 60952420000 presentation timestamp: 60952420000
```

I am running on a RaspberryPi 4 and using a Logitech C920e to record both audio and video.

Could you suggest as to what this error means and how to address it?

**status: 0x52000085**

MushMal, 2020-02-24:
@mistersingh179 the error you see is defined here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L185

Unfortunately, the public AWS docs have not been updated to reflect this.

The error happens in multi-track mode when the PIC is attempting to AV/sync the frames. It has to hold some of the frames from one track so the other track can 'catch up' to allow the frames from this track to proceed. This error indicates that one of the tracks does not produce frames and as a result the frames from the other track (the one that's putKinesisVideoFrame is called on) exceeds the limit. 

Normally, this happens when the video frames get delayed.

Could you enable the verbose logging, collect the logs and attach them?

mistersingh179, 2020-02-24:
hi @MushMal I just re-ran it with `GST_DEBUG=4`. can you confirm that this is what you meant with –  enabling verbose logging?

Also, this is hard to reproduce as it doesn't fail right away, but only after I let it run for many hours.

What would you suggest I do to prevent it from completely failing. Right now, when it hits this error it never recovers and I have to restart the process to resolve it. Is it possible to have it flush the old frames and just stream new data?

I do have the following set to get higher resolution. Could this be contributing to the problem?

```
video_caps_string = "video/x-raw,framerate=10/1, format=I420, width=1280, height=720";
```

mistersingh179, 2020-02-24:
btw so far the logs show this:

```
[DEBUG][2020-02-24 22:33:18] fragmentAckReceivedHandler invoked
[DEBUG][2020-02-24 22:33:18] Reporting fragment ack received. Ack timecode 325833
[DEBUG][2020-02-24 22:33:18] postReadCallback(): Wrote 2348 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2020-02-24 22:33:18] postReadCallback(): Pausing CURL read for upload handle: 0
0:05:32.000177278  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:29.500000000 out ts 0:05:31.285745669
0:05:32.100203868  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:29.600000000 out ts 0:05:31.385737932
0:05:32.200219753  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:29.700000000 out ts 0:05:31.485732379
0:05:32.300301009  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:29.800000000 out ts 0:05:31.595626550
0:05:32.400308172  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:29.900000000 out ts 0:05:31.695631109
0:05:32.500338021  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:30.000000000 out ts 0:05:31.795616761
0:05:32.604106770  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:30.100000000 out ts 0:05:31.895862104
0:05:32.704131228  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:30.200000000 out ts 0:05:31.995845645
0:05:32.804129520  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:30.300000000 out ts 0:05:32.095863409
0:05:32.904214941  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:30.400000000 out ts 0:05:32.195795710
```

I take it this means that when the delta is higher than 60 seconds, is when the software fails?

How can I increase this buffer from 60 to 600? Would you recommend I do that?

MushMal, 2020-02-24:
@mistersingh179 it's hard to tell whether the media pipeline will recover by itself or not - it depends on the actual encoder and the media source. 

Here is the location where you can change the value of the number of frames it stores in the temp buffer before flushing for multi-track.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/mkvgen/include/com/amazonaws/kinesis/video/mkvgen/Include.h#L134

Overflowing this buffer will cause the error you are seeing. For 30 fps this means 2 seconds drift of audio/video before this error happens.

Suggest a quick fix of increasing 60 to something more - say double it. NOTE: This will also increase the memory pressure and add latency in case this buffer starts getting full (until the frames are flushed).

There are no public methods to flush the frames in. 

Also, when collecting the logs, it would be good to collect gstreamer and KVS logs both in verbose mode. For GSTreamer GST_DEBUG=4 is good. For KVS, you could use 

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/samples/KvsVideoOnlyStreamingSample.c#L93

or depending on which asset you are using

mistersingh179, 2020-02-24:
thanks @MushMal.  i am experimenting with increasing the frame buffer count.

Can you tell me what in the logs will tell me how many frames drift i have? Is it this:

> 0:05:32.000177278  4937  0x1956c60 INFO                 v4l2src gstv4l2src.c:957:gst_v4l2src_create:<videosrc> sync to 0:05:29.500000000 out ts 0:05:31.285745669

MushMal, 2020-02-25:
I am not really familiar with the gst_v412src logging. Need to investigate this further. The important question I guess is to know whether this issue is persistent or transient and if persistent whether video frames are still being produced or not (and, if they are produced then what's the delta between the frames).

It's possible that your video source simply stopped working?

mistersingh179, 2020-02-26:
hi @MushMal i tried today with different values of frame buffer and the issue is persistent. After a while it stops streaming. The logs are below. The remainder of the logs are just the same failure message repeated.

What do you recommend my next step be?

```
[2020-02-25 16:55:23]-DEBUG-Kinesis Video client and stream metrics
	>> Overall storage byte size: 134217728
	>> Available storage byte size: 134012785
	>> Allocated storage byte size: 204943
	>> Total view allocation byte size: 576064
	>> Total streams frame rate (fps): 35222
	>> Total streams transfer rate (bps): 154352 (150 Kbps)
	>> Current view duration (ms): 991
	>> Overall view duration (ms): 9041
	>> Current view byte size: 34603
	>> Overall view byte size: 194365
	>> Current frame rate (fps): 35222.4
	>> Current transfer rate (bps): 154352 (150 Kbps)
[2020-02-25 16:55:23]-DEBUG-postReadCallback(): Wrote 34603 bytes to Kinesis Video. Upload stream handle: 7
[2020-02-25 16:55:23]-DEBUG-postReadCallback(): Pausing CURL read for upload handle: 7
[2020-02-25 16:55:23]-DEBUG-postWriteCallback(): Curl post body write function for stream with handle: test and upload handle: 7 returned: {"EventType":"RECEIVED","FragmentTimecode":1230352,"FragmentNumber":"91343852333183928753240643324634405752952168380"}

[2020-02-25 16:55:23]-DEBUG-fragmentAckReceivedHandler invoked
[2020-02-25 16:55:23]-DEBUG-Reporting fragment ack received. Ack timecode 1230352
[2020-02-25 16:55:23]-DEBUG-postWriteCallback(): Curl post body write function for stream with handle: test and upload handle: 7 returned: {"EventType":"BUFFERING","FragmentTimecode":1239393,"FragmentNumber":"91343852333183928758192403481775929094058995378"}

[2020-02-25 16:55:23]-DEBUG-fragmentAckReceivedHandler invoked
[2020-02-25 16:55:23]-DEBUG-Reporting fragment ack received. Ack timecode 1239393
[2020-02-25 16:55:23]-DEBUG-postWriteCallback(): Curl post body write function for stream with handle: test and upload handle: 7 returned: {"EventType":"PERSISTED","FragmentTimecode":1230352,"FragmentNumber":"91343852333183928753240643324634405752952168380"}

[2020-02-25 16:55:23]-DEBUG-fragmentAckReceivedHandler invoked
[2020-02-25 16:55:23]-DEBUG-Reporting fragment ack received. Ack timecode 1230352
[2020-02-25 16:55:23]-DEBUG-postReadCallback(): Wrote 4967 bytes to Kinesis Video. Upload stream handle: 7
[2020-02-25 16:55:23]-DEBUG-postReadCallback(): Pausing CURL read for upload handle: 7
[2020-02-25 16:55:28]-WARN-putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 149463080000 presentation timestamp: 149463080000
[2020-02-25 16:55:28]-WARN-putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 149463506666 presentation timestamp: 149463506666
[2020-02-25 16:55:28]-WARN-putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 149463933333 presentation timestamp: 149463933333
```

MushMal, 2020-02-26:
Seems that you are getting debug logs. Do you see the printout from putFrame:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/client/src/Stream.c#L716

This should show the trackId and we can see whether the video simply stopped all together.

mistersingh179, 2020-02-28:
@MushMal no i don't see this line in my logs ever. so would this mean that i am not logging properly?

Below is my current kvs_log_configuration. What do you suggest i change it to get the correct logs:

```
log4cplus.rootLogger=DEBUG, KvsConsoleAppender, KvsFileAppender

#log4cplus.logger.MemoryCheck=TRACE, KvsConsoleAppender

#KvsConsoleAppender:
log4cplus.appender.KvsConsoleAppender=log4cplus::ConsoleAppender
log4cplus.appender.KvsConsoleAppender.layout=log4cplus::PatternLayout
log4cplus.appender.KvsConsoleAppender.layout.ConversionPattern=[%-5p][%d] %m%n

#KvsFileAppender
log4cplus.appender.KvsFileAppender=log4cplus::DailyRollingFileAppender
log4cplus.appender.KvsFileAppender.File=./log/kvs.log
log4cplus.appender.KvsFileAppender.Schedule=HOURLY
log4cplus.appender.KvsFileAppender.CreateDirs=true
log4cplus.appender.KvsFileAppender.layout=log4cplus::PatternLayout
log4cplus.appender.KvsFileAppender.layout.ConversionPattern=[%D]-%p-%m%n
```

zhiyua-git, 2020-02-28:
Hi @mistersingh179 ,

You can rerun the app with environment variable "export DEBUG_DUMP_FRAME_INFO=TRUE". That should give you the print of frame info with timestamp and track id. Thanks.

mistersingh179, 2020-02-28:
@zhiyua-git thank you , this helps. I just re-ran it with DEBUG_DUMP_FRAME_INFO set to true. I can now see that its printing as each frame is uploaded. Will wait till it crashes and then send logs.

MushMal, 2020-02-28:
@mistersingh179 - was it crashing before? If so, could you please open a separate issue and include the crash stack/dump?

mistersingh179, 2020-02-29:
@MushMal by crash i meant to say that it leads to the error i created this ticket with:

```
[2020-02-25 16:55:28]-WARN-putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 149463080000 presentation timestamp: 149463080000
[2020-02-25 16:55:28]-WARN-putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 149463506666 presentation timestamp: 149463506666
[2020-02-25 16:55:28]-WARN-putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 149463933333 presentation timestamp: 149463933333
```

I have been running it now with `DEBUG_DUMP_FRAME_INFO=true` for a 6 hours now and so far it has not resulted in error `0x52000085`.

MushMal, 2020-03-02:
@mistersingh179 have you had a chance to get more data on this?

mistersingh179, 2020-03-02:
@MushMal i am unable to reproduce this issue anymore. this would mean an issue with either the OS or camera. Closing the issue. Thank you for your help @MushMal @zhiyua-git

closed 2020-03-02 by mistersingh179