awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #355: Outdated OpenSSL

chmod750, 2020-02-25:
Hello,
The OpenSSL version used is 1_1_0f. Would it be possible to upgrade it to 1.1.1d?
Thanks

disa6302, 2020-02-25:
Hello @chmod750 ,

The install-script will automatically install openssl1.1.1d for you.

chmod750, 2020-02-26:
> Hello @chmod750 ,
> 
> The install-script will automatically install openssl1.1.1d for you.

THanks for your reply. I noticed there is outdated usage in installation script. For example in kinesis-video-native-build/windows-install.bat:
`if not exist "openssl-OpenSSL_1_1_0f" (
    echo "Openssl doesn't exist, installing into %CURRENTDIR%"
    if not exist "openssl-1.1.0f.zip" (
        powershell -Command "[Net.ServicePointManager]::SecurityProtocol = [Net.SecurityProtocolType]::Tls12; Invoke-WebRequest https://github.com/openssl/openssl/archive/OpenSSL_1_1_0f.zip -OutFile openssl-1.1.0f.zip"
    )`

disa6302, 2020-02-27:
Ah! Understood. Good catch. Must have missed updating this while updating install-script with openssl1.1.1d. Will have a patch up soon for this. Thank you for pointing it out!

PR #359 (2020-02-27): [Issue-355] Update Windows install script to install openssl 1.1.1d
*Issue #, if available: #355*
 
*Description of changes:*
- Updated Windows install script to install openssl 1.1.1d instead of 1.1.0f
- Tested the change on 64 bit Windows 10 Enterprise system

Resolves #355 

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
closed 2020-02-27 by zhiyua-git