awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #357: Potential NAL_ADAPTATION_FLAGS issue introduced by FFMPEG decodable bitstream

farleylai, 2020-02-26:
The use case is to upload an H.264 bitstream in NALUs from a NVR to KVS.
The bitstream seems regular with SPS and PPS prepended to IDR NALUs as key frames and non-IDR NALUs as other frames.
FFMPEG based players such as ffplay can play the saved bitstream showing the pixel format as YUVJ420p without noticeable issues or warnings but when using the `cproducer` in the SDK to upload the frames with the key frame flag set properly and the combination of `NAL_ADAPTATION_FLAGS`, either `putKinesisVideoFrame()` failed with error code or the console shows `Missing codec private data error` (STATUS 400):

- NAL_ADAPTATION_FLAG_NONE => Missing codec private data error
- NAL_ADAPTATION_ANNEXB_NALS => 0x3200000d
- NAL_ADAPTATION_AVCC_NALS => 0x3200000e
- NAL_ADAPTATION_ANNEXB_CPD_NALS => Missing codec private data error
- NAL_ADAPTATION_ANNEXB_NALS |  NAL_ADAPTATION_ANNEXB_CPD_NALS => 0x3200000d

Since FFMPEG is able to decode the bitstream correctly, any idea to set the flags correctly for KVS?

disa6302, 2020-02-26:
Hello @farleylai ,

Can you dump the frames into raw bytes and post it here so that we can see where the issue might be?

farleylai, 2020-02-26:
@disa6302 

Sure, the first SPS(31 bytes)/PPS(9 bytes)/IDR frame and the 2nd non-IDR frame are attached FYI.
All NALUs are prefixed with the start code 0x00000001.

[bitstream-frames.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4257658/bitstream-frames.zip)

Let me know if you find anything wrong.

Thanks.

MushMal, 2020-02-26:
0x3200000d is indicative that the stream is not properly formatted in Annex-B. Namely, it has more than 3 zeroes which is not allowed.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/mkvgen/include/com/amazonaws/kinesis/video/mkvgen/Include.h#L41

Could you please include the raw frames you are submitting? 

Some background info. The KVS console playback and HLS/MPEG-DASH will work only on AvCC format streams with h264/h265 encoding and when the CPD (Codec Private Data) is present.

When your stream is in Annex-B format, you might need to set both of the flags (NAL_ADAPTATION_ANNEXB_NALS | NAL_ADAPTATION_ANNEXB_CPD_NALS) to get the stream adapted and it possible automatically extract the CPD (NOTE: the stream should have type set of video/h264)

MushMal, 2020-02-26:
@farleylai the frames you've attached are good. Those won't cause the issue. Please ensure you specify the correct size for the data when submitting the frames to the producer SDK.

Can you dump the frame that's causing the issue. Ensure you record the size and we can look at it

farleylai, 2020-02-26:
@MushMal 

This what I got puzzled since the same code (based on the kinesis video only sample) works with other camera bitstreams but not this particular one.
I would like to clarify:
- the default settings remain, using the `NAL_ADAPTATION_ANNEXB_NALS | NAL_ADAPTATION_ANNEXB_CPD_NALS` that leads to 0x3200000d immediately
- the saved bitstream frame is exactly the same as the one passed to `putKinesisVideoFrame()` that got the error

I did one more time and made sure the uploaded frame size is exactly the same as the saved bitstream.
and the following log messages that print the frame info including the size matching the saved raw bitstream size attached FYI.

BTW, the SDK is based on tag 2.1.0.
Let me know if you have any other insight.

```
2020-02-26 19:19:25 DEBUG   heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2020-02-26 19:19:25 DEBUG   heapInitialize(): Creating AIV heap.
2020-02-26 19:19:25 DEBUG   heapInitialize(): Heap is initialized OK
2020-02-26 19:19:25 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-02-26 19:19:25 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2020-02-26 19:19:25 DEBUG   createDeviceResultEvent(): Create device result event.
2020-02-26 19:19:25 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-02-26 19:19:25 DEBUG   createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-02-26 19:19:25 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-02-26 19:19:25 DEBUG   writeHeaderCallback(): RequestId: e321d331-0db4-4ad5-a666-37233648fe45
2020-02-26 19:19:25 DEBUG   describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.582624109213E9,"DataRetentionInHours":6,"DeviceName":"JY9757ACX1EVE7NM","KmsKeyId":"arn:aws:kms:us-east-1:734908081819:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:734908081819:stream/calstore-ucb/1582624109213","StreamName":"calstore-ucb","Version":"IIUOwnPBmov1e6IS8iT1"}}
2020-02-26 19:19:25 DEBUG   describeStreamResultEvent(): Describe stream result event.
2020-02-26 19:19:25 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2020-02-26 19:19:26 DEBUG   writeHeaderCallback(): RequestId: ceb32437-9c2e-458d-bafd-d90c385af870
2020-02-26 19:19:26 DEBUG   getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
2020-02-26 19:19:26 DEBUG   getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2020-02-26 19:19:26 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
2020-02-26 19:19:26 DEBUG   getStreamingTokenResultEvent(): Get streaming token result event.
2020-02-26 19:19:26 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-02-26 19:19:26 DEBUG   defaultStreamReadyCallback(): Reported streamReady callback for stream handle 94426580188793

pFrame.index=0
pFrame.size=205022
pFrame.flags=1
pFrame.decodingTs=15827447675999135
pFrame.presentationTs=15827447675999135

2020-02-26 19:19:27 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
2020-02-26 19:19:27 DEBUG   putStreamResultEvent(): Put stream result event.
2020-02-26 19:19:27 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
2020-02-26 19:19:27 DEBUG   putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x3200000d decoding timestamp: 15827447675999135 presentation timestamp: 15827447675999135
2020-02-26 19:19:27 DEBUG   stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 000055e166c4fe79.
2020-02-26 19:19:27 DEBUG   writeHeaderCallback(): RequestId: ece8f764-39ab-16cf-ba38-319b7f7b8d73
2020-02-26 19:19:27 DEBUG   defaultStreamClosedCallback(): Reported streamClosed callback for stream handle 94426580188793. Upload handle 0
2020-02-26 19:19:27 DEBUG   postReadCallback(): Reported end-of-stream for stream calstore-ucb. Upload handle: 0
2020-02-26 19:19:27 DEBUG   postReadCallback(): Wrote 0 bytes to Kinesis Video. Upload stream handle: 0
2020-02-26 19:19:27 DEBUG   freeKinesisVideoStream(): Freeing Kinesis Video stream.
2020-02-26 19:19:27 DEBUG   curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
2020-02-26 19:19:31 DEBUG   defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 94426580188793
2020-02-26 19:19:31 DEBUG   freeKinesisVideoClient(): Freeing Kinesis Video Client
2020-02-26 19:19:31 DEBUG   curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
2020-02-26 19:19:31 DEBUG   freeKinesisVideoClientInternal(): Total allocated memory 0
2020-02-26 19:19:31 DEBUG   curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
```

[bitstream-1st.h264.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4257778/bitstream-1st.h264.zip)

MushMal, 2020-02-26:
Hmm, this is odd. The first few bytes of the frame are:

0000 0001 274d 6028 8d68 0780 227e 5c05
b200 0003 0002 0000 0300 1e1e 2845 4000
0000 0128 ee1c **9200 0000 0001** 25b8 0000
d05f f383 62ca 023a 7ae5 2ecf fdc2 7f37

It's SPS|PPS|I-frame but what's odd is that the PPS ends with 0x00 and follows with the Annex-B start code of 3 zeroes and 1. The resulting code breaks Annex-B format. Not sure if this is valid and need to do some research. Either way, this is an edge case that we will need to handle. This might take some time. What's your timeline like? This simple case can be handled by using the 2 zero and 1 version of Annex-B start code.

farleylai, 2020-02-26:
@MushMal 

Yeah, I just figured out that too by scanning the pattern in the bitstream and compared with other camera streams. It turns out that there is no such patterns from the other camera streams but only those particular two. The expectation is the get the streams to KVS as soon as possible (hopefully by Friday).

Regarding the workaround, do you mean to have the encoder to emit start code with 0x000001? Since we have no control of the cameras/NVR, this has to be done by rewrite the whole frame. Alternatively, mutating the last byte of PPS seems to work with replacement of 0x01. Nonetheless, an official clarification would be appreciated.

MushMal, 2020-02-26:
@farleylai we should fix this properly and we can prioritize this fix but I can't give the exact timelines for now.

I think "hacking" this by removing the zero or modifying it might produce some artifacts on some of the players downstream.

I will keep you posted on this

farleylai, 2020-02-26:
Agreed!

I did some research and it seems like there should be an RBSP stop bit set to 1 in the last byte.
If the last data byte is indeed zero, an additional byte should be inserted with the MSb set to 1.
Therefore, it can be either case for the last zero byte of PPS, that is, the encoder forgets to set the stop bit despite some prior 0 data bits or insert an additional trailing byte with MSb set to 1.

Also, FFMPEG seems to deal with the situation by skipping that trailing zero byte as follows ([h264.c](https://github.com/balabit/FFmpeg/blob/a75c056386c5586894b966157341c9a4f55ce2ab/libavcodec/h264.c#L340)):

```c
/**
 * Identify the exact end of the bitstream
 * @return the length of the trailing, or 0 if damaged
 */
static int decode_rbsp_trailing(H264Context *h, const uint8_t *src)
{
    int v = *src;
    int r;

    ff_tlog(h->avctx, "rbsp trailing %X\n", v);

    for (r = 1; r < 9; r++) {
        if (v & 1)
            return r;
        v >>= 1;
    }
    return 0;
}
```

```
static int get_bit_length(H264Context *h, const uint8_t *buf,
                          const uint8_t *ptr, int dst_length,
                          int i, int next_avc)
{
    if ((h->workaround_bugs & FF_BUG_AUTODETECT) && i + 3 < next_avc &&
        buf[i]     == 0x00 && buf[i + 1] == 0x00 &&
        buf[i + 2] == 0x01 && buf[i + 3] == 0xE0)
        h->workaround_bugs |= FF_BUG_TRUNCATED;

    if (!(h->workaround_bugs & FF_BUG_TRUNCATED))
        while (dst_length > 0 && ptr[dst_length - 1] == 0)
            dst_length--;

    if (!dst_length)
        return 0;

    return 8 * dst_length - decode_rbsp_trailing(h, ptr + dst_length - 1);
}
```

MushMal, 2020-06-26:
This is being tracked as part of a hack to support broken encoders: https://github.com/awslabs/amazon-kinesis-video-streams-pic/pull/68

and this is also a dup. Resolving this one

closed 2020-06-26 by MushMal