awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #358: ./min-install-script is stuck when building on Raspberry Pi(OS is Ubuntu-MATE)

joyuiyeong, 2020-02-27:
_I'm trying to build the sdk on Raspberry Pi(OS is Ubuntu-MATE)._

I installed all dependencies following [this article](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#install-steps-for-ubuntu-17x-and-raspbian-stretch-using-apt-get)

But, if executing ./min-install-script , 

[image]

it showed above logs, and stuck there(stuck at 94%).

There is no ERROR or HINT why script is stop.

Please comment about this situation.

disa6302, 2020-02-27:
Hello @joyuiyeong , 

Can you run `git clean -ffxd` and re-run `./min-install-script` from the kinesis-video-native-build directory?

joyuiyeong, 2020-02-27:
@disa6302 Oh, Thank you.

I run the commands in your comment.

And It's WORKED!!

Building is success, but streaming is failed..

After building, run ```./kinesis_video_gstreamer_sample_app RaspberryPi2Video``` (RaspberryPi2Video is Kinesis video stream name)

Log messages are, 
[image]

log4cplus:ERROR Unable to open file: ./log/kvs.log
[INFO ] [27-02-2020 08:24:09:172.963 GMT] Using region: ap-northeast-1
[INFO ] [27-02-2020 08:24:09:173.036 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [27-02-2020 08:24:09:173.066 GMT] No session token was detected.
[INFO ] [27-02-2020 08:24:09:177.560 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [27-02-2020 08:24:09:177.780 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [27-02-2020 08:24:09:177.819 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [27-02-2020 08:24:09:177.908 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [27-02-2020 08:24:09:178.010 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [27-02-2020 08:24:09:178.076 GMT] getSecurityTokenHandler invoked
[DEBUG] [27-02-2020 08:24:09:178.158 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1582791849178127240 Expiration: 0
[INFO ] [27-02-2020 08:24:09:178.194 GMT] New credentials expiration is 1582794249
[DEBUG] [27-02-2020 08:24:09:178.266 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [27-02-2020 08:24:09:178.311 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [27-02-2020 08:24:09:178.432 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [27-02-2020 08:24:09:178.504 GMT] clientReadyHandler invoked
[DEBUG] [27-02-2020 08:24:09:178.569 GMT] Client is ready
[INFO ] [27-02-2020 08:24:09:178.671 GMT] Creating Kinesis Video Stream RaspberryPi2Video
[INFO ] [27-02-2020 08:24:09:178.726 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [27-02-2020 08:24:09:178.972 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [27-02-2020 08:24:09:571.230 GMT] writeHeaderCallback(): RequestId: bda37913-0193-4ba5-8120-aa47afe22e3b
[DEBUG] [27-02-2020 08:24:09:571.551 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.582684511009E9,"DataRetentionInHours":24,"DeviceName":null,"KmsKeyId":"arn:aws:kms:ap-northeast-1:100825101287:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-northeast-1:100825101287:stream/RaspberryPi2Video/1582684511009","StreamName":"RaspberryPi2Video","Version":"vp4AfIC7f1iHQD4KBQPY"}}
[INFO ] [27-02-2020 08:24:09:580.522 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [27-02-2020 08:24:09:580.667 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [27-02-2020 08:24:09:580.704 GMT] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [27-02-2020 08:24:09:580.749 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [27-02-2020 08:24:09:817.747 GMT] writeHeaderCallback(): RequestId: 49b35f17-b8cd-4b94-8649-1ed018881556
[DEBUG] [27-02-2020 08:24:09:818.006 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-96e4f3e9.kinesisvideo.ap-northeast-1.amazonaws.com"}
[INFO ] [27-02-2020 08:24:09:826.918 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [27-02-2020 08:24:09:827.130 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [27-02-2020 08:24:09:827.217 GMT] getStreamingTokenHandler invoked
[DEBUG] [27-02-2020 08:24:09:827.289 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1582791849827273949 Expiration: 1582794249
[INFO ] [27-02-2020 08:24:09:827.355 GMT] New credentials expiration is 1582794249
[INFO ] [27-02-2020 08:24:09:827.461 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [27-02-2020 08:24:09:827.557 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [27-02-2020 08:24:09:827.678 GMT] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 187650985848137
[DEBUG] [27-02-2020 08:24:09:827.738 GMT] streamReadyHandler invoked
[DEBUG] [27-02-2020 08:24:09:827.971 GMT] Stream is ready
[INFO ] [27-02-2020 08:24:09:882.870 GMT] Streaming from live source
[DEBUG] [27-02-2020 08:24:09:883.461 GMT] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512
Unable to set the source to ready state.
[INFO ] [27-02-2020 08:24:09:910.673 GMT] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 0000aaaae65a6d49.
[DEBUG] [27-02-2020 08:24:09:910.802 GMT] defaultStreamClosedCallback(): Reported streamClosed callback for stream handle 187650985848137. Upload handle 18446744073709551615
[DEBUG] [27-02-2020 08:24:09:910.846 GMT] streamClosedHandler invoked for upload handle: 18446744073709551615
[... cut ...]
[INFO ] [27-02-2020 08:24:10:311.961 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[WARN ] [27-02-2020 08:24:10:312.073 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [27-02-2020 08:24:10:712.637 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [27-02-2020 08:24:10:712.859 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
log4cplus:ERROR Failed to open file ./log/kvs.log

I also setting all Environment varialbles..

MushMal, 2020-02-27:
@joyuiyeong seems like the GStreamer is unable to get the video source to a ready state. 

Could you please run the troubleshooting steps highlighted here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#raspberry-pi-failure-to-load-the-camera-device

disa6302, 2020-02-27:
Additionally, you can also do `export GST_DEBUG=4` to get Gstreamer logs and understand what might be going on in the pipeline.

joyuiyeong, 2020-03-03:
Hey, guys Sorry for late..

I try to build again and fail to building..
(It stuck like above.. I think the cause is that running **Ubuntu-MATE on Raspberry Pi 3**)

Sometimes, building is success and sometimes building is fail.
All of Two case takes times 30 minutes ~ 1 hour
(Stuck at some point and some case went to next builing step, but some case did nothing and I pressed ctrl+C to get out)

NOW,  I'm trying to stream video using Docker Container of Gstreamer Plugin.
I installed a docker image "546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-raspberry-pi:latest" although OS is Ubuntu-MATE.

BUT, when I start stream video with ```gst-launch-1.0``` ,
(Run ```sudo docker pull 546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-raspberry-pi:latest```)
FAIL like avobe [comment](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/358#issuecomment-591843679)

 
Docker Container Logs are,
```
0:00:00.000176625    49   0x837c00 INFO                GST_INIT gst.c:510:init_pre: Initializing GStreamer Core Library version 1.10.4
0:00:00.000358510    49   0x837c00 INFO                GST_INIT gst.c:511:init_pre: Using library installed in /usr/lib/arm-linux-gnueabihf
0:00:00.000500705    49   0x837c00 INFO                GST_INIT gst.c:522:init_pre: Linux 695dca2edf76 4.15.0-1055-raspi2 #59-Ubuntu SMP PREEMPT Mon Feb 3 20:23:12 UTC 2020 aarch64
0:00:00.001980319    49   0x837c00 INFO                GST_INIT gstmessage.c:126:_priv_gst_message_initialize: init messages
0:00:00.004117106    49   0x837c00 INFO                GST_INIT gstcontext.c:83:_priv_gst_context_initialize: init contexts
0:00:00.005435565    49   0x837c00 INFO      GST_PLUGIN_LOADING gstplugin.c:316:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.006021796    49   0x837c00 INFO      GST_PLUGIN_LOADING gstplugin.c:224:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.006129875    49   0x837c00 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.006253684    49   0x837c00 INFO            GST_REGISTRY gstregistry.c:1738:ensure_current_registry: reading registry cache: /root/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.049564654    49   0x837c00 INFO            GST_REGISTRY gstregistrybinary.c:619:priv_gst_registry_binary_read_cache: loaded /root/.cache/gstreamer-1.0/registry.arm.bin in 0.043204 seconds
0:00:00.049851545    49   0x837c00 INFO            GST_REGISTRY gstregistry.c:1594:scan_and_update_registry: Validating plugins from registry cache: /root/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.077949892    49   0x837c00 INFO            GST_REGISTRY gstregistry.c:1696:scan_and_update_registry: Registry cache has not changed
0:00:00.078036824    49   0x837c00 INFO            GST_REGISTRY gstregistry.c:1773:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.078083754    49   0x837c00 INFO                GST_INIT gst.c:720:init_post: GLib runtime version: 2.50.3
0:00:00.078137715    49   0x837c00 INFO                GST_INIT gst.c:722:init_post: GLib headers version: 2.50.3
0:00:00.078179957    49   0x837c00 INFO                GST_INIT gst.c:723:init_post: initialized GStreamer successfully
0:00:00.078394500    49   0x837c00 INFO            GST_PIPELINE gstparse.c:325:gst_parse_launch_full: parsing pipeline description 'v4l2src do-timestamp=TRUE device=/dev/vchiq ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc control-rate=1 target-bitrate=5120000 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name=RaspberryPi2 access-key=AKIARO6NE6PT72FBHKXL secret-key=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 aws-region=ap-northeast-1 '
0:00:00.094095641    49   0x837c00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideo4linux2.so" loaded
0:00:00.094205022    49   0x837c00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "v4l2src"
0:00:00.098350098    49   0x837c00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseSrc@0x87c0a0> adding pad 'src'
0:00:00.099799919    49   0x837c00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideoconvert.so" loaded
0:00:00.099937687    49   0x837c00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "videoconvert"
0:00:00.101118118    49   0x837c00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseTransform@0x92c1d8> adding pad 'sink'
0:00:00.101592103    49   0x837c00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstBaseTransform@0x92c1d8> adding pad 'src'
0:00:00.124611050    49   0x837c00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstomx-rpi.so" loaded
0:00:00.124723973    49   0x837c00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "omxh264enc"
[... cut ...]
0:00:01.630473054    49   0x837c00 INFO         GST_REFCOUNTING gstelement.c:3071:gst_element_finalize:<pipeline0> 0x96e208 finalize
0:00:01.630551496    49   0x837c00 INFO         GST_REFCOUNTING gstelement.c:3077:gst_element_finalize:<pipeline0> 0x96e208 finalize parent
0:00:01.630642074    49   0x837c00 INFO                GST_INIT gst.c:989:gst_deinit: deinitializing GStreamer
0:00:01.640276833    49   0x837c00 INFO                GST_INIT gst.c:1130:gst_deinit: deinitialized GStreamer
log4cplus:ERROR Failed to open file ./log/kvs.log

```

chehefen, 2020-03-03:
can you try this pipeline instead: 
gst-launch-1.0 videotestsrc is-live=TRUE ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 speed-preset=veryfast key-int-max=30 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name="YourStreamName" access-key="YourAccessKey" secret-key="YourSecretKey" aws-region="YourAWSRegion"

joyuiyeong, 2020-03-03:
@disa6302 @chehefen 

_**Is there any problem to build SDK in ARM - 64 bit side raspberry pi?**_
_**Or just is building SDK burden to this raspberry pi? **_

_well, i need to install and build [userland](https://github.com/raspberrypi/userland) with ```./buildme --aarch64``` option (which makes buidling native and to cross compiles)_
_I wonder the I have to do something more with this architecture when building SDK._

**AND**
```  
$ gst-launch-1.0 videotestsrc is-live=TRUE ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 speed-preset=veryfast key-int-max=30 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1,profile=baseline ! kvssink stream-name="YourStreamName" access-key="YourAccessKey" secret-key="YourSecretKey" aws-region="YourAWSRegion"
```

I tried above command, but streaming was failed with these logs,
```
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
0:00:07.347950051    16  0x16abc00 INFO               GST_EVENT gstevent.c:1382:gst_event_new_latency: creating latency event 0:00:00.000000000
0:00:07.348194774    16  0x16abc00 INFO        GST_ELEMENT_PADS gstelement.c:1564:gst_element_default_send_event: can't send latency event on element kvssink0
0:00:07.348340962    16  0x16abc00 WARN                     bin gstbin.c:2802:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
0:00:07.348783172    16  0x16abc00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<kvssink0> current PAUSED pending VOID_PENDING, desired next PLAYING
0:00:07.349015396    16  0x16abc00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<kvssink0> completed state change to PLAYING
0:00:07.349136377    16  0x16abc00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<kvssink0> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:07.349279232    16  0x16abc00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'kvssink0' changed state to 4(PLAYING) successfully
0:00:07.349403130    16  0x16abc00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<capsfilter1> current PAUSED pending VOID_PENDING, desired next PLAYING
0:00:07.349511404    16  0x16abc00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<capsfilter1> completed state change to PLAYING
0:00:07.349605408    16  0x16abc00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<capsfilter1> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:07.349723994    16  0x16abc00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'capsfilter1' changed state to 4(PLAYING) successfully
0:00:07.349850444    16  0x16abc00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<h264parse0> current PAUSED pending VOID_PENDING, desired next PLAYING
0:00:07.349914450    16  0x16abc00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<h264parse0> completed state change to PLAYING
0:00:07.349981945    16  0x16abc00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<h264parse0> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:07.350049701    16  0x16abc00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'h264parse0' changed state to 4(PLAYING) successfully
0:00:07.350120009    16  0x1803f20 INFO                 x264enc gstx264enc.c:561:gst_x264_enc_add_x264_chroma_format: This x264 build supports 8-bit depth
0:00:07.350165318    16  0x16abc00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<x264enc0> current PAUSED pending VOID_PENDING, desired next PLAYING
0:00:07.350223908    16  0x16abc00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<x264enc0> completed state change to PLAYING
0:00:07.350265207    16  0x16abc00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<x264enc0> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:07.350327703    16  0x16abc00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'x264enc0' changed state to 4(PLAYING) successfully
0:00:07.350388272    16  0x16abc00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<capsfilter0> current PAUSED pending VOID_PENDING, desired next PLAYING
0:00:07.350425301    16  0x1803f20 INFO                 x264enc gstx264enc.c:561:gst_x264_enc_add_x264_chroma_format: This x264 build supports 8-bit depth
0:00:07.350472641    16  0x16abc00 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<capsfilter0> completed state change to PLAYING
0:00:07.350549928    16  0x16abc00 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<capsfilter0> notifying about state-changed PAUSED to PLAYING (VOID_PENDING pending)
0:00:07.350614403    16  0x16abc00 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<pipeline0> child 'capsfilter0' changed state to 4(PLAYING) successfully
0:00:07.350678878    16  0x16abc00 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<videoconvert0> current PAUSED pending VOID_PENDING, desired next PLAYING
[... cut ...]
[DEBUG][2020-03-03 23:42:23] stepStateMachine(): PIC Client State Machine - Current state: 0xf7035a28, Next state: 0x40
[INFO ][2020-03-03 23:42:23] putStreamResultEvent(): Put stream result event.
[DEBUG][2020-03-03 23:42:23] stepStateMachine(): PIC Stream State Machine - Current state: 0x0, Next state: 0x80
0:00:08.262822553    16  0x1803f20 INFO               baseparse gstbaseparse.c:3994:gst_base_parse_set_latency:<h264parse0> min/max latency 0:00:00.033333333, 0:00:00.033333333
Bus error (core dumped)
```

**What does mean ```Bus error(core dumped)```?**

joyuiyeong, 2020-03-03:
In docker image(_546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-raspberry-pi_), I fail to stream with ```gst-launch-1.0``` command or ```./kinesis_video_gstreamer_sample_app``` sample app.

BUT, With ```./kinesis_video_cproducer_video_only_sample``` I success to stream!

Is audio source the cause???

chehefen, 2020-03-04:
kinesis_video_cproducer_video_only_sample streams pre-recorded h264 files so there is no encoding nor gstreamer involved. Getting gstreamer running is very platform specific. You can check the gstreamer doc to see how to get gstreamer working in your setup: https://gstreamer.freedesktop.org/documentation/tutorials/index.html?gi-language=c

joyuiyeong, 2020-03-04:
@chehefen Ok, I check the document. Thanks for your help.

closed 2020-03-04 by joyuiyeong