awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #360: No Valid Credential Method Was Found

patricktophatrobotics, 2020-02-27:
I am using a Raspberry Pi 4 4GB ram with Ubuntu 18.04 installed. When running this series of commands, I receive the following error. I believe I am using the right access and secret access keys as well. Any help is greatly appreciated.
[image]

disa6302, 2020-02-27:
Hello @patricktophatrobotics , 
What is the error that you are seeing?

patricktophatrobotics, 2020-02-27:
I have updated the post sorry about that and thank you for the quick response!

zhiyua-git, 2020-02-27:
Hi @patricktophatrobotics ,

You can use sudo -E ./kinesis_video_gstreamer_sample_app to use your environment variable. BTW, we will still recommend you to deactivate your access key/secret key and create a new one. Thanks.

patricktophatrobotics, 2020-02-27:
ubuntu@ubuntu:/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$ sudo -E ./kinesis_video_gstreamer_sample_app Unit_Main
log4cplus:ERROR Unable to open file: ./log/kvs.log
[INFO ] [27-02-2020 22:44:05:840.552 GMT] Using region: us-east-2
[INFO ] [27-02-2020 22:44:05:840.635 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [27-02-2020 22:44:05:840.678 GMT] No session token was detected.
[INFO ] [27-02-2020 22:44:05:843.334 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [27-02-2020 22:44:05:843.557 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [27-02-2020 22:44:05:843.606 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [27-02-2020 22:44:05:843.698 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [27-02-2020 22:44:05:843.774 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [27-02-2020 22:44:05:843.832 GMT] getSecurityTokenHandler invoked
[DEBUG] [27-02-2020 22:44:05:843.909 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1582843445843883303 Expiration: 0
[INFO ] [27-02-2020 22:44:05:843.956 GMT] New credentials expiration is 1582845845
[DEBUG] [27-02-2020 22:44:05:844.027 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [27-02-2020 22:44:05:852.271 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [27-02-2020 22:44:05:852.460 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [27-02-2020 22:44:05:852.543 GMT] clientReadyHandler invoked
[DEBUG] [27-02-2020 22:44:05:852.628 GMT] Client is ready
[INFO ] [27-02-2020 22:44:05:852.748 GMT] Creating Kinesis Video Stream Unit_Main
[INFO ] [27-02-2020 22:44:05:852.823 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [27-02-2020 22:44:05:853.090 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [27-02-2020 22:44:06:209.598 GMT] writeHeaderCallback(): RequestId: 8c6258df-7aa5-44b6-8f0a-c86132acafd5
[WARN ] [27-02-2020 22:44:06:209.799 GMT] curlCompleteSync(): HTTP Error 404 : Response: {"Message":"The requested stream is not found or not active."}
Request URL: https://kinesisvideo.us-east-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAYM2LQ3ZWQMEFT5GS/20200227/us-east-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=20891bb1d8409de49189511189d4d0e48fd5b847ca83aaab06e9692d1b9b2e24
    content-length: 30
    content-type: application/json
    host: kinesisvide
[DEBUG] [27-02-2020 22:44:06:210.013 GMT] describeStreamCurlHandler(): DescribeStream API response: {"Message":"The requested stream is not found or not active."}
[INFO ] [27-02-2020 22:44:06:214.266 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [27-02-2020 22:44:06:214.424 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000004
[DEBUG] [27-02-2020 22:44:06:219.066 GMT] createStreamCurlHandler(): createStreamCurlHandler response 0xffff7c064180
[INFO ] [27-02-2020 22:44:06:387.728 GMT] writeHeaderCallback(): RequestId: f5bf45ff-649d-41ee-9113-41a293b512a9
[DEBUG] [27-02-2020 22:44:06:387.852 GMT] createStreamCurlHandler(): CreateStream API response: {"StreamARN":"arn:aws:kinesisvideo:us-east-2:577294622317:stream/Unit_Main/1582843446351"}
[INFO ] [27-02-2020 22:44:06:389.951 GMT] createStreamResultEvent(): Create stream result event.
[DEBUG] [27-02-2020 22:44:06:390.047 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000004, Next state: 0x0000000000000020
[INFO ] [27-02-2020 22:44:06:564.374 GMT] writeHeaderCallback(): RequestId: fa3346fd-a3ba-404a-b33c-1fdc14a2b210
[DEBUG] [27-02-2020 22:44:06:564.572 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-171c51c7.kinesisvideo.us-east-2.amazonaws.com"}
[INFO ] [27-02-2020 22:44:06:569.042 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [27-02-2020 22:44:06:569.184 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [27-02-2020 22:44:06:569.302 GMT] getStreamingTokenHandler invoked
[DEBUG] [27-02-2020 22:44:06:569.371 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1582843446569359947 Expiration: 1582845845
[INFO ] [27-02-2020 22:44:06:569.480 GMT] New credentials expiration is 1582845846
[INFO ] [27-02-2020 22:44:06:569.553 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [27-02-2020 22:44:06:569.630 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [27-02-2020 22:44:06:569.714 GMT] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 187650475785625
[DEBUG] [27-02-2020 22:44:06:569.770 GMT] streamReadyHandler invoked
[DEBUG] [27-02-2020 22:44:06:569.930 GMT] Stream is ready
[INFO ] [27-02-2020 22:44:07:169.608 GMT] Streaming from live source
[DEBUG] [27-02-2020 22:44:07:170.237 GMT] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512
Unable to set the source to ready state.
[... cut ...]
[INFO ] [27-02-2020 22:44:07:680.760 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[WARN ] [27-02-2020 22:44:07:680.838 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [27-02-2020 22:44:08:081.255 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [27-02-2020 22:44:08:081.421 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
log4cplus:ERROR Failed to open file ./log/kvs.log

disa6302, 2020-02-27:
Can you do `export GST_DEBUG=4` and send us a log? It looks like GStreamer is unable to set the source to ready state.

patricktophatrobotics, 2020-02-28:
Hello and thank you for your help, I ran that command however where can I locate the log for me to share?

zhiyua-git, 2020-02-28:
Hi @patricktophatrobotics ,

With the GST_DEBUG=4 you will get lots of log lines looks like 
```
0:00:00.006209000 16055 0x7ff3a8d01e00 INFO                GST_INIT gst.c:586:gboolean init_pre(GOptionContext *, GOptionGroup *, gpointer, GError **): Initializing GStreamer Core Library version 1.16.0
0:00:00.006667000 16055 0x7ff3a8d01e00 INFO                GST_INIT gst.c:587:gboolean init_pre(GOptionContext *, GOptionGroup *, gpointer, GError **): Using library installed in /usr/local/Cellar/gstreamer/1.16.0_1/lib
0:00:00.006733000 16055 0x7ff3a8d01e00 INFO                GST_INIT gst.c:607:gboolean init_pre(GOptionContext *, GOptionGroup *, gpointer, GError **): Darwin 4c327594216b.ant.amazon.com 18.7.0 Darwin Kernel Version 18.7.0: Thu Jan 23 06:52:12 PST 2020; root:xnu-4903.278.25~1/RELEASE_X86_64 x86_64
0:00:00.013471000 16055 0x7ff3a8d01e00 INFO                GST_INIT gstmessage.c:128:void _priv_gst_message_initialize(void): init messages
0:00:00.016258000 16055 0x7ff3a8d01e00 INFO                GST_INIT gstcontext.c:84:void _priv_gst_context_initialize(void): init contexts
0:00:00.017356000 16055 0x7ff3a8d01e00 INFO      GST_PLUGIN_LOADING gstplugin.c:318:void _priv_gst_plugin_initialize(void): registering 0 static plugins
0:00:00.021470000 16055 0x7ff3a8d01e00 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gboolean gst_plugin_register_static(gint, gint, const gchar *, const gchar *, GstPluginInitFunc, const gchar *, const gchar *, const gchar *, const gchar *, const gchar *): registered static plugin "staticelements"
0:00:00.021507000 16055 0x7ff3a8d01e00 INFO      GST_PLUGIN_LOADING gstplugin.c:228:gboolean gst_plugin_register_static(gint, gint, const gchar *, const gchar *, GstPluginInitFunc, const gchar *, const gchar *, const gchar *, const gchar *, const gchar *): added static plugin "staticelements", result: 1
0:00:00.022775000 16055 0x7ff3a8d01e00 INFO            GST_REGISTRY gstregistry.c:1723:gboolean ensure_current_registry(GError **): reading registry cache: /Users/zhiyua/.cache/gstreamer-1.0/registry.x86_64.bin
0:00:00.071140000 16055 0x7ff3a8d01e00 INFO            GST_REGISTRY gstregistrybinary.c:621:gboolean priv_gst_registry_binary_read_cache(GstRegistry *, const char *): loaded /Users/zhiyua/.cache/gstreamer-1.0/registry.x86_64.bin in 0.047446 seconds
0:00:00.071269000 16055 0x7ff3a8d01e00 INFO            GST_REGISTRY gstregistry.c:1579:GstRegistryScanAndUpdateResult scan_and_update_registry(GstRegistry *, const gchar *, gboolean, GError **): Validating plugins from registry cache: /Users/zhiyua/.cache/gstreamer-1.0/registry.x86_64.bin
0:00:00.116545000 16055 0x7ff3a8d01e00 INFO            GST_REGISTRY gstregistry.c:1681:GstRegistryScanAndUpdateResult scan_and_update_registry(GstRegistry *, const gchar *, gboolean, GError **): Registry cache has not changed
0:00:00.116623000 16055 0x7ff3a8d01e00 INFO            GST_REGISTRY gstregistry.c:1758:gboolean ensure_current_registry(GError **): registry reading and updating done, result = 1
0:00:00.116657000 16055 0x7ff3a8d01e00 INFO                GST_INIT gst.c:807:gboolean init_post(GOptionContext *, GOptionGroup *, gpointer, GError **): GLib runtime version: 2.60.6
0:00:00.116684000 16055 0x7ff3a8d01e00 INFO                GST_INIT gst.c:809:gboolean init_post(GOptionContext *, GOptionGroup *, gpointer, GError **): GLib headers version: 2.60.4
0:00:00.116699000 16055 0x7ff3a8d01e00 INFO                GST_INIT gst.c:810:gboolean init_post(GOptionContext *, GOptionGroup *, gpointer, GError **): initialized GStreamer successfully
[INFO ] [28-02-2020 18:54:30:970.805 UTC] Streaming from live source
```

Please share that with us. Thanks.

patricktophatrobotics, 2020-02-28:
[image]

Nothing outputs when I run that command am I doing something wrong?

disa6302, 2020-02-28:
Hello @patricktophatrobotics ,
You need to run the sample again after exporting. That is when you see the logs.

patricktophatrobotics, 2020-02-28:
[image]

I still do not see anything

disa6302, 2020-02-28:
@patricktophatrobotics , 

Are you sure you set your AWS_ACCESS_KEY and AWS_SECRET_ACCESS_KEY? 
export AWS_ACCESS_KEY=<your key>
export AWS_SECRET_ACCESS_KEY=<your value>

patricktophatrobotics, 2020-02-28:
[image]
Yes here it is

disa6302, 2020-02-28:
Since you are running it with sudo, run the executable with -E flag.

patricktophatrobotics, 2020-02-28:
ubuntu@ubuntu:/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build$ sudo -E ./kinesis_video_gstreamer_sample_app -w 1920 -h 1080 Main_Unit
log4cplus:ERROR Unable to open file: ./log/kvs.log
[INFO ] [28-02-2020 21:08:35:644.188 GMT] Using region: us-east-2
[INFO ] [28-02-2020 21:08:35:644.394 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [28-02-2020 21:08:35:644.426 GMT] No session token was detected.
[INFO ] [28-02-2020 21:08:35:648.520 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [28-02-2020 21:08:35:648.971 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [28-02-2020 21:08:35:649.050 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [28-02-2020 21:08:35:649.150 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [28-02-2020 21:08:35:649.223 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [28-02-2020 21:08:35:649.283 GMT] getSecurityTokenHandler invoked
[DEBUG] [28-02-2020 21:08:35:649.390 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1582924115649329927 Expiration: 0
[INFO ] [28-02-2020 21:08:35:649.430 GMT] New credentials expiration is 1582926515
[DEBUG] [28-02-2020 21:08:35:649.504 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [28-02-2020 21:08:35:649.547 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [28-02-2020 21:08:35:649.584 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [28-02-2020 21:08:35:649.616 GMT] clientReadyHandler invoked
[DEBUG] [28-02-2020 21:08:35:649.657 GMT] Client is ready
[INFO ] [28-02-2020 21:08:35:649.735 GMT] Creating Kinesis Video Stream -w
[INFO ] [28-02-2020 21:08:35:649.777 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [28-02-2020 21:08:35:649.998 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [28-02-2020 21:08:35:837.571 GMT] writeHeaderCallback(): RequestId: 82201b8d-dd49-4c15-a70c-63907df60993
[DEBUG] [28-02-2020 21:08:35:837.796 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.582921643595E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-east-2:577294622317:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-2:577294622317:stream/-w/1582921643595","StreamName":"-w","Version":"YGYTkMdbiMhEXHfWqW9P"}}
[INFO ] [28-02-2020 21:08:35:840.524 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [28-02-2020 21:08:35:840.716 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [28-02-2020 21:08:36:035.294 GMT] writeHeaderCallback(): RequestId: b67a4226-841a-4838-9063-61a3ee5a4b08
[DEBUG] [28-02-2020 21:08:36:035.447 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-171c51c7.kinesisvideo.us-east-2.amazonaws.com"}
[INFO ] [28-02-2020 21:08:36:037.809 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [28-02-2020 21:08:36:037.946 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [28-02-2020 21:08:36:037.985 GMT] getStreamingTokenHandler invoked
[DEBUG] [28-02-2020 21:08:36:038.019 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1582924116038008852 Expiration: 1582926515
[INFO ] [28-02-2020 21:08:36:038.045 GMT] New credentials expiration is 1582926516
[INFO ] [28-02-2020 21:08:36:038.078 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [28-02-2020 21:08:36:038.116 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [28-02-2020 21:08:36:038.159 GMT] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 187651547303321
[DEBUG] [28-02-2020 21:08:36:038.218 GMT] streamReadyHandler invoked
[DEBUG] [28-02-2020 21:08:36:038.341 GMT] Stream is ready
0:00:00.000150796  2685 0xaaab07cefe00 INFO                GST_INIT gst.c:586:init_pre: Initializing GStreamer Core Library version 1.14.5
0:00:00.000220704  2685 0xaaab07cefe00 INFO                GST_INIT gst.c:587:init_pre: Using library installed in /usr/lib/aarch64-linux-gnu
0:00:00.000257852  2685 0xaaab07cefe00 INFO                GST_INIT gst.c:607:init_pre: Linux ubuntu 5.3.0-1018-raspi2 #20~18.04.1-Ubuntu SMP Wed Feb 5 04:56:34 UTC 2020 aarch64
0:00:00.000801573  2685 0xaaab07cefe00 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.001851164  2685 0xaaab07cefe00 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.002313386  2685 0xaaab07cefe00 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.002584275  2685 0xaaab07cefe00 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.002625904  2685 0xaaab07cefe00 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
[... cut ...]
[INFO ] [28-02-2020 21:08:36:503.542 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[WARN ] [28-02-2020 21:08:36:503.573 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [28-02-2020 21:08:36:903.921 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [28-02-2020 21:08:36:904.133 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
log4cplus:ERROR Failed to open file ./log/kvs.log

disa6302, 2020-02-28:
Hello @patricktophatrobotics , 

What kind of camera are you using? 

If it is a USB camera, can you do the following?
1. Check current entries of /dev/video* by running `ls /dev/video*`.
2. Check if the Pi recognizes the camera. Run `lsusb`
3. If yes, check the dev node for it. Run `ls /dev/video*`. A new `/dev/video*` will be available if the Pi recognizes the camera. 

If it is a RTSP camera, you need to provide the rtsp url of the camera is the gst pipeline.

zhiyua-git, 2020-02-28:
Hi @patricktophatrobotics ,

Seems on your device you don't have a camera call "/dev/video0"

0:00:00.063948949 2685 0xaaab07cefe00 WARN v4l2 v4l2_calls.c:609:gst_v4l2_open: error: Cannot identify device '/dev/video0'.

You need to check what device do you have and specify the correct device name in [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L723). You can find your devices by running gst-device-monitor-1.0. For more details, check [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#discovering-audio-and-video-devices-available-in-your-system). Thanks.

One example here (see device.path = /dev/video0):

```
Device found:

	name  : 1080p Pro Stream Webcam
	class : Video/Source
	caps  : video/x-raw, format=(string)YUY2, width=(int)2304, height=(int)1536, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
	        video/x-raw, format=(string)YUY2, width=(int)2304, height=(int)1296, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
	        video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1;
	        video/x-raw, format=(string)YUY2, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)960, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)864, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)432, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)320, height=(int)180, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        video/x-raw, format=(string)YUY2, width=(int)160, height=(int)90, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        image/jpeg, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        image/jpeg, width=(int)960, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        image/jpeg, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        image/jpeg, width=(int)864, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        image/jpeg, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
	        image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[... cut ...]
		device.product.name = "1080p\ Pro\ Stream\ Webcam"
		device.serial = 046d_1080p_Pro_Stream_Webcam_19CD5B9F
		device.capabilities = :capture:
		device.api = v4l2
		device.path = /dev/video0
		v4l2.device.driver = uvcvideo
		v4l2.device.card = "1080p\ Pro\ Stream\ Webcam"
		v4l2.device.bus_info = usb-3f980000.usb-1.2
		v4l2.device.version = 267058 (0x00041332)
		v4l2.device.capabilities = 2225078273 (0x84a00001)
		v4l2.device.device_caps = 69206017 (0x04200001)
	gst-launch-1.0 v4l2src ! ...
```

MushMal, 2020-03-17:
Resolving this issue as it diverged from the original question. Please feel free to open a separate issue which will target the problem better.

closed 2020-03-17 by MushMal

Nritesh-akshay, 2022-09-01:
[image]
..........................similar issue i am facing