awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #361: video image quality is low / pixelated

mistersingh179, 2020-03-03:
The image quality of the video streamed to KVS appears to be low and pixelated. Below is a screenshot of the image from KVS side by side with the the camera directly opened via photo booth app on mac.

I am running `./kinesis_video_gstreamer_audio_video_sample_app` on a mac with `video_caps_string = "video/x-raw,framerate=5000000/83333, format=I420, width=1920, height=1080";`

What do you recommend I do to improve the video quality?

[image]

chehefen, 2020-03-04:
Hi, it depends on how you tune the gstreamer pipeline. Please refer to the gstreamer documentation for more details: https://gstreamer.freedesktop.org/documentation/tutorials/index.html?gi-language=c. You will need to fine tune your encoder settings such as increasing bitrate.

mistersingh179, 2020-03-04:
thanks @chehefen for pointing me in the right direction.

Just in case this is helpful to the next person:

I was able to fix the pixelated issue by increasing the bitrate to 8192. 

The change is at https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_audio_video_sample_app.cpp#L871

Based on this support article from google I took the value of 8Mbps for bitrate
https://support.google.com/youtube/answer/1722171?hl=en

closed 2020-03-04 by mistersingh179