awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #366: putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x3200000b decoding timestamp: 311950000 presentation timestamp: 311950000

Remainfox, 2020-03-12:
I'm passing on a time stamp from the collection.But the report was wrong.

2020-03-12 08:03:31 WARN    putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x3200000b decoding timestamp: 311950000 presentation timestamp: 311950000

frame.duration 				= obj->video_frameIndex ? VIDEO_DURATION : (pts - obj->previousVideoTimestamp);
	frame.duration				= -1;
	frame_index    				= obj->video_frameIndex++;
	frame.decodingTs	 		= pts * HUNDREDS_OF_NANOS_IN_A_MILLISECOND;
	frame.presentationTs 		= frame.decodingTs;
	obj->previousVideoTimestamp = pts;		
	
	frame.index 	= frame_index;
	frame.flags 	= isIDR ? FRAME_FLAG_KEY_FRAME : FRAME_FLAG_NONE;
	frame.size		= size;
	frame.frameData = data;
	frame.version	= FRAME_CURRENT_VERSION;
	frame.trackId	= DEFAULT_VIDEO_TRACK_ID;
	
	ret = putKinesisVideoFrame(obj->streamHandle, &frame);

zhiyua-git, 2020-03-12:
Hi @Remainfox ,

You are hitting max frame timecode. It means you are not setting key frame for >32767 ms = 32.7 seconds. As you are setting key frame for IDR frame, this might mean your encoder setting doesn’t generate IDR frame frequently enough. You need to tune encoder to provide IDR frames. Thanks.

You can refer to https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html#producer-sdk-errors-mkvgen for error codes.

Remainfox, 2020-03-18:
thanks for replay, i I solved this problem by referring to kvsaacvideosample

closed 2020-03-18 by MushMal

gal1l0, 2022-06-14:
Wait how do did you do that ?