awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #367: How to have souphttpsrc working with Gstreamer with this tutorial

satyajitmails, 2020-03-12:
Dear team,

I have followed the tutorial for Linux to install Gstreamer and amazon-kinesis-video-streams-producer-sdk-cpp.I am trying to set up a flow where I will stream from a http based video/audio url.I am able to run gstreamer to upload to KVS few mkv/mp4 file. But when I am trying the souphttpsrc it is not recognising it.I also used gst-inspect-1.0 souphttpsrc to inspect and got this message "No such element or plugin 'souphttpsrc'.

The command start like .
gst-launch-1.0 -v souphttpsrc location=…

and the message I am getting is "WARNING: erroneous pipeline: no element "souphttpsrc"
I tried everything from setting up flags in .configure inside install_script ,trying to compile C++ files to produce a separate .so , to multiple times installing reinstalling the gst-plugins in multiple VM's, to going through a whole bunch of documents but no luck here. :-(

Please let me know if you require more information.Appreciate a quick help.

zhiyua-git, 2020-03-12:
Hi @satyajitmails ,

According to http://gstreamer-devel.966125.n4.nabble.com/manually-compile-souphttpsrc-td4687511.html, you might need to install libsoup dependency. Any of these gstreamer related questions would get better answered on http://gstreamer-devel.966125.n4.nabble.com/ or  https://gstreamer.freedesktop.org/documentation/tutorials/index.html?gi-language=c. Thanks.

MushMal, 2020-03-17:
Resolving. Please open a separate issue for KVS related issues here.

closed 2020-03-17 by MushMal

satyajitmails, 2020-03-20:
Thanks team, this stuff is at back burner for a while, will update how it went when I get into it again.