awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #368: Problem streaming with docker.

pdalmaso1, 2020-03-13:
Im running a poc that consists to stream IP cameras to KVS.
So in my first tests I was streaming using the docker image inside /docker_native_scripts that I build myself via Dockerfile. 
As I ran against a few problems I was instructed to use the latest docker image (546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux:latest).
So now Im streaming using this latest image but I cant see anything in the KVS media playback at AWS console. All I got is the message "No media found. Verify that you are streaming to us-east-1 and the fragment timecodes are correct, or download the Kinesis Video Streams Producer SDK to set up your device."

Earlier when I was using the docker container that I build from the scripts I used to start the container (and the streaming) with the following command. 
`docker run -it <IMAGE_ID> <AWS_ACCESS_KEY_ID> <AWS_SECRET_ACCESS_KEY> <RTSP_URL> <STREAM_NAME>`

Now with the latest docker image Im using the followings commands, since Im not able to run it like I was running before.

```
sudo docker run -it --network="host" --device=/dev/video0 546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux:latest /bin/bash

gst-launch-1.0 -v rtspsrc location=<rtsp-url> short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink stream-name=Channel4 storage-size=128 access-key=<access-key> secret-key=<secret-key> aws-region="us-east-1"
```
The logs appears to be all ok and the region is being printed correct in the logs as well. 
Am I doing something wrong?

disa6302, 2020-03-13:
Hello @pdalmaso1 ,

Can you share the logs?

pdalmaso1, 2020-03-13:
[logsKVS.txt]

Is there any way to run the latest version of the docker and start streaming?? Should I edit the docker to do so?

pdalmaso1, 2020-03-13:
[logsKVS.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4330726/logsKVS.txt)

Is there any way I can run the latest container and already start streaming? Maybe should I edit the container to do so?

zhiyua-git, 2020-03-13:
Hi @pdalmaso1 ,

Your log looks correct and everything is working as expected. Can you refresh the AWS Console page and check the media player again while you are streaming? You might need to click the skip button on the right bottom. Please note the player would be freeze while you are not staying on the page and it will not automatically restart if it already get into status of "No media found. Verify that you are streaming to us-east-1 and the fragment timecodes are correct, or download the Kinesis Video Streams Producer SDK to set up your device.".

You can also check the playback using HLS https://aws-samples.github.io/amazon-kinesis-video-streams-media-viewer/. Thanks.

closed 2020-03-16 by pdalmaso1