awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #369: Unable to stream from Webcam MacOS Cataline

harsimranb, 2020-03-13:
I have done the following:
1. Disable System Integrity Protection
2. Successfully run `./install-script`.
3. Update LD_LIBRARY_PATH, GST_PLUGIN_PATH, and PATH
4. Set AWS Credentials in environment variables 

However, when I try to stream using the same application or gst-launch, programs seems to exit gracefully without any explicit errors.  Nothing is streamed to Kinesis, my webcam never turns on.

# Something Curious
When I run `gst-device-monitor-1.0` (gst-device-monitor-1.0 version 1.16.2 GStreamer 1.16.2) from my brew installed, it lists the webcam device in the output.

When I run `gst-device-monitor-1.0` (gst-device-monitor-1.0 version 1.12.3, GStreamer 1.12.3) from `./downloads/local/bin` it **does not list the webcam device in the output**.

Right now my hunch is, this might be related to the problem.

# Commands I've tried:
## gst-launch
With gst-launch, I tried `autovideosrc device=/dev/video0 ! videoconvert` and `autovideosrc !videoconvert`. Ex below:
```
gst-launch-1.0 autovideosrc device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=1280,height=720,framerate=30/1 ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=500 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="KVS1" storage-size=512  access-key=$AWS_ACCESS_KEY_ID secret-key=$AWS_SECRET_ACCESS_KEY aws-region="us-east-1"
```
### Log
```
Setting pipeline to PAUSED ...
[INFO ] [13-03-2020 22:27:29:611.117 UTC] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [13-03-2020 22:27:29:611.233 UTC] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [13-03-2020 22:27:29:611.256 UTC] heapInitialize(): Creating AIV heap.
[INFO ] [13-03-2020 22:27:29:611.287 UTC] heapInitialize(): Heap is initialized OK
[DEBUG] [13-03-2020 22:27:29:611.308 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [13-03-2020 22:27:29:611.328 UTC] getSecurityTokenHandler invoked
[DEBUG] [13-03-2020 22:27:29:611.348 UTC] Refreshing credentials. Force refreshing: 0 Now time is: 1584138449611343 Expiration: 0
[DEBUG] [13-03-2020 22:27:29:611.370 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [13-03-2020 22:27:29:611.392 UTC] createDeviceResultEvent(): Create device result event.
[DEBUG] [13-03-2020 22:27:29:611.411 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [13-03-2020 22:27:29:616.863 UTC] clientReadyHandler invoked
[INFO ] [13-03-2020 22:27:29:617.011 UTC] try creating stream
[INFO ] [13-03-2020 22:27:29:617.072 UTC] Creating Kinesis Video Stream KVS1
[INFO ] [13-03-2020 22:27:29:617.099 UTC] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [13-03-2020 22:27:29:617.177 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [13-03-2020 22:27:29:931.904 UTC] writeHeaderCallback(): RequestId: a277c211-4c77-4ad7-b467-a57ba8c56531
[DEBUG] [13-03-2020 22:27:29:932.030 UTC] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.584056183786E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-east-1:296799816296:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:296799816296:stream/KVS1/1584056183786","StreamName":"KVS1","Version":"iCMaaqNPPjRsEnFkCBGq"}}
[INFO ] [13-03-2020 22:27:29:932.304 UTC] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [13-03-2020 22:27:29:932.339 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [13-03-2020 22:27:30:338.430 UTC] writeHeaderCallback(): RequestId: 03248a1c-5dd5-473a-87bd-40c7a940fa1b
[DEBUG] [13-03-2020 22:27:30:338.575 UTC] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
[INFO ] [13-03-2020 22:27:30:339.025 UTC] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [13-03-2020 22:27:30:339.090 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [13-03-2020 22:27:30:339.120 UTC] getStreamingTokenHandler invoked
[DEBUG] [13-03-2020 22:27:30:339.152 UTC] Refreshing credentials. Force refreshing: 1 Now time is: 1584138450339144 Expiration: 18446744073709551615
[INFO ] [13-03-2020 22:27:30:339.179 UTC] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [13-03-2020 22:27:30:339.204 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [13-03-2020 22:27:30:339.236 UTC] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 140600670693776
[DEBUG] [13-03-2020 22:27:30:339.260 UTC] streamReadyHandler invoked
Stream is ready
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Abort trap: 6
```
## Sample Application
```
DEBUG_DUMP_FRAME_INFO=TRUE AWS_ACCESS_KEY_ID=$AWS_ACCESS_KEY_ID AWS_SECRET_ACCESS_KEY=$AWS_SECRET_ACCESS_KEY AWS_DEFAULT_REGION=us-east-1 ./kinesis_video_gstreamer_sample_app -w 1280 -h 720 -f 25 -b 512 KVS1
```
### Output
```
[INFO ] [13-03-2020 22:24:13:333.123 UTC] Using region: us-east-1
[INFO ] [13-03-2020 22:24:13:333.785 UTC] Using aws credentials for Kinesis Video Streams
[INFO ] [13-03-2020 22:24:13:333.851 UTC] No session token was detected.
[INFO ] [13-03-2020 22:24:13:336.358 UTC] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [13-03-2020 22:24:13:336.420 UTC] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[... 20 lines cut ...]
    content-type: application/json
    host: kinesisvide
[DEBUG] [13-03-2020 22:24:13:487.923 UTC] describeStreamCurlHandler(): DescribeStream API response: {"Message":"The requested stream is not found or not active."}
[INFO ] [13-03-2020 22:24:13:488.472 UTC] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [13-03-2020 22:24:13:488.544 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000004
[DEBUG] [13-03-2020 22:24:13:488.758 UTC] createStreamCurlHandler(): createStreamCurlHandler response 0x7fa680800a00
[INFO ] [13-03-2020 22:24:13:752.964 UTC] writeHeaderCallback(): RequestId: 3a2d06e9-d2cd-4fda-affa-f7835a489ae1
[DEBUG] [13-03-2020 22:24:13:753.077 UTC] createStreamCurlHandler(): CreateStream API response: {"StreamARN":"arn:aws:kinesisvideo:us-east-1:296799816296:stream/-w/1584138253711"}
[INFO ] [13-03-2020 22:24:13:753.434 UTC] createStreamResultEvent(): Create stream result event.
[DEBUG] [13-03-2020 22:24:13:753.494 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000004, Next state: 0x0000000000000020
[INFO ] [13-03-2020 22:24:13:894.536 UTC] writeHeaderCallback(): RequestId: da1ae011-3d16-4c4e-8cb6-c598b4a0b6e9
[DEBUG] [13-03-2020 22:24:13:894.647 UTC] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
[INFO ] [13-03-2020 22:24:13:894.949 UTC] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [13-03-2020 22:24:13:894.992 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [13-03-2020 22:24:13:895.021 UTC] getStreamingTokenHandler invoked
[DEBUG] [13-03-2020 22:24:13:895.052 UTC] Refreshing credentials. Force refreshing: 1 Now time is: 1584138253895044 Expiration: 1584140653
[INFO ] [13-03-2020 22:24:13:895.079 UTC] New credentials expiration is 1584140653
[INFO ] [13-03-2020 22:24:13:895.113 UTC] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [13-03-2020 22:24:13:895.138 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [13-03-2020 22:24:13:895.169 UTC] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 140353080585104
[DEBUG] [13-03-2020 22:24:13:895.223 UTC] streamReadyHandler invoked
[DEBUG] [13-03-2020 22:24:13:895.329 UTC] Stream is ready
[INFO ] [13-03-2020 22:24:14:494.680 UTC] Streaming from live source
[DEBUG] [13-03-2020 22:24:14:494.946 UTC] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512
```

zhiyua-git, 2020-03-14:
Hi @harsimranb ,

You are mixing up two sets of gstreamer here, from install script a gstreamer is built and installed in Kinesis-video-native-build/downloads, and you are using it together with your brew installed version.

Can you try removing the downloads directory and follow this instruction about using min-install-script? It should be fast and avoid that mix. https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md#alternate-option-to-build-the-sdk-and-sample-applications-using-libraries-installed-by-homebrew

Thank you.

harsimranb, 2020-03-14:
Hi @zhiyua-git - I have also tried `./min-install-script`, that too gives me a weird error: `No package 'gstreamer-1.0' found`.  gstreamer in the Cellar.

**Output of gst-launch-1.0 --version**
```
gst-launch-1.0 version 1.16.2
GStreamer 1.16.2
Unknown package origin
```

## Min Script Output
```
Setting up for KVS Producer SDK installation
======= make sure the following libraries have been installed (library name from homebrew) =======
* pkg-config
* openssl
* cmake
* gstreamer
* gst-plugins-base
* gst-plugins-good
* gst-plugins-bad
* gst-plugins-ugly
* log4cplus
jsmn lib not found. Installing
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   119  100   119    0     0    355      0 --:--:-- --:--:-- --:--:--   355
100 11035    0 11035    0     0  18001      0 --:--:-- --:--:-- --:--:-- 92731
x jsmn-1.0.0/
x jsmn-1.0.0/.travis.yml
x jsmn-1.0.0/LICENSE
x jsmn-1.0.0/Makefile
x jsmn-1.0.0/README.md
x jsmn-1.0.0/example/
x jsmn-1.0.0/example/jsondump.c
x jsmn-1.0.0/example/simple.c
x jsmn-1.0.0/jsmn.c
x jsmn-1.0.0/jsmn.h
x jsmn-1.0.0/library.json
x jsmn-1.0.0/test/
x jsmn-1.0.0/test/test.h
x jsmn-1.0.0/test/tests.c
x jsmn-1.0.0/test/testutil.h
rm -f *.o example/*.o
rm -f *.a *.so
rm -f simple_example
rm -f jsondump
cc -c -fPIC jsmn.c -o jsmn.o
ar rc libjsmn.a jsmn.o
Checking jsmn at /Users/harsimranb/Projects/Spring2020/CloudComputing/Assignment2/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libjsmn.a
Clean up artifacts
done Clean up artifacts
-- The C compiler identification is AppleClang 11.0.0.11000033
-- The CXX compiler identification is AppleClang 11.0.0.11000033
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- System name is Darwin
-- Kinesis Video PIC path is ../kinesis-video-pic
-- Kinesis Video Producer path is ../kinesis-video-producer
-- Kinesis Video JNI path is ../kinesis-video-producer-jni
-- Kinesis Video Gstreamer demp app path is ../kinesis-video-gst-demo
-- Kinesis Video Open Source path is /Users/harsimranb/Projects/Spring2020/CloudComputing/Assignment2/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads
-- Configuration Unix Makefiles
-- Bitness 64 bits
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.29.2") 
found release log4cplus
-- Found Log4cplus: /usr/local/lib/liblog4cplus.dylib  
-- Found OpenSSL: /usr/local/opt/openssl/lib/libcrypto.dylib (found version "1.1.1d")  
-- Found CURL: /opt/local/lib/libcurl.dylib (found version "7.64.1")  
-- Checking for module 'glib-2.0'
--   Found glib-2.0, version 2.58.3
-- Checking for module 'gstreamer-1.0'
--   No package 'gstreamer-1.0' found
CMake Error at /opt/local/share/cmake-3.14/Modules/FindPkgConfig.cmake:457 (message):
  A required package was not found
Call Stack (most recent call first):
  /opt/local/share/cmake-3.14/Modules/FindPkgConfig.cmake:642 (_pkg_check_modules_internal)
  CMakeLists.txt:159 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/Users/harsimranb/Projects/Spring2020/CloudComputing/Assignment2/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/CMakeFiles/CMakeOutput.log".
```

disa6302, 2020-03-14:
@harsimranb ,

Looks like pkg-config is not installed. Pleaserun this to install it:

`brew install pkg-config`

harsimranb, 2020-03-14:
@disa6302 Brew says its already installed.
```
Warning: pkg-config 0.29.2 is already installed and up-to-date
To reinstall 0.29.2, run `brew reinstall pkg-config`
```

But i reinstalled just in case. 
```
$ brew reinstall pkg-config
==> Reinstalling pkg-config 
==> Downloading https://homebrew.bintray.com/bottles/pkg-config-0.29.2.catalina.bottle.1.tar.gz
==> Downloading from https://akamai.bintray.com/c0/c0a6927c8e404f6db8b14d6644a218b22ddb0d27be1fa0a69d15bf0d9a6875ae?__gda__=exp=1584207600~hmac=7
######################################################################## 100.0%
==> Pouring pkg-config-0.29.2.catalina.bottle.1.tar.gz
🍺  /usr/local/Cellar/pkg-config/0.29.2: 11 files, 623KB
```

Still getting the same error when I run `./min-install-script`.

zhiyua-git, 2020-03-14:
Hi @harsimranb ,

Can you try setting the path to find gstreamer? You can check brew list gstreamer. In the output there should be a directory like /usr/local/Cellar/gstreamer/1.16.2_1/lib

Use your directory path to set your PKG_CONFIG_PATH, and then run ./min-install-script
export PKG_CONFIG_PATH=/usr/local/Cellar/gstreamer/1.16.2_1/lib/**pkgconfig**/

Thanks.

harsimranb, 2020-03-15:
@zhiyua-git That almost worked.  Now, its able to find `gstreamer` but missing another package:  'gstreamer-app-1.0'.
```
-- Checking for module 'glib-2.0'
--   Found glib-2.0, version 2.58.3
-- Checking for module 'gstreamer-1.0'
--   Found gstreamer-1.0, version 1.16.2
-- Checking for module 'gstreamer-app-1.0'
--   No package 'gstreamer-app-1.0' found
```

zhiyua-git, 2020-03-15:
Hi @harsimranb ,

One quick thing you can try is opening a new terminal and run the ./min-install-script. There might be some stale env var from install-script avoid the brew gstreamer being found. Running in a new terminal might help.

If that doesn’t work then likely you will need to do the similar PKG_CONFIG_PATH trick to other components like gstreamer-app.

harsimranb, 2020-03-15:
@zhiyua-git Thanks!  I got `min-install-script` to run successfully.  Here's the var export for those that run into this issue: `export PKG_CONFIG_PATH=/usr/local/Cellar/gstreamer/1.16.2/lib/pkgconfig/:/usr/local/Cellar/gst-plugins-base/1.16.2/lib/pkgconfig/`

However, I'm still getting an error when I run the sample app: `DEBUG_DUMP_FRAME_INFO=TRUE AWS_ACCESS_KEY_ID=$AWS_ACCESS_KEY_ID AWS_SECRET_ACCESS_KEY=$AWS_SECRET_ACCESS_KEY AWS_DEFAULT_REGION=us-east-1 ./kinesis_video_gstreamer_sample_app -w 1280 -h 720 -f 25 -b 512 KVS1`

**Output:**
```
[INFO ] [15-03-2020 03:14:19:545.485 UTC] Using region: us-east-1
[INFO ] [15-03-2020 03:14:19:545.861 UTC] Using aws credentials for Kinesis Video Streams
[INFO ] [15-03-2020 03:14:19:545.913 UTC] No session token was detected.
[INFO ] [15-03-2020 03:14:19:547.883 UTC] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [15-03-2020 03:14:19:547.936 UTC] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [15-03-2020 03:14:19:547.955 UTC] heapInitialize(): Creating AIV heap.
[INFO ] [15-03-2020 03:14:19:547.989 UTC] heapInitialize(): Heap is initialized OK
[DEBUG] [15-03-2020 03:14:19:548.021 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [15-03-2020 03:14:19:548.043 UTC] getSecurityTokenHandler invoked
[DEBUG] [15-03-2020 03:14:19:548.069 UTC] Refreshing credentials. Force refreshing: 0 Now time is: 1584242059548059 Expiration: 0
[INFO ] [15-03-2020 03:14:19:548.092 UTC] New credentials expiration is 1584244459
[DEBUG] [15-03-2020 03:14:19:548.117 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [15-03-2020 03:14:19:548.145 UTC] createDeviceResultEvent(): Create device result event.
[DEBUG] [15-03-2020 03:14:19:548.163 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [15-03-2020 03:14:19:548.186 UTC] clientReadyHandler invoked
[DEBUG] [15-03-2020 03:14:19:548.210 UTC] Client is ready
[INFO ] [15-03-2020 03:14:19:548.254 UTC] Creating Kinesis Video Stream -w
[INFO ] [15-03-2020 03:14:19:548.276 UTC] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [15-03-2020 03:14:19:548.348 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [15-03-2020 03:14:19:772.789 UTC] writeHeaderCallback(): RequestId: b13c7187-3d68-4fb4-8071-ac05eae723ed
[DEBUG] [15-03-2020 03:14:19:772.994 UTC] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.584138253711E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-east-1:<hidden>:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:<hidden>:stream/-w/1584138253711","StreamName":"-w","Version":"zZcMfJ3nrhDugHON73Db"}}
[INFO ] [15-03-2020 03:14:19:775.807 UTC] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [15-03-2020 03:14:19:775.855 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [15-03-2020 03:14:20:032.812 UTC] writeHeaderCallback(): RequestId: 8bdfa8f9-2f9a-4e28-a03f-5a8e7e20c8fb
[DEBUG] [15-03-2020 03:14:20:032.990 UTC] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
[INFO ] [15-03-2020 03:14:20:036.061 UTC] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [15-03-2020 03:14:20:036.141 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [15-03-2020 03:14:20:036.170 UTC] getStreamingTokenHandler invoked
[DEBUG] [15-03-2020 03:14:20:036.198 UTC] Refreshing credentials. Force refreshing: 1 Now time is: 1584242060036191 Expiration: 1584244459
[INFO ] [15-03-2020 03:14:20:036.221 UTC] New credentials expiration is 1584244460
[INFO ] [15-03-2020 03:14:20:036.251 UTC] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [15-03-2020 03:14:20:036.273 UTC] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [15-03-2020 03:14:20:036.301 UTC] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 140429685370256
[DEBUG] [15-03-2020 03:14:20:036.354 UTC] streamReadyHandler invoked
[DEBUG] [15-03-2020 03:14:20:036.439 UTC] Stream is ready
[INFO ] [15-03-2020 03:14:20:067.358 UTC] Streaming from live source
[DEBUG] [15-03-2020 03:14:20:067.620 UTC] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512

(process:32069): GLib-CRITICAL **: 23:14:20.101: g_datalist_id_set_data_full: assertion 'key_id > 0' failed

(process:32069): GLib-GObject-CRITICAL **: 23:14:20.101: g_param_spec_pool_lookup: assertion 'pool != NULL' failed

(process:32069): GLib-GObject-WARNING **: 23:14:20.101: g_object_set_is_valid_property: object class '(null)' has no property named 'caps'

(process:32069): GLib-CRITICAL **: 23:14:20.101: g_datalist_id_set_data_full: assertion 'key_id > 0' failed

(process:32069): GLib-GObject-CRITICAL **: 23:14:20.101: g_param_spec_pool_lookup: assertion 'pool != NULL' failed

[... cut ...]
(process:32069): GLib-GObject-CRITICAL **: 23:14:20.102: g_param_spec_pool_lookup: assertion 'pool != NULL' failed

(process:32069): GLib-GObject-WARNING **: 23:14:20.102: g_object_set_is_valid_property: object class '(null)' has no property named 'emit-signals'

(process:32069): GLib-GObject-WARNING **: 23:14:20.102: gsignal.c:2523: signal 'new-sample' is invalid for instance '0x7fb8588003a0' of type '(null)'

(process:32069): GLib-GObject-WARNING **: 23:14:20.112: gsignal.c:2523: signal 'message::error' is invalid for instance '0x7fb85607cb10' of type '(null)'
Abort trap: 6
```

zhiyua-git, 2020-03-15:
Hi @harsimranb ,

Can you try running the app from a new terminal? Also what is the output if you run gst-launch-1.0 autovideosrc ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=500 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! fakesink ? If that gst-launch-1.0 command also fail to run with abort trap you have a problem wi thy your gstreamer, consider reinstalling it and rerun min-install-script after gstreamer is working. Thanks.

harsimranb, 2020-03-15:
Hi, it is running now!  Now, the webcam turns on, however kinesis isn't showing the video stream in Media Playback section.  `DEBUG_DUMP_FRAME_INFO=TRUE AWS_ACCESS_KEY_ID=$AWS_ACCESS_KEY_ID AWS_SECRET_ACCESS_KEY=$AWS_SECRET_ACCESS_KEY AWS_DEFAULT_REGION=us-east-1 ./kinesis_video_gstreamer_sample_app KVS1 -w 1280 -h 720 -f 25 -b 512`

```
[DEBUG] [15-03-2020 03:59:28:712.247 UTC] streamReadyHandler invoked
[DEBUG] [15-03-2020 03:59:28:712.361 UTC] Stream is ready
[INFO ] [15-03-2020 03:59:28:741.798 UTC] Streaming from live source
[DEBUG] [15-03-2020 03:59:28:741.955 UTC] Streaming with live source and width: 1280, height: 720, fps: 25, bitrateInKBPS512

(process:33741): GLib-CRITICAL **: 23:59:28.813: g_datalist_id_set_data_full: assertion 'key_id > 0' failed

(process:33741): GLib-GObject-CRITICAL **: 23:59:28.813: g_param_spec_pool_lookup: assertion 'pool != NULL' failed

(process:33741): GLib-GObject-WARNING **: 23:59:28.813: g_object_set_is_valid_property: object class '(null)' has no property named 'caps'

(process:33741): GLib-CRITICAL **: 23:59:28.813: g_datalist_id_set_data_full: assertion 'key_id > 0' failed

(process:33741): GLib-GObject-CRITICAL **: 23:59:28.813: g_param_spec_pool_lookup: assertion 'pool != NULL' failed

(process:33741): GLib-GObject-WARNING **: 23:59:28.813: g_object_set_is_valid_property: object class '(null)' has no property named 'allow-frame-reordering'

(process:33741): GLib-CRITICAL **: 23:59:28.813: g_datalist_id_set_data_full: assertion 'key_id > 0' failed

(process:33741): GLib-GObject-CRITICAL **: 23:59:28.813: g_param_spec_pool_lookup: assertion 'pool != NULL' failed

(process:33741): GLib-GObject-WARNING **: 23:59:28.813: g_object_set_is_valid_property: object class '(null)' has no property named 'caps'

(process:33741): GLib-CRITICAL **: 23:59:28.813: g_datalist_id_set_data_full: assertion 'key_id > 0' failed

(process:33741): GLib-GObject-CRITICAL **: 23:59:28.813: g_param_spec_pool_lookup: assertion 'pool != NULL' failed

(process:33741): GLib-GObject-WARNING **: 23:59:28.813: g_object_set_is_valid_property: object class '(null)' has no property named 'emit-signals'

(process:33741): GLib-GObject-WARNING **: 23:59:28.813: gsignal.c:2523: signal 'new-sample' is invalid for instance '0x7fd60004eba0' of type '(null)'

(process:33741): GLib-GObject-WARNING **: 23:59:28.820: gsignal.c:2523: signal 'message::error' is invalid for instance '0x7fd60084c310' of type '(null)'
```

zhiyua-git, 2020-03-15:
Hi @harsimranb ,

Your gstreamer pipeline is still not working. Having some critical error (process:33741): GLib-CRITICAL **: 23:59:28.813: g_datalist_id_set_data_full: assertion 'key_id > 0' failed. You need to first fix the setup of gstreamer.

zhiyua-git, 2020-03-15:
Hi @harsimranb ,

After doing some search seems your installation of gstreamer could be broken: https://github.com/sdroege/gstreamer-rs/issues/60. Can you try reinstalling your gstreamer and glib to see if it helps?

disa6302, 2020-03-16:
@harsimranb , 

Can you try the following? 

- Uninstall every GStreamer related packages that you installed. Run `brew uninstall gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly gst-libav`

- Save any code changes you might have done

- Run `git clean -ffxd` from your repo directory. This will remove all new directories that were created after your most recent `git pull`

- Run `brew install gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly gst-libav`

- Run `./min-install-script` again from kinesis-native-build directory.

Considering your GStreamer installation has some issues, starting clean can help. 

Let us know if this helped fix the issues you are facing.

harsimranb, 2020-03-16:
@disa6302  Thank you!  It is working now.  Such strange behavior.

harsimranb, 2020-03-16:
@disa6302 BTW, is it necessary to disable System Integrated Protection on Catalina?  A friend mentioned he did that to make it work.

closed 2020-03-16 by harsimranb