awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #370: Are there a way to graceful stop and resume the kvssink?

jfguo, 2020-03-16:
I have a GStremer pipeline (_v4l2src ... -> ... valve ... -> kivsink_) to send data to AWS Kinesis stream. 
The element _valve_ is used to stop and resume the uploading through setting its property _stop_.

My testing result is a little bit different with what https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/289 described.

Generally it's working, but sometimes the following errors happen, one MAX_FRAGMENT_DURATION_REACHED followed by endless INVALID_MKV_DATAs.

Any ideas or suggestions can avoid the error?

MushMal, 2020-03-17:
@jfguo KVS GStreamer plugin is primarily designed for a continuous streaming. As you've noticed, the solution that was described in #289 really terminates and re-creates the pipeline with the sink in order to achieve the intermittent producer scenario. In your case, it seems that from the perspective of the KVS sink you simply stop producing frames into it.

There are three cases that this could lead to after the new frames come in.

1) The new frames happen to complete the fragment within 10 seconds (https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/limits.html) which falls within the max fragment duration so all is OK.
2) The new frames are P-frames and the KVS sink continues the fragment reaching past the max fragment duration - this is likely what you are seeing.
3) The frames don't come within 30 seconds causing the connection to timeout and re-try (continuously).

We are working on providing out-of-the-box solution which would work for KVS plugin for the 3) so the SDK will simply not retry if there are no incoming frames and the buffer is depleted (there are couple of edge cases to handle). 

As for the 2), could you ensure the new frames start off from I-frames?

MushMal, 2020-03-17:
Also, would you please include the logs to understand whether INVALID_MKV_DATA is persistent and a transient issue that is caused by the fragment overflow?

jfguo, 2020-03-17:
@MushMal , thanks for the quick response.
I didn't dig into the video processing too long. So don't know whether the new frames start off from I or P frames. If I find it finally, I'll come back to you.

The following are some logs when the error happened. Once this happens, endless INVALID_MKV_DATA related messages are printed out quickly.

> 2020-03-17 14:52:53 [139982254028544] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: jianfengguo_laptop_camera and upload handle: 3 returned: {"EventType":"PERSISTED","FragmentTimecode":1584409968909,"FragmentNumber":"91343852333196777917216084829674756817675926618"}
> 
> 2020-03-17 14:52:53 [139982254028544] DEBUG - fragmentAckReceivedHandler invoked
> 2020-03-17 14:52:53 [139982254028544] DEBUG - postReadCallback(): Wrote 7960 bytes to Kinesis Video. Upload stream handle: 3
> 2020-03-17 14:52:53 [139982254028544] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 3
> DEBUG - postReadCallback(): Wrote 56283 bytes to Kinesis Video. Upload stream handle: 3
> 2020-03-17 14:52:54 [139982254028544] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 3
> WARN - curlCompleteSync(): curl perform failed for url https://s-ca658586.kinesisvideo.ap-southeast-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
> 2020-03-17 14:53:29 [139982254028544] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
> Request URL: https://s-ca658586.kinesisvideo.ap-southeast-2.amazonaws.com/putMedia
> Request Headers:
>     Authorization: AWS4-HMAC-SHA256 Credential=ASIAWBP62X6VLHEU365Z/20200317/ap-southeast-2/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=2e4d2028c26dd05ebdfb0904e30aa1b78939b
> 2020-03-17 14:53:29 [139982254028544] DEBUG - putStreamCurlHandler(): Network thread for Kinesis Video stream: jianfengguo_laptop_camera with upload handle: 3 exited. http status: 0
> 2020-03-17 14:53:29 [139982254028544] WARN - putStreamCurlHandler(): Stream with streamHandle 139982519989344 has exited without triggering end-of-stream. Service call result: 599
> 2020-03-17 14:53:29 [139982254028544] INFO - kinesisVideoStreamTerminated(): Stream 0x7f503860f460 terminated upload handle 3 with service call result 599.
> 2020-03-17 14:53:29 [139982254028544] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
> 2020-03-17 14:53:29 [139982254028544] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000001
> 2020-03-17 14:53:29 [139982254028544] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
> 2020-03-17 14:53:29 [139982254028544] INFO - writeHeaderCallback(): RequestId: 7e370b01-cb90-484d-b745-02589d3a74d2
> 2020-03-17 14:53:29 [139981902677760] DEBUG - describeStreamCurlHandler(): DescribeStream API response: { ................ }
> 2020-03-17 14:53:29 [139981902677760] INFO - describeStreamResultEvent(): Describe stream result event.
> 2020-03-17 14:53:29 [139981902677760] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
> 2020-03-17 14:53:29 [139981902677760] INFO - writeHeaderCallback(): RequestId: 679987b9-cd7b-46f7-b473-67ea5a2972ae
> 2020-03-17 14:53:29 [139982254028544] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-ca658586.kinesisvideo.ap-southeast-2.amazonaws.com"}
> 2020-03-17 14:53:29 [139982254028544] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
> 2020-03-17 14:53:29 [139982254028544] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
> 2020-03-17 14:53:29 [139982254028544] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
> 2020-03-17 14:53:29 [139982254028544] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
> 2020-03-17 14:53:29 [139982254028544] DEBUG - defaultStreamReadyCallback(): Reported streamReady callback for stream handle 139982519989344
> 2020-03-17 14:53:29 [139982254028544] DEBUG - streamReadyHandler invoked
> 2020-03-17 14:53:29 [139982254028544] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
> 2020-03-17 14:53:29 [139982254028544] INFO - putStreamResultEvent(): Put stream result event. New upload handle 4
> 2020-03-17 14:53:29 [139982254028544] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
> 2020-03-17 14:53:29 [139982254028544] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 4
> 2020-03-17 14:53:30 [139981902677760] INFO - writeHeaderCallback(): RequestId: f57f1fc9-dcc4-32fb-a395-30135e4bb432
> 2020-03-17 14:53:30 [139981902677760] DEBUG - postReadCallback(): Wrote 17977 bytes to Kinesis Video. Upload stream handle: 4
> 2020-03-17 14:53:30 [139981902677760] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 4
[... cut ...]
> 2020-03-17 14:53:35 [139981902677760] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 6
> DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 6
> 2020-03-17 14:53:35 [139981902677760] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 6
> 2020-03-17 14:53:35 [139981902677760] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: jianfengguo_laptop_camera and upload handle: 6 returned: {"EventType":"ERROR","ErrorCode":"INVALID_MKV_DATA","ErrorId":4006}
> ...

MushMal, 2020-03-17:
From the fragment of logs above, you are hitting case 3). The operation times out as you are not producing frames so the SDK is not sending any data and the backend is not returning any ACKs

WARN - curlCompleteSync(): curl perform failed for url https://s-ca658586.kinesisvideo.ap-southeast-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds

Here is the metrics printed once the frames resume
DEBUG - Kinesis Video client and stream metrics
	>> Overall storage byte size: 134217728
	>> Available storage byte size: 133446377
	>> Allocated storage byte size: 771351
	>> Total view allocation byte size: 144072
	>> Total streams frame rate (fps): 420
	>> Total streams transfer rate (bps): 3870800 (3780 Kbps)
	>> Current view duration (ms): 1776
	>> Overall view duration (ms): 32244
	>> Current view byte size: 683741
	>> Overall view byte size: 766980
	>> Current frame rate (fps): 420.402
	>> Current transfer rate (bps): 3870800 (3780 Kbps)

This indicates that you are 1.7 seconds behind the buffer - this is OK as it takes time for the state machine to re-create a new session and get it to the streaming state. What's kind of odd is that you are 32 seconds behind of persisted ACKs or you have no persistence configured for this stream which would fill out the buffer entirely. 

As the new frames come in the new session rolls back to re-stream either from rollback_duration or last persisted ACK (likely the former case as you have over 38 seconds in the overall buffer duration depending on the value of the rollback duration).

This then results in max duration breached.

The resulted error kicks off the re-streaming which results in invalid mkv errors.

While I am not entirely sure what's going on as the logs verbosity doesn't include DEBUG level to see what frames are produced, one thing that can cause the max duration reached is the case of the new frame coming in later which is not an I-frame which is case 2) as I mentioned before.

The logs cut after the second instance of invalid MKV so I don't see the timestamp.

Can I suggest the following:

* Ensure you use the latest commit in the SDK branches. Make sure to update the git submodules as well by running `git submodule update`
* Set debug level verbosity by setting AWS_KVS_LOG_LEVEL env variable to 2
* Specify whether the KVS plugin is used with the default parameters and if not which parameter was modified
* Fix the media pipeline to ensure the new frames start from an I-frame.

Please capture the logs so we can see what might be wrong.

MushMal, 2020-03-21:
@jfguo do you have any update on this?

MushMal, 2020-03-23:
Closing due to staleness.

closed 2020-03-23 by MushMal