awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #371: Significantly Reduced Streaming Performance with Kvssink and Audio

suggestedfixes, 2020-03-17:
System:
Ubuntu 18.04 LTS
i7 with 16GB of memory

Build:
48c68b92c957c67101e4d746ea96aa5468f7b1c1

Video Stream:
```
gst-launch-1.0 -v
videotestsrc ! x264enc ! kvssink stream-name=test storage-size=120 name=sink
```

Video Audio Stream:
```
gst-launch-1.0 -v
videotestsrc ! x264enc ! kvssink stream-name=test storage-size=120 name=sink
audiotestsrc ! audioconvert ! avenc_aac ! sink.
```

Symptom with Audio, significantly increased probability of full screen of 
"... Failed to submit frame to Kinesis Video client ... "

suggestedfixes, 2020-03-17:
In addition, with live mode with videojs, the chances of the screen loading increases significantly. To the point where the loading takes 99% of time.

MushMal, 2020-03-17:
@suggestedfixes could you please include the logs to see what's failing exactly? When streaming audio/video multi-track, the SDK handles the A/V sync which on some pipelines can cause errors due to time shifts between the tracks.

suggestedfixes, 2020-03-17:
Easier to reproduce with storage-size=10
Case 1: full screen of failed put frames
```
2020-03-16 19:47:00 [140487373616896] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
835221768 presentation timestamp: 15844132995648081
2020-03-16 19:47:00 [140487774107392] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
835333333 presentation timestamp: 15844132996092979
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
835453968 presentation timestamp: 15844132995880281
2020-03-16 19:47:00 [140487774107392] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
835666666 presentation timestamp: 15844132997759646
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
835686167 presentation timestamp: 15844132996112480
2020-03-16 19:47:00 [140487774107392] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
835918367 presentation timestamp: 15844132996344680
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
836000000 presentation timestamp: 15844132997092979
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
836150566 presentation timestamp: 15844132996576879
2020-03-16 19:47:00 [140487774107392] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
836333333 presentation timestamp: 15844132996759646
2020-03-16 19:47:00 [140487765714688] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 1
2020-03-16 19:47:00 [140487373616896] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
836382766 presentation timestamp: 15844132996809079
2020-03-16 19:47:00 [140487774107392] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 1
2020-03-16 19:47:00 [140487373616896] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 1
2020-03-16 19:47:00 [140487373616896] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 1
2020-03-16 19:47:00 [140487373616896] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 1
2020-03-16 19:47:00 [140487373616896] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 1
2020-03-16 19:47:00 [140487373616896] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 1
2020-03-16 19:47:00 [140487373616896] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
836614965 presentation timestamp: 15844132997041278
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
836666666 presentation timestamp: 15844132997426312
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
836847165 presentation timestamp: 15844132997273478
2020-03-16 19:47:00 [140487774107392] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
837000000 presentation timestamp: 15844132999092979
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
837079365 presentation timestamp: 15844132997505678
2020-03-16 19:47:00 [140487774107392] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 1
2020-03-16 19:47:00 [140487373616896] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
837311564 presentation timestamp: 15844132997737877
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
837333333 presentation timestamp: 15844132998426312
[... 226 lines cut ...]
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
847000000 presentation timestamp: 15844133007426312
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
847063945 presentation timestamp: 15844133007490258
2020-03-16 19:47:00 [140487774107392] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
847296145 presentation timestamp: 15844133007722458
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
847333333 presentation timestamp: 15844133008092979
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
[... cut ...]
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
847666666 presentation timestamp: 15844133009759646
2020-03-16 19:47:00 [140487765714688] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
847760544 presentation timestamp: 15844133008186857
2020-03-16 19:47:00 [140487774107392] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x5200002e decoding timestamp:
847992743 presentation timestamp: 15844133008419056

```

MushMal, 2020-03-17:
0x5200002e is returned when you are out-of-memory for the content store. A/V sync uses a temporary storage outside of the content store so the "held" frames for A/V sync won't be counted against the content store while they are being accumulated. However, once the appropriate frame from the other track is put, the remainder of the frames are "released" in a burst which could overwhelm a small content store.

What's the size of the frames that you are streaming? Could you ensure that you have enough storage for a couple of fragments to eliminate this issue?

The alternative is that the content store is filled due to longer buffer duration. We recently introduced a new API that allows setting buffer duration based on a pre-set content store size which is in the C SDK: https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/commit/4bf15b655b19f72c90712ea2564e57f9989f01be

suggestedfixes, 2020-03-17:
It would be awesome if the content store were leaky?
I'm generating random test sources with 1.5mbps. Having 8MB should hold around 42 second of data? After that shouldn't the kvssink start to dump extra fragments?
How to use the buffer duration with kvssink?

MushMal, 2020-03-17:
@suggestedfixes there are couple of things here. Let me cover them

Generally, you have two types of pressures 1) content store getting full 2) temporal (or buffer duration) pressure.

When we get the content store full we will have to error on the putFrame as there is a single content store that's shared among multiple streams. The decision we made was to fail the most recent frame in.

For the second case, you are operating within the stream itself that has the local knowledge for handling it. Here, the pressure will force the earliest frame (the tail) out. 

In the offline-mode the putFrame will block until there is availability on both space and buffer duration.

Here is a formula that can give some understanding of how much content store size you would need in order to support a given stream density for a given duration: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-reference-structures-producer.html

`Size = NumberOfStreams * AverageFrameSize * FramesPerSecond * BufferDurationInSeconds * DefragmentationFactor`

KVS sink has a list of parameters that you can configure. The best source of info is here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp#L450

suggestedfixes, 2020-03-17:
Since p1 is correlated to p2, wouldn't it make sense to have the flexibility of choose
Only p1, or Only p2? Having floor(p1, p2) is too confusing :)

When we get the content store full we will have to error on the putFrame as there is a single content store that's shared among multiple streams. The decision we made was to fail the most recent frame in.
- The way I think of content store, I think of it like a cache, so I would expect that this cache to evict a least important item, possibly the earliest. The part of pressure of forcing the earliest frame makes sense. Error out on the most recent put frame does not make sense.

In the offline-mode the putFrame will block until there is availability on both space and buffer duration.
- It would be awesome if we have an option to evict frames if we are out of space.

suggestedfixes, 2020-03-17:
In terms treating this like a producer and consumer problem, if the producer generates items at a faster rate than consumer (kvssink). And we block this storage. We might end up blocking forever. So we probably need to drop something?

MushMal, 2020-03-17:
@suggestedfixes things get a little more complicated when we have multiple streams. Content store storage pressure can be thought of as temporal cache or as a simple storage which can return out of memory. We do have a storage pressure notification which could be used to react on the pressure but it won't be easy to simply evict the earliest frame. We can certainly think about adding a flag to drive this behavior. In this case, the frames from this stream will be evicted (perhaps numerous earlier P-frames to make space for the encoded I-frame).

To respond to your second question. The putFrame thread will be blocked in the offline mode only. This happens when your source is a file source. In this case, the media thread will block until the networking thread (internal to KVS sink) will clear up enough space (temporal and/or storage) to unblock the media thread. If this isn't the case, the file source will quickly overwhelm the storage/buffer duration as it loads at the IO speed which is a lot faster than the network speed.

For the purpose of this issue though, I would leave the evicting the earliest frames behavior out of this issue and open a separate one. Is there anything else for this issue. If not, would you please go ahead and close this? Otherwise, please clarify if you need any other help.

MushMal, 2020-03-17:
Added an issue to the PIC repository to handle this: https://github.com/awslabs/amazon-kinesis-video-streams-pic/issues/22

suggestedfixes, 2020-03-17:
I think we went off topic a little. Having an eviction is certainly good. I think as you have suspected, the timestamps might have been out of sync. For example, the rtsp-timestamp-fixup might improve the synchronization between the two streams. I will provide realistic logs.

MushMal, 2020-03-17:
We rely on the GStreamer to do the timeline mapping and we then do AV/Sync. The main issue is that being a plugin we won't have a full exposure on which clocks the different streams use. For example, audio can use high-precision media clock while video might use system clock. Similarly, the implementation of the rtsp can timestamp based on the source or the destination. GStreamer has it's own logic on mapping the clocks.

suggestedfixes, 2020-03-17:
Is there a Gstreamer technique to sync the clocks for the kvssink? Probably time for me to read this I guess. https://gstreamer.freedesktop.org/documentation/application-development/advanced/clocks.html?gi-language=c

MushMal, 2020-03-17:
This is a good article describing the clocks and timestamps but it doesn't talk about how the sync works. We use collect pad that handles some of the work: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp#L605

gst_collect_pads_clip_running_time clips the frame timestamps to the running time and gst_kvs_sink_collect_compare then allows us to introduce custom comparator.

I am not super familiar with GStreamer intricacies so I am sure some folks would have a better understanding.

suggestedfixes, 2020-03-17:
After some skimming through gstreamer docs, I think it is probably possible to feed the same clock provider for video and audio. In terms of aac format, what are the recommended aac parameters?

MushMal, 2020-03-17:
It really depends on your scenario but in this case you would configure your upstream aac encoder in your application. Try the defaults first and see if it works for your scenario.

suggestedfixes, 2020-03-17:
Just figured it is possible to stream audio only using kvssink. I'm going to try audio only, video only, and streaming both together. There is a slight chance that converting audio is the bottleneck.

MushMal, 2020-03-17:
Normally, audio takes few CPU cycles even on a constrained device encoded in software. Please do some experimentation and see what works for your scenario. 

Please go ahead and resolve this issue if satisfied. Open a separate issue specific to the audio/video sync if any or related to the buffer issues.

suggestedfixes, 2020-03-17:
More realistic log entries with master branch:
```
verall storage byte size: 33554432
        >> Available storage byte size: 33372070
        >> Allocated storage byte size: 182362
        >> Total view allocation byte size: 576072
        >> Total streams frame rate (fps): 9865
        >> Total streams transfer rate (bps): 1997504 (1950 Kbps)
        >> Current view duration (ms): 700
        >> Overall view duration (ms): 1000
        >> Current view byte size: 103788
        >> Overall view byte size: 181018
        >> Current frame rate (fps): 9865.26
        >> Current transfer rate (bps): 1997504 (1950 Kbps)
2020-03-17 09:27:04 [140678742918912] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x32000002 decoding timestamp:
2650637827 presentation timestamp: 15844624246476753
2020-03-17 09:27:04 [140678742918912]
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstRtpSession:rtpsession2:
stats = application/x-rtp-session-stats, rtx-drop-count=(uint)0,
sent-nack-count=(uint)0, recv-nack-count=(uint)0, source-stats=(GValueArray)<
"application/x-rtp-source-stats\,\ ssrc\=\(uint\)3142888549\,\
internal\=\(boolean\)true\,\ validated\=\(boolean\)true\,\
received-bye\=\(boolean\)false\,\ is-csrc\=\(boolean\)false\,\
is-sender\=\(boolean\)false\,\ seqnum-base\=\(int\)-1\,\
clock-rate\=\(int\)-1\,\ octets-sent\=\(guint64\)0\,\
packets-sent\=\(guint64\)0\,\ octets-received\=\(guint64\)0\,\
packets-received\=\(guint64\)0\,\ bitrate\=\(guint64\)0\,\
packets-lost\=\(int\)0\,\ jitter\=\(uint\)0\,\ sent-pli-count\=\(uint\)0\,\
recv-pli-count\=\(uint\)0\,\ sent-fir-count\=\(uint\)0\,\
recv-fir-count\=\(uint\)0\,\ sent-nack-count\=\(uint\)0\,\
recv-nack-count\=\(uint\)0\,\ have-sr\=\(boolean\)false\,\
sr-ntptime\=\(guint64\)0\,\ sr-rtptime\=\(uint\)0\,\
sr-octet-count\=\(uint\)0\,\ sr-packet-count\=\(uint\)0\;" >,
rtx-count=(uint)0;
DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream
handle: 8
2020-03-17 09:27:04 [140678164117248] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:04 [140678164117248] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:04 [140678164117248] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:04 [140678164117248] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:04 [140678164117248] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:04 [140678164117248] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:04 [140678164117248] DEBUG - postReadCallback(): Wrote 6805
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:04 [140678164117248] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 8
2020-03-17 09:27:04 [140678164117248] DEBUG -
defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback
for stream handle 94545492640912. Current buffer duration in 100ns:
18446744073709151616
2020-03-17 09:27:04 [140678742918912] DEBUG - streamLatencyPressureHandler
invoked
2020-03-17 09:27:04 [140678742918912] DEBUG - Reported streamLatencyPressure
callback for stream handle 94545492640912. Current buffer duration in 100ns:
18446744073709151616
2020-03-17 09:27:04 [140678742918912] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: test and upload handle: 8 returned:
{"EventType":"RECEIVED","FragmentTimecode":1584462423721,"FragmentNumber":"91343852333189949524139309343498352002124588088"}

2020-03-17 09:27:04 [140678164117248] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-17 09:27:04 [140678164117248] DEBUG - postReadCallback(): Wrote 5757
bytes to Kinesis Video. Upload stream handle: 8
[... 485 lines cut ...]
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:09 [140678164117248] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:09 [140678164117248] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:09 [140678164117248] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:09 [140678164117248] DEBUG - postReadCallback(): Wrote 6033
bytes to Kinesis Video. Upload stream handle: 8
2020-03-17 09:27:09 [140678164117248] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 8
2020-03-17 09:27:09 [140678164117248] DEBUG -
defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback
for stream handle 94545492640912. Current buffer duration in 100ns:
18446744073709131616
2020-03-17 09:27:09 [140678742918912] DEBUG - streamLatencyPressureHandler
invoked
2020-03-17 09:27:09 [140678742918912] DEBUG - Reported streamLatencyPressure
callback for stream handle 94545492640912. Current buffer duration in 100ns:
18446744073709131616
2020-03-17 09:27:09 [140678742918912] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: test and upload handle: 8 returned:
{"EventType":"RECEIVED","FragmentTimecode":1584462428721,"FragmentNumber":"91343852333189949548898110129205958912324264321"}

```

suggestedfixes, 2020-03-17:
Test string used
```
gst-launch-1.0 -v \
rtspsrc location=rtsp://link protocols=tcp short-header=TRUE name=rsrc \
! rtph264depay ! video/x-h264,stream-format=avc,alignment=au ! h264parse ! kvssink stream-name=test storage-size=32 name=sink \
rsrc. ! rtppcmudepay ! mulawdec ! audioconvert ! avenc_aac ! sink. \
```

suggestedfixes, 2020-03-17:
I gave it a generous storage of 1024. I'm also observing a dejavu effect, where the audio replayed from 1 second before on live mode.

suggestedfixes, 2020-03-17:
What is code status: 0x32000002?

MushMal, 2020-03-18:
Few things going on. Just to let you know that I just published a new PR for the content store OOM policy: https://github.com/awslabs/amazon-kinesis-video-streams-pic/pull/24

Once this is in, we will re-base the C producer dependency on top of this commit and add more flexibility at the top layer to utilize these policies.

Regarding 0x32000002. All status codes are in the public Include.h files but can be also referenced from here (a little stale though)
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html

This is STATUS_MKV_INVALID_FRAME_TIMESTAMP which is generated when we encounter a frame timestamp which is less than the key-frame timestamp that precedes it. It's an indication that you might have b-frames in the stream.

Stream latency pressure is called when you accumulate more than max latency duration in the buffer. 

https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L869

So, even if you have enough storage, your SDK is configured to call stream latency pressure as soon as it accumulated more than max latency duration in the buffer without streaming it out.

In your logs you don't seem to have much buffer - only 450ms but for some reason (BUG BUG BUG) we trigger an underflow with the new frame as it reports 18446744073709131616 in the duration.

Could you please open a separate issue for this so we can track it separately from this issue?

Regarding this particular one. It seems that your rtsp stream is a little off

suggestedfixes, 2020-03-18:
If I see a bunch of those in a row, the observation is that there is a lot of dropped packets
```
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstRtpSession:rtpsession0:
stats = application/x-rtp-session-stats, rtx-drop-count=(uint)0,
sent-nack-count=(uint)0, recv-nack-count=(uint)0, source-stats=(GValueArray)<
"application/x-rtp-source-stats\,\ ssrc\=\(uint\)2057518414\,\
internal\=\(boolean\)false\,\ validated\=\(boolean\)true\,\
received-bye\=\(boolean\)false\,\ is-csrc\=\(boolean\)false\,\
is-sender\=\(boolean\)true\,\ seqnum-base\=\(int\)-1\,\
clock-rate\=\(int\)90000\,\ octets-sent\=\(guint64\)0\,\
packets-sent\=\(guint64\)0\,\ octets-received\=\(guint64\)30566650\,\
packets-received\=\(guint64\)23793\,\ bitrate\=\(guint64\)910240\,\
packets-lost\=\(int\)0\,\ jitter\=\(uint\)50\,\ sent-pli-count\=\(uint\)0\,\
recv-pli-count\=\(uint\)0\,\ sent-fir-count\=\(uint\)0\,\
recv-fir-count\=\(uint\)0\,\ sent-nack-count\=\(uint\)0\,\
recv-nack-count\=\(uint\)0\,\ have-sr\=\(boolean\)false\,\
sr-ntptime\=\(guint64\)0\,\ sr-rtptime\=\(uint\)0\,\
sr-octet-count\=\(uint\)0\,\ sr-packet-count\=\(uint\)0\,\
sent-rb\=\(boolean\)true\,\ sent-rb-fractionlost\=\(uint\)0\,\
sent-rb-packetslost\=\(int\)0\,\ sent-rb-exthighestseq\=\(uint\)23793\,\
sent-rb-jitter\=\(uint\)50\,\ sent-rb-lsr\=\(uint\)0\,\
sent-rb-dlsr\=\(uint\)0\,\ have-rb\=\(boolean\)false\,\
rb-fractionlost\=\(uint\)0\,\ rb-packetslost\=\(int\)0\,\
rb-exthighestseq\=\(uint\)0\,\ rb-jitter\=\(uint\)0\,\ rb-lsr\=\(uint\)0\,\
rb-dlsr\=\(uint\)0\,\ rb-round-trip\=\(uint\)0\;",
"application/x-rtp-source-stats\,\ ssrc\=\(uint\)4240445261\,\
internal\=\(boolean\)true\,\ validated\=\(boolean\)true\,\
received-bye\=\(boolean\)false\,\ is-csrc\=\(boolean\)false\,\
is-sender\=\(boolean\)false\,\ seqnum-base\=\(int\)-1\,\
clock-rate\=\(int\)-1\,\ octets-sent\=\(guint64\)0\,\
packets-sent\=\(guint64\)0\,\ octets-received\=\(guint64\)0\,\
packets-received\=\(guint64\)0\,\ bitrate\=\(guint64\)0\,\
packets-lost\=\(int\)0\,\ jitter\=\(uint\)0\,\ sent-pli-count\=\(uint\)0\,\
recv-pli-count\=\(uint\)0\,\ sent-fir-count\=\(uint\)0\,\
recv-fir-count\=\(uint\)0\,\ sent-nack-count\=\(uint\)0\,\
recv-nack-count\=\(uint\)0\,\ have-sr\=\(boolean\)false\,\
sr-ntptime\=\(guint64\)0\,\ sr-rtptime\=\(uint\)0\,\
sr-octet-count\=\(uint\)0\,\ sr-packet-count\=\(uint\)0\;" >,
rtx-count=(uint)0;
```

MushMal, 2020-03-18:
I am not an expert in the RTSP nor the RTSP element of the GStreamer. Perhaps you need to troubleshoot your device/network. GStreamer forums might be the best source for this information. There are some sample GStreamer pipelines in the Readmes that you might find useful too.

As this thread is getting too long and has a lot of branching, I would recommend closing this and opening a KVS specific issue if any.

suggestedfixes, 2020-03-19:
I found an audio clock that worked a little bit better, what remains unresolved is how do I get those clocks synced?
```
avenc_aac perfect-timestamp=TRUE
```

suggestedfixes, 2020-03-19:
I agree this is too general. Let me think about how to break this issue down to multiple issues.

MushMal, 2020-03-19:
@suggestedfixes take a look at the kvs plugin if you are building your app on top of gstreamer directly. See if you can use gstreamer functionality that I mentioned to map the timestamps to what gstreamer calls the running time. Basically, both tracks will start off from 0 - at least that's what the gstreamer will attempt to do.

I am going to resolve this issue. Please re-open or open separate issue for questions related to KVS.

closed 2020-03-19 by MushMal