awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #372: Spurious EOS signals

suggestedfixes, 2020-03-20:
System:
Ubuntu 18.04 LTS
i7 with 16GB of memory

Build:
48c68b9

gst-launch-1.0 -v 
rtspsrc location=rtsp://url short-header=TRUE ! rtph264depay ! video/x-h264,stream-format=avc,alignment=au ! h264parse ! kvssink stream-name=test storage-size=120

I understand that we can tweak parameters, but what are the main reasons for those streams ending?
```
2020-03-20 15:38:47 [140205046626048] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:38:47 [140205046626048] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:47 [140205046626048] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:47 [140205046626048] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:47 [140205046626048] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:47 [140205046626048] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:47 [140205046626048] DEBUG - postReadCallback(): Wrote 13525
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:47 [140205046626048] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 0
2020-03-20 15:38:47 [140205046626048] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: video-only and upload handle: 0
returned:
{"EventType":"RECEIVED","FragmentTimecode":1584743927375,"FragmentNumber":"91343852333181437705920711240162808583694468159"}

2020-03-20 15:38:48 [140205046626048] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:38:48 [140205046626048] DEBUG - postReadCallback(): Wrote 1537
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:48 [140205046626048] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 0
2020-03-20 15:38:48 [140205046626048] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: video-only and upload handle: 0
returned:
{"EventType":"BUFFERING","FragmentTimecode":1584743928375,"FragmentNumber":"91343852333181437710872471397304329753755836002"}

2020-03-20 15:38:48 [140205046626048] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:38:48 [140205046626048] DEBUG - postReadCallback(): Wrote 1790
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:48 [140205046626048] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 0
2020-03-20 15:38:48 [140205046626048] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: video-only and upload handle: 0
returned:
{"EventType":"PERSISTED","FragmentTimecode":1584743927375,"FragmentNumber":"91343852333181437705920711240162808583694468159"}

2020-03-20 15:38:48 [140205046626048] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:38:48 [140205046626048] DEBUG - postReadCallback(): Wrote 1446
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:48 [140205046626048] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 0
0:01:15.820094719  1259 0x7f842c00a9e0 INFO                    task
gsttask.c:316:gst_task_func:<rtpjitterbuffer0:src> Task going to paused
2020-03-20 15:38:48 [140205046626048] INFO - stopKinesisVideoStreamSync():
Synchronously stopping Kinesis Video Stream 000055c8f68c1b60.
2020-03-20 15:38:48 [140205885957888]
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstFunnel:funnel1.GstPad:src:
caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstRtpSession:rtpsession0.GstPad:sync_src:
caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstRtpSsrcDemux:rtpssrcdemux0.GstPad:rtcp_sink:
caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstRtpSession:rtpsession0.GstPad:recv_rtcp_sink:
caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstFunnel:funnel3.GstPad:src:
caps = application/x-rtcp
0:01:15.820363763  1259 0x55c8f6866230 INFO                    task
gsttask.c:316:gst_task_func:<task0> Task going to paused
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstRtpSession:rtpsession1.GstPad:sync_src:
caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstRtpSsrcDemux:rtpssrcdemux1.GstPad:rtcp_sink:
caps = application/x-rtcp
/GstPipeline:pipeline0/GstRTSPSrc:rsrc/GstRtpBin:manager/GstRtpSession:rtpsession1.GstPad:recv_rtcp_sink:
caps = application/x-rtcp
INFO - getStreamData(): Handle 0 waiting for last persisted ack with ts
15847439283750000
2020-03-20 15:38:49 [140205046626048] DEBUG - postReadCallback(): Wrote 11126
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:49 [140205046626048] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 0 waiting for last ack.
2020-03-20 15:38:49 [140205046626048] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: video-only and upload handle: 0
returned:
{"EventType":"RECEIVED","FragmentTimecode":1584743928375,"FragmentNumber":"91343852333181437710872471397304329753755836002"}

2020-03-20 15:38:49 [140205046626048] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:38:49 [140205046626048] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: video-only and upload handle: 0
returned:
{"EventType":"PERSISTED","FragmentTimecode":1584743928375,"FragmentNumber":"91343852333181437710872471397304329753755836002"}

2020-03-20 15:38:49 [140205046626048] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:38:49 [140205046626048] INFO - getStreamData(): Indicating an EOS
after last persisted ACK is received for stream upload handle 0
2020-03-20 15:38:49 [140205046626048] DEBUG - defaultStreamClosedCallback():
Reported streamClosed callback for stream handle 94321618197344. Upload handle
0
2020-03-20 15:38:49 [140205046626048] DEBUG - streamClosedHandler invoked for
upload handle: 0
2020-03-20 15:38:49 [140205046626048] DEBUG - Reported streamClosed callback
for stream handle 94321618197344. Upload handle 0
2020-03-20 15:38:49 [140205046626048] INFO - postReadCallback(): Reported
end-of-stream for stream video-only. Upload handle: 0
2020-03-20 15:38:49 [140205046626048] DEBUG - postReadCallback(): Wrote 0 bytes
to Kinesis Video. Upload stream handle: 0
2020-03-20 15:38:49 [140205046626048] INFO - Sending eos
0:01:16.721534366  1259 0x55c8f68661e0 INFO                    task
gsttask.c:316:gst_task_func:<queue0:src> Task going to paused
2020-03-20 15:38:49 [140205885957888] Got EOS from element "pipeline0".
Execution ended after 0:01:15.505255773
Setting pipeline to PAUSED ...

```

suggestedfixes, 2020-03-20:
```
2020-03-20 15:44:44 [139828012242688] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:44:44 [139828012242688] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: video-only and upload handle: 0
returned:
{"EventType":"PERSISTED","FragmentTimecode":1584744283635,"FragmentNumber":"91343852333181442830992473881637242339967773148"}

2020-03-20 15:44:44 [139828012242688] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:44:44 [139828012242688] DEBUG - postReadCallback(): Wrote 4997
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:44 [139828012242688] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 0
2020-03-20 15:44:44 [139828012242688] DEBUG - Kinesis Video client and stream
metrics
        >> Overall storage byte size: 104857600
        >> Available storage byte size: 104646341
        >> Allocated storage byte size: 211259
        >> Total view allocation byte size: 144072
        >> Total streams frame rate (fps): 20
        >> Total streams transfer rate (bps): 2040064 (1992 Kbps)
        >> Current view duration (ms): 501
        >> Overall view duration (ms): 1000
        >> Current view byte size: 106433
        >> Overall view byte size: 210251
        >> Current frame rate (fps): 20.2518
        >> Current transfer rate (bps): 2040064 (1992 Kbps)
2020-03-20 15:44:45 [139828477490944] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Wrote 9570
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: video-only and upload handle: 0
returned:
{"EventType":"RECEIVED","FragmentTimecode":1584744284635,"FragmentNumber":"91343852333181442835944234038778763503893255502"}

2020-03-20 15:44:45 [139828012242688] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Wrote 1793
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: video-only and upload handle: 0
returned:
{"EventType":"BUFFERING","FragmentTimecode":1584744285635,"FragmentNumber":"91343852333181442840895994195920284929446941365"}

2020-03-20 15:44:45 [139828012242688] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Wrote 1740
bytes to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:45 [139828012242688] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 0
2020-03-20 15:44:45 [139828012242688] INFO - stopKinesisVideoStreamSync():
Synchronously stopping Kinesis Video Stream 00005627963b1400.
[... 113 lines cut ...]
{"EventType":"PERSISTED","FragmentTimecode":1584744285635,"FragmentNumber":"91343852333181442840895994195920284929446941365"}

2020-03-20 15:44:46 [139828012242688] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:44:46 [139828012242688] INFO - getStreamData(): Indicating an EOS
after last persisted ACK is received for stream upload handle 0
2020-03-20 15:44:46 [139828012242688] DEBUG - defaultStreamClosedCallback():
Reported streamClosed callback for stream handle 94728024167424. Upload handle
0
2020-03-20 15:44:46 [139828012242688] DEBUG - streamClosedHandler invoked for
upload handle: 0
2020-03-20 15:44:46 [139828012242688] DEBUG - Reported streamClosed callback
for stream handle 94728024167424. Upload handle 0
2020-03-20 15:44:46 [139828012242688] INFO - postReadCallback(): Reported
end-of-stream for stream video-only. Upload handle: 0
2020-03-20 15:44:46 [139828012242688] DEBUG - postReadCallback(): Wrote 0 bytes
to Kinesis Video. Upload stream handle: 0
2020-03-20 15:44:46 [139828012242688] INFO - Sending eos
0:01:52.361656663  1331 0x7f2c5c034de0 INFO                    task
gsttask.c:316:gst_task_func:<rtpjitterbuffer0:src> Task going to paused
2020-03-20 15:44:46 [139828477490944] Got EOS from element "pipeline0".
Execution ended after 0:01:51.162844420
Setting pipeline to PAUSED ...

```

suggestedfixes, 2020-03-20:
```
        >> Overall storage byte size: 12582912
        >> Available storage byte size: 12367914
        >> Allocated storage byte size: 214998
        >> Total view allocation byte size: 576072
        >> Total streams frame rate (fps): 231
        >> Total streams transfer rate (bps): 2282744 (2229 Kbps)
        >> Current view duration (ms): 600
        >> Overall view duration (ms): 1000
        >> Current view byte size: 111944
        >> Overall view byte size: 213702
        >> Current frame rate (fps): 231.155
        >> Current transfer rate (bps): 2282744 (2229 Kbps)
2020-03-20 15:48:34 [140118624122624] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x32000002 decoding timestamp:
1069500336 presentation timestamp: 15847445146886375
2020-03-20 15:48:34 [140118624122624] WARN - putKinesisVideoFrame(): Failed to
submit frame to Kinesis Video client. status: 0x32000002 decoding timestamp:
1070780336 presentation timestamp: 15847445148166375
2020-03-20 15:48:34 [140118624122624] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Wrote 16372
bytes to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Wrote 9273
bytes to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG -
defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback
for stream handle 93954716590112. Current buffer duration in 100ns:
18446744073707401616
2020-03-20 15:48:34 [140118624122624] DEBUG - streamLatencyPressureHandler
invoked
2020-03-20 15:48:34 [140118624122624] DEBUG - Reported streamLatencyPressure
callback for stream handle 93954716590112. Current buffer duration in 100ns:
18446744073707401616
2020-03-20 15:48:34 [140118624122624] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: test and upload handle: 3 returned:
{"EventType":"RECEIVED","FragmentTimecode":1584744513887,"FragmentNumber":"91343852333191697450908938886248542626426072692"}

2020-03-20 15:48:34 [140118031574784] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Wrote 6418
bytes to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postReadCallback(): Pausing CURL
read for upload handle: 3
2020-03-20 15:48:34 [140118031574784] DEBUG - postWriteCallback(): Curl post
body write function for stream with handle: test and upload handle: 3 returned:
{"EventType":"BUFFERING","FragmentTimecode":1584744514887,"FragmentNumber":"91343852333191697455860699043390064069604354508"}

2020-03-20 15:48:34 [140118031574784] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:48:34 [140118031574784] DEBUG -
defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback
for stream handle 93954716590112. Current buffer duration in 100ns:
18446744073707181616
2020-03-20 15:48:34 [140118624122624] DEBUG - streamLatencyPressureHandler
invoked
2020-03-20 15:48:34 [140118624122624] DEBUG - Reported streamLatencyPressure
callback for stream handle 93954716590112. Current buffer duration in 100ns:
[... 141 lines cut ...]
{"EventType":"PERSISTED","FragmentTimecode":1584744515887,"FragmentNumber":"91343852333191697460812459200531585512822858715"}

2020-03-20 15:48:36 [140118031574784] DEBUG - fragmentAckReceivedHandler
invoked
2020-03-20 15:48:36 [140118031574784] INFO - getStreamData(): Indicating an EOS
after last persisted ACK is received for stream upload handle 3
2020-03-20 15:48:36 [140118031574784] DEBUG - defaultStreamClosedCallback():
Reported streamClosed callback for stream handle 93954716590112. Upload handle
3
2020-03-20 15:48:36 [140118031574784] DEBUG - streamClosedHandler invoked for
upload handle: 3
2020-03-20 15:48:36 [140118031574784] DEBUG - Reported streamClosed callback
for stream handle 93954716590112. Upload handle 3
2020-03-20 15:48:36 [140118031574784] INFO - postReadCallback(): Reported
end-of-stream for stream test. Upload handle: 3
2020-03-20 15:48:36 [140118031574784] DEBUG - postReadCallback(): Wrote 0 bytes
to Kinesis Video. Upload stream handle: 3
2020-03-20 15:48:36 [140118031574784] INFO - Sending eos
0:01:50.699492745  1423 0x7f6fdc0492d0 INFO                    task
gsttask.c:316:gst_task_func:<rtpjitterbuffer2:src> Task going to paused
2020-03-20 15:48:36 [140118039967488] Got EOS from element "pipeline0".
Execution ended after 0:01:49.215529421
Setting pipeline to PAUSED ...

```

MushMal, 2020-03-20:
@suggestedfixes the EOS is an internal constant that's communicated from the PIC indicating to the upper layer (C -SDK) that the session should terminate and the resource associated to the session cleaned. In your case your pipeline is tearing down the GST plugin resulting in the EOS during the teardown.

2020-03-20 15:38:48 [140205046626048] INFO - stopKinesisVideoStreamSync():
Synchronously stopping Kinesis Video Stream 000055c8f68c1b60.

Incidentally, you are also getting 0x32000002 STATUS_MKV_INVALID_FRAME_TIMESTAMP which indicates that you have a time drift in your source - the frame that's produced has a timestamp that's less than it's fragment starts timestamp. This can happen when the encoder is bad, or gets restarted - which might be your case. Occasionally, you can see this in streams that encode b-frames.

Once you get this, the content view gets confused and we end up with underflow which causes latency pressure callback to fire.

I would like us to get to the bottom of this issue. Could you please describe in details what your scenarios is? What's the pipeline, what asset are you using (GST plugin?), what device you are using, 

I don't have a full log to understand the timeline of events. Perhaps the EoS is a consequence of the plugin itself reacting on bad state or something extraordinary that's happening. 

Please share the details and the full log so we could look into it.

suggestedfixes, 2020-03-21:
I was forwarding a device specific rtsp stream to a PC, let me see if I can replicate it on a public rtsp test source.
```
gst-launch-1.0 -v
rtspsrc location=rtsp://url short-header=TRUE ! rtph264depay ! video/x-h264,stream-format=avc,alignment=au ! h264parse ! kvssink stream-name=test storage-size=120
```
But having timestamp confusion that causes an eos signal sounds too catastrophic. I understand that the WebRTC repo drops a lot of frames with invalid time stamps, I wonder if this is the case here. Or better yet, a mechanism to do some timestamp correction.

So latency pressure causes the program to stop? Also I was reading on gstreamer and it appears to be latency is taken into account when clocks are being autocorrected?

suggestedfixes, 2020-03-21:
I have a brilliant idea, I should test the stream with a fake sink first.

suggestedfixes, 2020-03-21:
Then we can narrow down if it is a kvssink issue or gstreamer issue.

suggestedfixes, 2020-03-21:
Found relevant code
```
    isDroppable =   GST_BUFFER_FLAG_IS_SET(buf, GST_BUFFER_FLAG_CORRUPTED) ||
                    GST_BUFFER_FLAG_IS_SET(buf, GST_BUFFER_FLAG_DECODE_ONLY) ||
                    (GST_BUFFER_FLAGS(buf) == GST_BUFFER_FLAG_DISCONT) ||
                    (GST_BUFFER_FLAG_IS_SET(buf, GST_BUFFER_FLAG_DISCONT) && GST_BUFFER_FLAG_IS_SET(buf, GST_BUFFER_FLAG_DELTA_UNIT)) ||
                    // drop if buffer contains header and has invalid timestamp
                    (GST_BUFFER_FLAG_IS_SET(buf, GST_BUFFER_FLAG_HEADER) && (!GST_BUFFER_PTS_IS_VALID(buf) || !GST_BUFFER_DTS_IS_VALID(buf)));

    if (isDroppable) {
        goto CleanUp;
    }
```

suggestedfixes, 2020-03-21:
Never mind, every frame goes through the same cleanup. I'm wrong.

suggestedfixes, 2020-03-21:
Are there any edge case where bad network leads to toggling between offline mode and online mode, and this leads to clock switching and error 0x32000002?

suggestedfixes, 2020-03-21:
I forgot the details of C++
```
frame->decodingTs = static_cast<UINT64>(dts.count()) / DEFAULT_TIME_UNIT_IN_NANOS;
frame->presentationTs = static_cast<UINT64>(pts.count()) / DEFAULT_TIME_UNIT_IN_NANOS;
```
Those get downgraded to signed int by first glimpse. But I could be wrong.

suggestedfixes, 2020-03-21:
Yep I'm wrong, wrote the following and got the right output
```
cout << static_cast<unsigned long long>(1e18) / 2 << endl;
```

suggestedfixes, 2020-03-21:
#define DEFAULT_RESTART_ON_ERROR TRUE
So this should restart the kvssink correct?

MushMal, 2020-03-21:
Sorry for the delay. Let me cover a few questions. The core of the SDK checks for the monotonically increasing timestamps - this is necessary for semantic validity of the MKV format, streamability and our backend enforces it. While I am not entirely sure why your rtsp source produces time drifts, I can guess that due to it's network conditions, it might be re-initializing. Perhaps this is why we see the pipeline restart. Restarting the pipeline is the only way that we are getting stopKinesisVideoStreamSync() call and it's not dependent on anything else.

We will have to investigate and fix the source of underflow in the PIC - though, this is a red herring and won't be relevant to this issue.

My recommendation is to run with GStreamer debug logging (set GST_DEBUG=4 environment variable) and run it with fake sink to eliminate KVS. This said, you might also want to ensure you have adequate bandwidth for the device as it is sharing the network connection ingesting RTSP and streaming out to KVS.

I will open a separate issue tracking the underflow: https://github.com/awslabs/amazon-kinesis-video-streams-pic/issues/33

suggestedfixes, 2020-03-21:
Thank you very much for your help! I also need to ramp up on how kvssink works to provide you with useful log. I feel like I don't understand it enough to provide useful information.

MushMal, 2020-03-21:
No worries at all, hope we can help.

MushMal, 2020-03-23:
Wonder if you were able to get more details from GStreamer debug logs on this issue?

suggestedfixes, 2020-03-25:
Almost forgot about this issue, I will get more examples.

MushMal, 2020-04-06:
@suggestedfixes Any updates on this issue?

MushMal, 2020-04-19:
I am going to resolve this issue. We have an ongoing investigation with regards to RTSP element producing some non-monotonic timestamps tracked at https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/386

closed 2020-04-19 by MushMal