awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #374: Process Alive while Stopped Streaming

suggestedfixes, 2020-03-23:
Reported by a colleague, after running a 2Mbps stream after 2 day:
```
allback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 9340000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - pts: 15834629316277564, dts: 1311608562492, duration: 0, size: 1951, trackId: 1, isKey: 0
2020-03-06 10:48:51 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 9000000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - pts: 15834629316610870, dts: 1311608564438, duration: 0, size: 1544, trackId: 1, isKey: 0
2020-03-06 10:48:51 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 8340000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - pts: 15834629317277509, dts: 1311608597484, duration: 0, size: 1960, trackId: 1, isKey: 0
2020-03-06 10:48:51 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 8000000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - pts: 15834629317610815, dts: 1311608894928, duration: 0, size: 1485, trackId: 1, isKey: 0
2020-03-06 10:48:51 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 7340000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 77
2020-03-06 10:48:51 [1784673360] DEBUG - pts: 15834629318277454, dts: 1311609581321, duration: 0, size: 2000, trackId: 1, isKey: 0
2020-03-06 10:48:51 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 7000000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 77
2020-03-06 10:48:51 [1784673360] DEBUG - pts: 15834629318610803, dts: 1311609923948, duration: 0, size: 1720, trackId: 1, isKey: 0
2020-03-06 10:48:51 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 6340000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - pts: 15834629319277491, dts: 1311610592901, duration: 0, size: 1792, trackId: 1, isKey: 0
2020-03-06 10:48:51 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 6000000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - pts: 15834629319610846, dts: 1311610927417, duration: 0, size: 1519, trackId: 1, isKey: 0
2020-03-06 10:48:51 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 5340000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - pts: 15834629320277534, dts: 1311611591293, duration: 0, size: 2128, trackId: 1, isKey: 0
2020-03-06 10:48:51 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 5000000
2020-03-06 10:48:51 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:51 [1818227792] DEBUG - pts: 15834629320610883, dts: 1311611892907, duration: 0, size: 2029, trackId: 1, isKey: 0
2020-03-06 10:48:52 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 4340000
2020-03-06 10:48:52 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:52 [1818227792] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 77
2020-03-06 10:48:52 [1784673360] DEBUG - pts: 15834629321277565, dts: 1311612569737, duration: 0, size: 2356, trackId: 1, isKey: 0
2020-03-06 10:48:52 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 4000000
2020-03-06 10:48:52 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:52 [1818227792] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 77
2020-03-06 10:48:52 [1784673360] DEBUG - pts: 15834629321610914, dts: 1311612915132, duration: 0, size: 2586, trackId: 1, isKey: 0
2020-03-06 10:48:52 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 3340000
2020-03-06 10:48:52 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:52 [1818227792] DEBUG - pts: 15834629322277596, dts: 1311613579227, duration: 0, size: 2259, trackId: 1, isKey: 0
2020-03-06 10:48:52 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 3000000
2020-03-06 10:48:52 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:52 [1818227792] DEBUG - pts: 15834629322610944, dts: 1311613916942, duration: 0, size: 1482, trackId: 1, isKey: 0
2020-03-06 10:48:52 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 2340000
2020-03-06 10:48:52 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:52 [1818227792] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 77
2020-03-06 10:48:52 [1784673360] DEBUG - pts: 15834629323277626, dts: 1311614590894, duration: 0, size: 2514, trackId: 1, isKey: 0
2020-03-06 10:48:52 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 2000000
2020-03-06 10:48:52 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:52 [1818227792] DEBUG - pts: 15834629323610974, dts: 1311614929264, duration: 0, size: 2124, trackId: 1, isKey: 0
2020-03-06 10:48:52 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 1340000
2020-03-06 10:48:52 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:52 [1818227792] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 77
2020-03-06 10:48:52 [1784673360] DEBUG - pts: 15834629324277656, dts: 1311615603756, duration: 0, size: 2383, trackId: 1, isKey: 0
2020-03-06 10:48:52 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 1000000
2020-03-06 10:48:52 [1818227792] DEBUG - bufferDurationOverflowPressureHandler invoked
2020-03-06 10:48:52 [1818227792] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 77
2020-03-06 10:48:52 [1784673360] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 77
2020-03-06 10:48:52 [1784673360] DEBUG - pts: 15834629324611004, dts: 1311617533483, duration: 0, size: 197886, trackId: 1, isKey: 1
2020-03-06 10:48:52 [1818227792] DEBUG - defaultBufferDurationOverflowCallback(): Reported bufferDurationOverflow callback for stream handle 6635397. Remaining duration in 100ns: 340000
[... cut ...]
2020-03-06 10:48:56 [1818227792] DEBUG - defaultDroppedFrameReportCallback(): Reported droppedFrame callback for stream handle 6635397. Dropped frame timecode in 100ns: 1311357520000
2020-03-06 10:48:56 [1818227792] DEBUG - droppedFrameReportHandler invoked
2020-03-06 10:48:56 [1818227792] WARN - Reported droppedFrame callback for stream handle 6635397. Dropped frame timecode in 100ns: 1311357520000
2020-03-06 10:48:56 [1818227792] WARN - curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty

```

MushMal, 2020-03-23:
@suggestedfixes could you please clarify what the issue is? From the log all I can see is that the current bandwidth of ~780Kbps is not enough to stream out and we accumulate the buffer which results in buffer overflow callbacks and eventually dropped frame.

Is this an application using KVS GStreamer plugin or gst-launch-1.0 call? How does the pipeline stop? Is it being terminated somehow but doesn't quit? Do we have a suspicion of a deadlock? Which commit are you using?

suggestedfixes, 2020-03-23:
This is an application based on rtsp demo "https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/demo/kvs_producer_plugin_rtsp_demo.cpp"

The issue is, there isn't any log activity, but the process is alive, but not uploading streams. I would say the commit is roughly a few commits around Mar. 6. There is a possibility of a deadlock.

suggestedfixes, 2020-03-23:
This may be relevant. My colleague's major concern is whether we could grab the status of putFrame from the application layer using a mechanism such as a callback function etc.

MushMal, 2020-03-23:
OK, seems that the application is trying to shutdown and this might cause a deadlock?

Would you mind attaching the complete logs? Is there a way this can be run under debugger or somehow core dump/stack traces could be generated for the threads in the application? The reason I am asking is that we are unable to hit this case in the scripted unit tests and I can't tell what's really going on.

PutFrame itself returns a status code. The main issue is that in the context of the KVS plugin, it's hard to "bubble up" the errors. We are considering adding GStreamer events when errors happen and on callbacks but haven't yet implemented it.

suggestedfixes, 2020-03-23:
A typical gstreamer application allows us to handle errors by specifying callbacks. We wonder if kvssink sends those error signals to application layer. By rough glimpse of the kvssink src, it might be able to capture some custom events by kvssink.

suggestedfixes, 2020-03-23:
This is the only log we have. We can try to duplicate the issue again.
Since this is an issue from a while ago, this PR will decrease the likelihood of a similar issue, correct? https://github.com/awslabs/amazon-kinesis-video-streams-pic/issues/22

suggestedfixes, 2020-03-23:
How difficult would it be to implement GStreamer events on putFrame status?

MushMal, 2020-03-23:
We need to ensure that the latest GStreamer is pulling the right dependent commit to make the tail eviction by default. However, this is not a stability fix. What you are describing makes me think that you might have hit an edge case that we haven't accounted for and there is either a live lock or some dead lock that the plugin is stuck on a teardown path. 

As soon as we have general GStreamer event support we should be able to expose all of these without any issues. Unfortunately, I don't have a timeline for now.

suggestedfixes, 2020-03-23:
Great find on the lock case!

suggestedfixes, 2020-03-23:
Added issue
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/375

MushMal, 2020-03-23:
Thank you. I will update the #375 as we make progress. Meanwhile, pleas please please keep an eye on this issue and if you could attach a debugger and dump the stack traces of all of the active threads then we can analyze and see what's going on.

suggestedfixes, 2020-03-25:
@MushMal 
New log from Colleague
```
2020-03-25 00:30:16 [1819276368] DEBUG - pts: 15850674183867075, dts: 301674952697, duration: 0, size: 182991, trackId: 1, isKey: 1
2020-03-25 00:30:16 [1819276368] DEBUG - Kinesis Video client and stream metrics
        >> Overall storage byte size: 8388608
        >> Available storage byte size: 2971858
        >> Allocated storage byte size: 5416750
        >> Total view allocation byte size: 36064
        >> Total streams frame rate (fps): 25
        >> Total streams transfer rate (bps): 2588320 (2527 Kbps)
        >> Current view duration (ms): 18700
        >> Overall view duration (ms): 22000
        >> Current view byte size: 4495890
        >> Overall view byte size: 5398669
        >> Current frame rate (fps): 25.675
        >> Current transfer rate (bps): 2588320 (2527 Kbps)
2020-03-25 00:30:16 [1819276368] DEBUG - pts: 15850674184200408, dts: 301674992308, duration: 0, size: 1153, trackId: 1, isKey: 0
2020-03-25 00:30:16 [1819276368] DEBUG - pts: 15850674184867076, dts: 301675270124, duration: 0, size: 2377, trackId: 1, isKey: 0
2020-03-25 00:30:16 [1819276368] DEBUG - pts: 15850674185200411, dts: 301675605474, duration: 0, size: 3398, trackId: 1, isKey: 0
2020-03-25 00:30:16 [1819276368] DEBUG - pts: 15850674185867079, dts: 301676272469, duration: 0, size: 3916, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674186200414, dts: 301676603708, duration: 0, size: 2521, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674186867083, dts: 301677271816, duration: 0, size: 3045, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674187200418, dts: 301677609346, duration: 0, size: 2799, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674187867086, dts: 301678279226, duration: 0, size: 3188, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674188200421, dts: 301678607830, duration: 0, size: 2868, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674188867089, dts: 301679277387, duration: 0, size: 3264, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674189200424, dts: 301679608149, duration: 0, size: 3317, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674189867093, dts: 301680278548, duration: 0, size: 3018, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674190200428, dts: 301680606341, duration: 0, size: 3322, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674190867096, dts: 301681277673, duration: 0, size: 3027, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674191200431, dts: 301681606789, duration: 0, size: 3688, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674191867100, dts: 301682276197, duration: 0, size: 3019, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674192200435, dts: 301682611875, duration: 0, size: 3765, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674192867103, dts: 301683271150, duration: 0, size: 3223, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674193200438, dts: 301683604963, duration: 0, size: 3331, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674193867107, dts: 301684923498, duration: 0, size: 178322, trackId: 1, isKey: 1
2020-03-25 00:30:17 [1819276368] DEBUG - Kinesis Video client and stream metrics
        >> Overall storage byte size: 8388608
        >> Available storage byte size: 2734199
        >> Allocated storage byte size: 5654409
        >> Total view allocation byte size: 36064
        >> Total streams frame rate (fps): 25
        >> Total streams transfer rate (bps): 2588320 (2527 Kbps)
        >> Current view duration (ms): 19700
        >> Overall view duration (ms): 23000
        >> Current view byte size: 4732729
        >> Overall view byte size: 5635508
        >> Current frame rate (fps): 25.8469
        >> Current transfer rate (bps): 2588320 (2527 Kbps)
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674194200442, dts: 301684962241, duration: 0, size: 1393, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674194867117, dts: 301685270423, duration: 0, size: 2294, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674195200455, dts: 301685602635, duration: 0, size: 2298, trackId: 1, isKey: 0
2020-03-25 00:30:17 [1819276368] DEBUG - pts: 15850674195867126, dts: 301686272980, duration: 0, size: 3907, trackId: 1, isKey: 0
2020-03-25 00:30:18 [1819276368] DEBUG - pts: 15850674196200465, dts: 301686603623, duration: 0, size: 2636, trackId: 1, isKey: 0
2020-03-25 00:30:18 [1819276368] DEBUG - pts: 15850674196867136, dts: 301687270144, duration: 0, size: 2902, trackId: 1, isKey: 0
2020-03-25 00:30:18 [1819276368] DEBUG - pts: 15850674197200474, dts: 301687608672, duration: 0, size: 2754, trackId: 1, isKey: 0
2020-03-25 00:30:18 [1819276368] DEBUG - pts: 15850674197867146, dts: 301688275592, duration: 0, size: 3092, trackId: 1, isKey: 0
2020-03-25 00:30:18 [1819276368] DEBUG - pts: 15850674198200484, dts: 301688605364, duration: 0, size: 2860, trackId: 1, isKey: 0
2020-03-25 00:30:18 [1819276368] DEBUG - pts: 15850674198867155, dts: 301689276756, duration: 0, size: 3292, trackId: 1, isKey: 0
[... cut ...]
2020-03-25 00:30:26 [1819276368] DEBUG - bufferDurationOverflowPressureHandler invoked
--Type <RET> for more, q to quit, c to continue without paging--

Thread 1039 "rtpjitterbuffer" received signal SIGPIPE, Broken pipe.
[Switching to LWP 3637]
0x76ba98b2 in write () from /lib/libc.so.6
(gdb) bt full
#0  0x76ba98b2 in write () from /lib/libc.so.6
No symbol table info available.
#1  0x76850c7a in ?? () from /mnt/plugin/vigilcamera/lib/libcrypto.so.1.1
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

```

suggestedfixes, 2020-03-25:
This is using a stripped-down version of gdb, so some symbols are missing.

suggestedfixes, 2020-03-25:
The new log should have been in a separate issue since the program crashed.
Still running 2Mbps.

MushMal, 2020-03-25:
@suggestedfixes here is what I see in the latest log you've attached

The streaming buffer falls behind due to bandwidth constrains and starts accumulating the frames and falling behind. At the last moment, you have close to 30 seconds of content in the buffer. Moreover, it seems that the network is stale as there is no bits are being pushed up and no ACKs are being received.

The SDK starts dropping frames due to the buffer pressure (based on temporal duration).

What happens next is not entirely clear to me. It seems that the process is broken into a debugger by someone pressing Ctrl+C. Was this interrupted as the logging stopped right there and after a while it was pressed?

The debugger contains the stack trace of a single thread that's not useful. 

Here are some recommendations to get to the bottom of this.

* Ensure we have the full scenario description. What exactly is happening. From what I understand, the KVS plugin in the context of your application streams fine, then, as it hits network issues, it accumulates the frames and as it drops the frames it gets in a deadlock?! Please clariy.
* Compile the debug version of the SDK and your app so we can get symbols
* Run the application with the GStreamer logs as well - `GST_DEBUG=4` environment variable
* Dump out all of the stacks by issuing `thread apply all bt`
* If this is network related issue, it's easy to reproduce by simply stopping the network interface. Simplest way is to turn the WiFi off for example.

suggestedfixes, 2020-03-29:
Sorry I dad been juggling between different issues. I will get back to you on this one ASAP.

sharadrj, 2020-03-29:
@suggestedfixes, can you update your code to latest master and try again (make sure submodules are updated also). We have some significant changes in buffering and frame drop logic which can help.

MushMal, 2020-04-16:
@suggestedfixes do we have more update on this? Can this be reproduced with fakesink?

suggestedfixes, 2020-04-23:
I forgot about this issue. Back to reproducing this.

MushMal, 2020-06-26:
I am resolving this due to staleness. Please cut a new issue with the details if this continues to happen

closed 2020-06-26 by MushMal