awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #379: Improve Documentation on IoT

suggestedfixes, 2020-03-24:
Currently I'm following this tutorial on IoT
```
https://github.com/awsdocs/AWS-Kinesis-Video-Documentation/blob/master/doc_source/how-iot.md
```

The final step involves putting in a test string such as this
```
$ gst-launch-1.0 rtspsrc location=rtsp://YourCameraRtspUrl short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au !
 h264parse ! kvssink stream-name="kvs_example_camera_stream" iot-certificate="iot-certificate,endpoint=iot-credential-endpoint-host-name,cert-path=/path/to/certificate.pem,key-path=/path/to/private.pem.key,ca-path=/path/to/cacert.pem,role-aliases=KvsCameraIoTRoleAlias"
```

Currently issue I'm experiencing is I'm getting unexpected ASSIGNMENT error. 
of the iot-certificate string, what are ambigious are iot-certificate, iot-credential-endpoint-host-name. For example endpoint could have a prefix of https:// or not. And I suspect iot-certificate should be assigned to a path.

Could someone verify the accuracy of the IoT instructions?

chehefen, 2020-03-24:
the endpoint should be whatever `aws iot describe-endpoint --endpoint-type iot:CredentialProvider` returns.

suggestedfixes, 2020-03-24:
I think it's how gstreamer parsing comma delimited string that's an issue. Maybe someone can give it a shot?

suggestedfixes, 2020-03-24:
I got the same set of certificates working for WebRTC, so it's not a mis-configuration issue on the AWS side. In addition, it is the gstreamer that is complaining about the input format.

chehefen, 2020-03-24:
I ran this and I didnt get a ASSIGNMENT error:
`
gst-launch-1.0 -v  videotestsrc is-live=TRUE ! timeoverlay ! videoconvert ! x264enc bframes=0 speed-preset=veryfast key-int-max=30 bitrate=512 ! video/x-h264,stream-format=byte-stream,alignment=au,profile=baseline ! h264parse ! kvssink stream-name="my-stream" iot-certificate="iot-certificate,endpoint=iot-credential-endpoint-host-name,cert-path=/path/to/certificate.pem,key-path=/path/to/private.pem.key,ca-path=/path/to/cacert.pem,role-aliases=KvsCameraIoTRoleAlias"
`

suggestedfixes, 2020-03-25:
v 1.14.5
Let's break it down the items in the quoted string
```
iot-certificate="iot-certificate,endpoint=iot-credential-endpoint-host-name,cert-path=/path/to/certificate.pem,key-path=/path/to/private.pem.key,ca-path=/path/to/cacert.pem,role-aliases=KvsCameraIoTRoleAlias"
```

iot-certificate - verbatim or path?
endpoint=iot-credential-endpoint-host-name - You explanation: no https:// prefix

suggestedfixes, 2020-03-25:
I tried this and expect a different error, still getting "unexpected ASSIGNMENT", I think it might have to do with how the os processes quotes
```
#!/bin/bash
gst-launch-1.0 -v \
location=rtsp://valid-url short-header=TRUE ! \
rtph264depay !  \
video/x-h264,stream-format=avc,alignment=au ! \
h264parse !  \
kvssink stream-name=what storage-size=8 \
iot-certificate="blash,endpoint=what,cert-path=/home/,key-path=/home/,ca-path=/home/,role-aliases=test"
```

suggestedfixes, 2020-03-25:
Made it work! Appreciated the help! It was something else in the pipeline that is generating this error. Thanks for confirming! Possibly use "cat" to clarify the format of the endpoint.

closed 2020-03-25 by suggestedfixes