awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #385: Stream Directly from Ip camera to KVS

IpDeveloper101, 2020-03-30:
Hi,

I have an RTSP enabled Ip camera and have been able to successfully stream to KVS using the rtsp url and the gstreamer sample app. However, I don't understand the steps to install producer libraries directly on the Ip camera for it to directly stream to KVS without needing a Windows, Mac, or Linux machine with access to the same network as my camera. In general, if you do have access to an ip camera's firmware/software image, what are the steps to specifically install a KVS producer GStreamer plugin based app on device? Does the device have to have any other requisite software to install the producer sdk?

disa6302, 2020-03-30:
@IpDeveloper101 , 

If you have access to an IP Camera's firmware/software image and you would like to install the SDK directly onto it, you will have to cross compile for the camera specific platform and scp the binaries and related libraries to it. 

We do not have steps for that for the producer SDK at this point of time.

MushMal, 2020-03-30:
@IpDeveloper101 It really depends on the camera manufacturer and the SDK that comes with the device. Once you have the "open" platform that you can install your software, you will have to think about the top level application. You could potentially model the application based on the samples that we provide. Generally, during any device bringup, you should think at the lower level on how you integrate with the security and the media pipeline. Some customers have chosen IoT certificate integration with the devices. Next, the media pipeline on the device is specific to the device and it's firmware. If you have access to the encoder then you could get the frames directly from the encoder and push into the KVS. If, however, your device only outputs RTSP you could potentially create a local RTSP pipeline using GStreamer and then use GStreamer plugin for example.

These are general guidelines. You might want to be more specific.

MushMal, 2020-04-06:
@IpDeveloper101 I am going to close this issue. Please cut a new issue if you need further help

closed 2020-04-06 by MushMal