awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #386: Error code 0x52000060

str3tch, 2020-03-31:
Hi there,

I am using the C++ Producer SDK to stream from an Axis camera. I am getting the following error when doing so:

```
putStreamResultEvent(): Put stream result event.
writeHeaderCallback(): RequestId: e9f5139b-2571-1cf4-bf09-a8746b6bdc0d
Reporting stream error. Errored timecode: 0 Status: 1375731808
continuousRetryStreamErrorReportHandler(): Reporting stream error. Errored timecode: 0 Status: 0x52000060
 getStreamData(): Indicating an end-of-stream for a terminated stream upload handle 4038
 postReadCallback(): Reported end-of-stream for stream ac-cc-8e-b9-99-41. Upload handle: 4038
```

In particular, I see 0x52000060, which the documentation states that I should not receive! :)

"The Producer SDK enforces timestamps, so clients using the SDK should not receive this error." from )https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html)

Can you help me understand why I would be seeing this error, and if there is anything I can do to fix it?

cheers,
Stretch

MushMal, 2020-03-31:
@str3tch Thanks for reporting and I agree the documentation is a little off :). The error really comes from the fact that the last frame timestamp of the fragment plus its duration is equal to the next fragments key frame timestamp. The SDK enforces greater or equal whereas the backend enforces string monotonicity. We are looking into this and will try to get this right ASAP.

Meanwhile, does the version of the SDK on these cameras properly auto-recover and skip over the "bad" fragment or it keeps retrying until it fails over at the tail? The latest codebase should skip over at least.

str3tch, 2020-03-31:
I do not see it recover from this, I get nothing at all into KVS.

I will pull the latest from Master, as I see there are a few changes that have gone in recently.

Keen to see if the bugfix corrects this issue.

cheers

MushMal, 2020-03-31:
Many thanks! Could I also ask if you could provide the logs with the issue after you pulled the latest?

str3tch, 2020-03-31:
Hi @MushMal, thanks for responding - attached is a log after running with the latest code.

I have my own code outputting debug well as the KVS producer SDK, so you will see some extra stuff in there. But you will recognise the SDK logging. Note it is not running on a camera, but it runs on a Greengrass server and connects to the Axis cam from there. It streams on a schedule, and for this particular camera, it streams for 20 seconds every 5 minutes. In the debug y
[kvs.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4408804/kvs.log)
ou will see some timers firing, which start and stop the streaming.

So as well as the 0x52000060 issue, I also see "FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS", which maybe related?

Also, "Reporting dropped frame. Frame timecode 0" as well, which you can see from around line 5393.. not sure if this is related (I expect so).

Note, this code has been tested on non-Axis cameras and it works well. So I guess there is something about the RTSP stream that the SDK does not like. If you have any insight that would be great!

cheers

MushMal, 2020-04-01:
0x52000060 is an error that's returned from the backend. The text representation that's sent down contains error name which is FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS - the #define in the native code defines it as STATUS_ACK_ERR_FRAGMENT_TIMECODE_NOT_MONOTONIC - those are the same things.

The backend currently enforces Fragment start timestamp which is the key frame timestamp should be strictly greater than the last Fragments last Frame timestamp, assuming that frames are not points in time but have durations. The SDK, however, has a slightly relaxed enforcement which is greater or equal, this still preserves the semantic validity of the stream and the MKV format.

It seems that your pipeline is producing frames which overlap causing the issue.

What's odd is that the logs contain stretches of successful streaming - for instance, all the last part starting from 08:16:05 and going until 08:26:15 has no issues.

I wonder if the RTSP link itself has some latency that produces this bizarre behavior.

Here is my recommendation:

1) Try to dump out the frame timestamps - those would be useful to clearly pinpoint the issue to what I described. You can do that by setting DEBUG_DUMP_FRAME_INFO env variable to true
2) Try to get as much information as possible as to why suddenly you have a couple of runs which are working just fine and some that get this bizarre error. Try to ensure you have a good networking throughput as the bandwidth is effectively doubling by streaming RTSP to the device and KVS stream out of the device. Try a different network, AP or region to see if this makes any difference.
3) We are attempting to track whether this might be caused by the RTSP GStreamer pipeline producing frames which are slightly apart from each other but due to a millisecond timestamp scale get squashed together to the same timestamp. If so, we are trying to remedy it by attempting to use the higher precision time code scale. Unfortunately, the default millisecond granularity allows to encode frames with signed 16 bit values which makes it around 32 second fragments. If we reduce it to 100micros granularity, we could only encode 3.2 second fragments max.

We will update this issue as we get more information.

ib1368, 2020-04-02:
I receive STATUS_ACK_ERR_FRAGMENT_TIMECODE_NOT_MONOTONIC errors as well. Happens a lot despite most of the time the SDK works. is there any active PR on fixing this?
Is https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/316 related?

MushMal, 2020-04-02:
@ib1368 I don't think these are related. The main issue that I have is that I am not sure how the RTSP plugin generates the timestamps on various edge cases - we are unfortunately relying on it for the correctness. 

Also, just to confirm - both in your cases, you are using video or audio/video?

ib1368, 2020-04-06:
@MushMal I have both video/audio and only video streams. I should check if it happens on both of them.
I found this commit: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/commit/21711fb8453631eb567f39f9424fb8fc4b8a95cf Does it fix the issue?

MushMal, 2020-04-06:
Yes, this would reduce the changes of hitting the main issue in multi-track case but won't entirely eliminate it. Could you please give it a try? We are trying to collect as much data as possible as it's hard to hit these cases.

MushMal, 2020-04-06:
@str3tch Could I ask you to run a verbose logging and send the following info to kinesis-video-support@amazon.com or paste the info here?

Account ID, Stream Name, Region, Time, Logs (most of these will be available in the logs anyway)

str3tch, 2020-04-14:
Hi @MushMal , here are some logs. Whilst taking these I did notice something.

We have used the gst demo multistream app as a base to be able to do multiple streams at once. When we see this issue, there are two rtsp streams pushing to KVS. We occasionally see the error in this case for ip 10.240.100.33. here is the log (two streams running concurrently): 
[two streams.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4473748/two.streams.log)

Here is just one of the streams, and there is no error for 10.240.100.33: 
[single.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4473750/single.log)

So it could be that callback is getting out of sync and gets a frame of data from the other source? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/824de900c2d4b42a403319ace5de9d68ec88b171/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_multistream_app.cpp#L203

Can you see anything in the logs that would indicate this is the case?

cheers! :)

MushMal, 2020-04-14:
@str3tch I looked at the two streams log and I really don't see anything wrong with the exception of the timestamps. The KVS SDK objects are either deeply isolated or they have proper interlocking - at least thats what the design is. So, there is no chance for the interfering with each other directly. I also assume GST is properly handling the multi-stream case. The timestamps in this case is generated from the RTSP plugin and I just don't know what it does - not familiar with it's source code.

One of the possibilities for the error could be the source side network congestion causing the bursty behavior on the RTSP side or dropping of the packets. So, your device needs the quadruple of the bandwidth to stream two RTSP streams (2 x the source and 2x the KVS out). 

Is there a way you could configure your RTSP cameras to produce less dense streams? Currently, the cumulative transfer rate is over 1.7 Mbps.

Also, one strange thing I noticed is the high frame rate - at times the combined frame rate is over 7K FPS. This might be due to multi-track audio though.

ib1368, 2020-04-16:
@MushMal I tested the latest revision. The number of 0x52000060 is much higher than v2.1.0

MushMal, 2020-04-18:
@ib1368 I am not really sure whether we can attribute it to the latest commits or it's the RTSP element producing "clumped" frames. I suggested to @str3tch to use lower bitrate and higher bandwidth to see if this helps with the RTSP element.

MushMal, 2020-06-26:
I am going to resolve this issue as it's not immediately actionable on our side.

closed 2020-06-26 by MushMal